ros-industrial/ros2_canopen #23: ros2_control integration

ipa-cmh, 2022-05-10:
**Feature request**
The CANopen stack should expose a ros2_control hardware interface, in order to make control of motion devices easier. Specifically, a SystemInterface could be interesting for enabling control of CANopen-based modular robot arms in ROS2.

The SystemInterface should be integrated into the URDF for example like this:
```xml
<ros2_control name=“CanopenSystemInterface" type="system">
  <hardware>
   <plugin>canopen_control/CanopenSystemInterface</plugin>
   <param name="bus_configuration">[bus.yml]</param>
   <param name="can_interface">can0</param>
   <param name="master_configuration">[master.dcf]</param>
   <param name="master_bin">[master.bin]</param>
  </hardware>
  <joint name="joint1">
   <command_interface name="position">
   </command_interface>
   <state_interface name="position"/>
  </joint>
  <joint name="joint2">
​    <command_interface name="position">
​    </command_interface>
    <state_interface name="position"/>
  </joint>
 </ros2_control>
```

Conventions:
* joint names need to correspond to the name of a device indicated in bus.yml

tonynajjar, 2022-05-13:
Hi, I'm very new to canopen so excuse my potentially wrong jargon. 

I have ros2_control all set up and am now looking to create my own hardware interface. I understand hardware interface to be the layer that converts values received by the controller (speeds, angles) to something the specific hardware can understand (e.g conversion from speeds to bytes), and I imagine that's why each hardware needs its own hardware interface. So my question is, what would CanopenSystemInterface do exactly and when would I use it? Thanks

ipa-cmh, 2022-05-13:
Yes. Basically, we aim to setup a ros2_control SystemInterface, that enables interfacing with all motion controller devices that follow CIA402 profile on your bus.

rbordalba, 2022-07-26:
when are you planning to add this feature to ros2_canopen? (approx)

Great work!

ipa-cmh, 2022-07-26:
We are currently working on it - no PR yet though. First alpha release for testing ros2_control system interface is planned for end of August.

closed 2022-09-16 by ipa-cmh

jclinton830, 2023-03-14:
Hi @ipa-cmh,

Have you published any detailed documentation on how to integrate ros2_canopen and ros2_control yet? I see that in your official manual, the ROS2 Control System Interface section is currently blank. The only way to understand how all of this works currently is to go through the code within canopen_ros2_control package.