ros-industrial/ros2_canopen #26: what():  CanController: Operation not permitted

AlessioMosca, 2022-06-13:
HI there, 
I am trying to use the ros2_canopen package in order to make the my configuration package on my PC (ubuntu 20.04 with ros galactic).
However,  after each point in the guide (https://ros-industrial.github.io/ros2_canopen/) when I run the launch script, i get the following error:

[INFO] [launch]: All log files can be found below /home/drfly/.ros/log/2022-06-13-15-32-43-174646-dti-12659-717498
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [device_manager_node-1]: process started with pid [717500]
[device_manager_node-1] [INFO] [1655127163.274893720] [device_manager_node]: Found 1 devices
[device_manager_node-1] [INFO] [1655127163.275023470] [device_manager_node]: Found Master.
[device_manager_node-1] [INFO] [1655127163.275188521] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_manager_node-1] [INFO] [1655127163.275655526] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1655127163.275683741] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1655127163.277072980] [device_manager_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_manager_node-1] [INFO] [1655127163.277106006] [device_manager_node]: Adding master with node id 1
[device_manager_node-1] terminate called after throwing an instance of 'std::system_error'
[device_manager_node-1]   what():  CanController: Operation not permitted
[ERROR] [device_manager_node-1]: process has died [pid 717500, exit code -6, cmd '/home/drfly/dev_ws/install/canopen_core/lib/canopen_core/device_manager_node --ros-args -r __node:=device_manager_node -r __ns:=/ --params-file /tmp/launch_params_e5ilwpr5'].

attached my configuration package

[everest.zip](https://github.com/ros-industrial/ros2_canopen/files/8892139/everest.zip)

ipa-cmh, 2022-06-20:
> HI there,
> I am trying to use the ros2_canopen package in order to make the my configuration package on my PC (ubuntu 20.04 with ros galactic).
> However, after each point in the guide [...]

This seems to be a problem with permissions for can0. Try running it as administrator for the moment that should solve this issue. We are working on resolving the problem.

ipa-cmh, 2022-06-20:
Just some comments to the code you posted:

**bus.yaml**:
```yaml
  sdo: # SDO executed during config
  - {index: 0x6081, sub_index: 0, value: 500000} # Set velocity
  - {index: 0x6083, sub_index: 0, value: 1000000} # Set acceleration
  - {index: 0x6083, sub_index: 0, value: 1000000} # Set deceleration
  - {index: 0x6085, sub_index: 0, value: 1000000} # Set quickstop deceleration
  - {index: 0x6098, sub_index: 0, value: 35} # Set default homing mode to 35
  - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
  - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
```
This part is at least to a degree device specific. We tested it with a steppper motor, depending on your device, this needs to be adapted.

AlessioMosca, 2022-06-20:
Dear @ipa-cmh,
Many thanks for the answer.

> > HI there,
> > I am trying to use the ros2_canopen package in order to make the my configuration package on my PC (ubuntu 20.04 with ros galactic).
> > However, after each point in the guide (https://ros-industrial.github.io/ros2_canopen/) when I run the launch script, i get the following error:
> > [INFO] [launch]: All log files can be found below /home/drfly/.ros/log/2022-06-13-15-32-43-174646-dti-12659-717498
> > [INFO] [launch]: Default logging verbosity is set to INFO
> > [INFO] [device_manager_node-1]: process started with pid [717500]
> > [device_manager_node-1] [INFO] [1655127163.274893720] [device_manager_node]: Found 1 devices
> > [device_manager_node-1] [INFO] [1655127163.275023470] [device_manager_node]: Found Master.
> > [device_manager_node-1] [INFO] [1655127163.275188521] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
> > [device_manager_node-1] [INFO] [1655127163.275655526] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
> > [device_manager_node-1] [INFO] [1655127163.275683741] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
> > [device_manager_node-1] [INFO] [1655127163.277072980] [device_manager_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
> > [device_manager_node-1] [INFO] [1655127163.277106006] [device_manager_node]: Adding master with node id 1
> > [device_manager_node-1] terminate called after throwing an instance of 'std::system_error'
> > [device_manager_node-1] what(): CanController: Operation not permitted
> > [ERROR] [device_manager_node-1]: process has died [pid 717500, exit code -6, cmd '/home/drfly/dev_ws/install/canopen_core/lib/canopen_core/device_manager_node --ros-args -r __node:=device_manager_node -r __ns:=/ --params-file /tmp/launch_params_e5ilwpr5'].
> > attached my configuration package
> 
> This seems to be a problem with permissions for can0. Try running it as administrator for the moment that should solve this issue. We are working on resolving the problem.
> 
> ```shell
> sudo su
> source install/setup.bash
> ros2 launch your_package your_launch_file
> ```
> 
> This error usually occurs when lely_canopen needs to reset the txqueue size. You can try to solve this without granting root permissions to the process by using the following commands:
> 
> ```shell
> sudo ip link set vcan0 down
> sudo ip link set vcan0 txqueuelen 1000
> sudo ip link set vcan0 up
> ```

I follow your suggestion and I was able to solve the problem concerning the permision of the can0.
However, a new error arises if I run the launch file by using sudo su, please see the following error:

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-06-20-18-43-28-799319-dti-12659-1501168
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [device_manager_node-1]: process started with pid [1501170]
[device_manager_node-1] [INFO] [1655743408.877121786] [device_manager_node]: Found 2 devices
[device_manager_node-1] [INFO] [1655743408.877175382] [device_manager_node]: Found Master.
[device_manager_node-1] [INFO] [1655743408.877274564] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_manager_node-1] [INFO] [1655743408.877761685] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1655743408.877793673] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1655743408.879622360] [device_manager_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_manager_node-1] [INFO] [1655743408.879637829] [device_manager_node]: Adding master with node id 1
[device_manager_node-1] error: /home/drfly/dev_ws/install/everest/share/everest/launch/../config/master.dcf: No such file or directory
[device_manager_node-1] error: /home/drfly/dev_ws/install/everest/share/everest/launch/../config/master.bin: No such file or directory
[device_manager_node-1] terminate called after throwing an instance of 'std::system_error'
[device_manager_node-1]   what():  Device: No such file or directory
[ERROR] [device_manager_node-1]: process has died [pid 1501170, exit code -6, cmd '/home/drfly/dev_ws/install/canopen_core/lib/canopen_core/device_manager_node --ros-args -r __node:=device_manager_node -r __ns:=/ --params-file /tmp/launch_params_b78xfbal'].
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-06-20-18-43-28-799319-dti-12659-1501168
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [device_manager_node-1]: process started with pid [1501170]
[device_manager_node-1] [INFO] [1655743408.877121786] [device_manager_node]: Found 2 devices
[device_manager_node-1] [INFO] [1655743408.877175382] [device_manager_node]: Found Master.
[device_manager_node
[everest.zip](https://github.com/ros-industrial/ros2_canopen/files/8942421/everest.zip)
-1] [INFO] [1655743408.877274564] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_manager_node-1] [INFO] [1655743408.877761685] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[... cut ...]
In the previous message I forgot to write that I am using the PCAN M2 -Peak interface (https://www.peak-system.com/PCAN-M-2.473.0.html?&L=1#:~:text=The%20PCAN%2DM.,)%2C%20and%20compact%20embedded%20applications.) and I would like to ask if it is correct if I run the peak driver by following commands?

```
 sudo modprobe peak_pci
 sudo ip link set can0 up type can bitrate 1000000
 sudo ip link set can0 txqueuelen 1000
 sudo ip link set up can0

```

The ros workspace folder has the following structure:

[image]

[everest.zip](https://github.com/ros-industrial/ros2_canopen/files/8942521/everest.zip)

Many thanks for your support :)

ipa-cmh, 2022-06-21:
> In the previous message I forgot to write that I am using the PCAN M2 -Peak interface [...]

This looks good to me, your device seems to support 1000kbit/s.

Regarding the generation, it could be that cmake configure is not run on your package. Therefore, to generate the dcf file please try the follwing first:
```bash
colcon build --packages-select everest --cmake-clean-first --cmake-force-configure
```
After that check in [ros_workspace]/install/everest/share/everest/config whether the dcf file was generated.
If there is no master.dcf in that folder, generation did not work using cmake, in that case, please post the build log!

You can manually generate the dcf.
* Source install/setup.bash (necessary as dcfgen is installed with lely_core_libraries)
* In terminal navigate to the config folder in everest package
* execute `dcfgen -rS bus.yml`
* copy generated files to [ros_workspace]/install/everest/share/everest/config

AlessioMosca, 2022-06-22:
Dear @ipa-cmh , many thanks for your supports.

So, I tried to follow your advice and when I run the launch script I got the following error:

```
root@dti-12659:/home/drfly/dev_ws# source install/setup.bash 
root@dti-12659:/home/drfly/dev_ws# ros2 launch everest bring_up.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-06-22-11-03-11-137695-dti-12659-37469
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [device_manager_node-1]: process started with pid [37471]
[device_manager_node-1] [INFO] [1655888591.207128959] [device_manager_node]: Found 2 devices
[device_manager_node-1] [INFO] [1655888591.207239160] [device_manager_node]: Found Master.
[device_manager_node-1] [INFO] [1655888591.207341020] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_manager_node-1] [INFO] [1655888591.207780275] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1655888591.207795547] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1655888591.209337719] [device_manager_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_manager_node-1] [INFO] [1655888591.209352033] [device_manager_node]: Adding master with node id 1
[device_manager_node-1] NMT: entering reset application state
[device_manager_node-1] NMT: entering reset communication state
[device_manager_node-1] NMT: running as master
[device_manager_node-1] NMT: entering pre-operational state
[device_manager_node-1] NMT: entering operational state
[device_manager_node-1] [INFO] [1655888591.223670321] [device_manager_node]: Found Driver ros2_canopen::MotionControllerDriver with lazy_load false
[device_manager_node-1] [INFO] [1655888591.223758978] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_402_driver/lib/libcanopen_402_driver.so
[device_manager_node-1] [INFO] [1655888591.225951199] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_manager_node-1] [INFO] [1655888591.225965837] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_manager_node-1] [INFO] [1655888591.227344279] [device_manager_node]: Added node of type ros2_canopen::MotionControllerDriver with name "canopen_402_driver" for node_id 2 to executor.
[device_manager_node-1] [INFO] [1655888591.227377002] [device_manager_node]: Adding ros2_canopen::MotionControllerDriver for node id 2 to master loop.
[device_manager_node-1] [INFO] [1655888591.232034353] [device_manager_node]: List of active components:
[device_manager_node-1] [INFO] [1655888591.232053613] [device_manager_node]: 1 : /master
[device_manager_node-1] [INFO] [1655888591.232079725] [device_manager_node]: 2 : /canopen_402_driver
[device_manager_node-1] [INFO] [1655888591.232085602] [device_manager_node]: Initialisation successful.
[device_manager_node-1] warning: error writing CAN frame: Operation canceled
[device_manager_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_manager_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_manager_node-1] [INFO] [1655888593.232554393] [canopen_402_driver]: Intitialising Device and Objects
[device_manager_node-1] Initialised object : 6041  0 RPDO: yes TPDO: no 
[device_manager_node-1] terminate called after throwing an instance of 'lely::canopen::SdoError'
[device_manager_node-1]   what():  SubmitRead:02:6040:00: Resource not available: SDO connection (060A0023): Resource not available: SDO connection
[ERROR] [device_manager_node-1]: process has died [pid 37471, exit code -6, cmd '/home/drfly/dev_ws/install/canopen_core/lib/canopen_core/device_manager_node --ros-args -r __node:=device_manager_node -r __ns:=/ --params-file /tmp/launch_params_lx1wlyve'].
```
Is it normal since I have no device device connected to the can bus network (note that i am using the eds and yaml file provided in the canopen_402 package)?

Is it possible to run multiple master on the same pc ? I have equipped my pc with a PCI Canbus with 4 channel and I would like to have a master for each channel. Do you know if it possible ?

Many thanks 
Best regards 

Alessio

ipa-cmh, 2022-06-23:
> Is it normal since I have no device device connected to the can bus network (note that i am using the eds and yaml file provided in the canopen_402 package)?

If you do not have a device connected to the CAN bus this looks correct. Basically the SDO requests for configuration are not answered and timeout.

> Is it possible to run multiple master on the same pc ? I have equipped my pc with a PCI Canbus with 4 channel and I would like to have a master for each channel. Do you know if it is possible ?

It is possible, but due to alpha state of the package, we have not tested it yet.
You would need to launch multiple masters and have multiple configuration files, one for each channel.

ipa-cmh, 2022-06-28:
I'll close this issue, as the original problem seems to be solved by #29

closed 2022-06-28 by ipa-cmh