ros-industrial/ros2_canopen #36: Launch File error

AlessioMosca, 2022-08-08:
Hi there,
i followed the Launch configuration creation guidelines (https://ros-industrial.github.io/ros2_canopen/configuration-package.html) but I am not able to run the master (bring_up.launch.py)

```
ros2 launch everest bring_up.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-08-08-16-52-54-408665-dti-12659-81500
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [device_manager_node-1]: process started with pid [81502]
[device_manager_node-1] [INFO] [1659970374.481639696] [device_manager_node]: Found 2 devices
[device_manager_node-1] [INFO] [1659970374.481656996] [device_manager_node]: Found Master.
[device_manager_node-1] [INFO] [1659970374.481755746] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_manager_node-1] [INFO] [1659970374.482189102] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1659970374.482203307] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1659970374.483814747] [device_manager_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_manager_node-1] [INFO] [1659970374.483850305] [device_manager_node]: Adding master with node id 1
[device_manager_node-1] NMT: entering reset application state
[device_manager_node-1] NMT: entering reset communication state
[device_manager_node-1] NMT: running as master
[device_manager_node-1] NMT: entering pre-operational state
[device_manager_node-1] NMT: entering operational state
[device_manager_node-1] [INFO] [1659970374.498616977] [device_manager_node]: Found Driver ros2_canopen::MotionControllerDriver with lazy_load false
[device_manager_node-1] [INFO] [1659970374.498690897] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_402_driver/lib/libcanopen_402_driver.so
[device_manager_node-1] [INFO] [1659970374.501282969] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_manager_node-1] [INFO] [1659970374.501302142] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_manager_node-1] [INFO] [1659970374.503049949] [device_manager_node]: Added node of type ros2_canopen::MotionControllerDriver with name "canopen_402_driver" for node_id 2 to executor.
[device_manager_node-1] [INFO] [1659970374.503084201] [device_manager_node]: Adding ros2_canopen::MotionControllerDriver for node id 2 to master loop.
[device_manager_node-1] [INFO] [1659970374.508052168] [device_manager_node]: List of active components:
[device_manager_node-1] [INFO] [1659970374.508075242] [device_manager_node]: 1 : /master
[device_manager_node-1] [INFO] [1659970374.508103012] [device_manager_node]: 2 : /canopen_402_driver
[device_manager_node-1] [INFO] [1659970374.508112513] [device_manager_node]: Initialisation successful.
[device_manager_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_manager_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_manager_node-1] [INFO] [1659970376.508307248] [canopen_402_driver]: Intitialising Device and Objects
[device_manager_node-1] Initialised object : 6041  0 RPDO: yes TPDO: no 
[device_manager_node-1] terminate called after throwing an instance of 'lely::canopen::SdoError'
[device_manager_node-1]   what():  SubmitRead:02:6040:00: Resource not available: SDO connection (060A0023): Resource not available: SDO connection
[ERROR] [device_manager_node-1]: process has died [pid 81502, exit code -6, cmd '/home/drfly/dev_ws/install/canopen_core/lib/canopen_core/device_manager_node --ros-args -r __node:=device_manager_node -r __ns:=/ --params-file /tmp/launch_params_6k6m3zc1'].

```

Before running the script I have also bring up the can interface reported at https://ros-industrial.github.io/ros2_canopen/

I have also tried to connect the drive but nothing has changed. Moreover, I have tried to monitor the can0 throung candump but i did not see any can frame message.

Attached the ros package that i am trying to build
[everest.zip](https://github.com/ros-industrial/ros2_canopen/files/9282988/everest.zip)

Could be a problem of the can peak m2 card (https://www.peak-system.com/PCAN-M-2.473.0.html?&L=1)?

ipa-cmh, 2022-08-10:
This seems to be an issue with the device. 
1. Are you certain, that your everest device is configured for node id 2? 

2. Have you tried communicating manually with device i.e. with cansend and candump? 
This sends an nmt reset command to node 2. You should receive a boot message from the device, if it is on node id 2.
```
cansend can0 000#8102
```

AlessioMosca, 2022-08-11:
Hi,
yes you are right. I was thinking that the node_id was possible to set as you want. I modified the yaml file and then I opened 2 new terminal, where in the first I run "ros2 launch everest bring_up.launch.py" and in the second I only run "candump can0" in order to see what it is writing on can0.
Here the result of the first terminal:

```
root@dti-12659:/home/drfly# ros2 launch everest bring_up.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-08-09-17-44-17-943513-dti-12659-19731
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [device_manager_node-1]: process started with pid [19733]
[device_manager_node-1] [INFO] [1660059858.020406927] [device_manager_node]: can_interface_name_: can0
[device_manager_node-1] [INFO] [1660059858.020724260] [device_manager_node]: Found 2 devices
[device_manager_node-1] [INFO] [1660059858.020757201] [device_manager_node]: Found Master.
[device_manager_node-1] [INFO] [1660059858.020864572] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_manager_node-1] [INFO] [1660059858.021369143] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1660059858.021384804] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_manager_node-1] [INFO] [1660059858.023059638] [device_manager_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_manager_node-1] [INFO] [1660059858.023076686] [device_manager_node]: Adding master with node id 1
[device_manager_node-1] NMT: entering reset application state
[device_manager_node-1] NMT: entering reset communication state
[device_manager_node-1] NMT: running as master
[device_manager_node-1] NMT: entering pre-operational state
[device_manager_node-1] NMT: entering operational state
[device_manager_node-1] [INFO] [1660059858.037976243] [device_manager_node]: Found Driver ros2_canopen::MotionControllerDriver with lazy_load false
[device_manager_node-1] [INFO] [1660059858.038034052] [device_manager_node]: Load Library: /home/drfly/dev_ws/install/canopen_402_driver/lib/libcanopen_402_driver.so
[device_manager_node-1] [INFO] [1660059858.040489325] [device_manager_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_manager_node-1] [INFO] [1660059858.040527518] [device_manager_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_manager_node-1] [INFO] [1660059858.042033955] [device_manager_node]: Added node of type ros2_canopen::MotionControllerDriver with name "canopen_402_driver" for node_id 32 to executor.
[device_manager_node-1] [INFO] [1660059858.042052274] [device_manager_node]: Adding ros2_canopen::MotionControllerDriver for node id 32 to master loop.
[device_manager_node-1] [INFO] [1660059858.047841605] [device_manager_node]: List of active components:
[device_manager_node-1] [INFO] [1660059858.047862197] [device_manager_node]: 1 : /master
[device_manager_node-1] [INFO] [1660059858.047868895] [device_manager_node]: 32 : /canopen_402_driver
[device_manager_node-1] [INFO] [1660059858.047875516] [device_manager_node]: Initialisation successful.
[device_manager_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 20: SDO protocol timed out
[device_manager_node-1] [INFO] [1660059860.047895390] [canopen_402_driver]: Intitialising Device and Objects
[device_manager_node-1] Initialised object :  6041  0 RPDO: yes TPDO: no 
[device_manager_node-1] Initialised object :  6040  0 RPDO: no TPDO: no 
[device_manager_node-1] Initialised object :  6061  0 RPDO: yes TPDO: no 
[device_manager_node-1] Initialised object :  6060  255 RPDO: no TPDO: no 
[device_manager_node-1] Initialised object :  6502  997 RPDO: no TPDO: no 
[device_manager_node-1] [INFO] [1660059860.052672659] [canopen_402_driver]: Fault reset
```
here the second
```
  can0  080   [0] 
  can0  080   [0] 
  can0  620   [8]  2B 40 60 00 00 01 00 00
  can0  5A0   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  620   [8]  2B 40 60 00 00 01 00 00
  can0  5A0   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  620   [8]  2B 40 60 00 00 01 00 00
  can0  5A0   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  620   [8]  2B 40 60 00 00 01 00 00
  can0  5A0   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  620   [8]  2B 40 60 00 00 01 00 00
  can0  5A0   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  620   [8]  2B 40 60 00 00 01 00 00
  can0  5A0   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
[... 370 lines cut ...]
  enable_lazy_load: false
  period: 50
#  tpdo:
#    1:
#      enabled: true
#      transmission: 0x01
#      mapping:
#        - {index: 0x6041, sub_index: 0}
#        - {index: 0x6061, sub_index: 0}
#        - {index: 0x6064, sub_index: 0}
#  rpdo:
#    1:
#      enabled: true
#      mapping:
#        - {index: 0x607a, sub_index: 0}
#        - {index: 0x6081, sub_index: 0}

```
I did not understand why the master did not understand what the everest drive answer. What am I doing wrong ? 

Attached all the package.

Best regards
Alessio
[everest.zip](https://github.com/ros-industrial/ros2_canopen/files/9308737/everest.zip)

ipa-cmh, 2022-08-12:
Your devices node_id is not known. 

You need to set it via lss - see manual https://doc.ingeniamc.com/eve-xcr/manuals/everest-xcr-reference-manual/canopen-servo-drives. ros2_canopen does not support lss.

But you can use this python library and instructions to do so:
https://github.com/christiansandberg/canopen
https://canopen.readthedocs.io/en/latest/lss.html#examples

AlessioMosca, 2022-08-17:
Many thanks,

If I understand well, before running the ros2_canopen I have to configure the drivre by using the python canopen,  correct? 
or I cannot use ros2_canopen with my drive ?

Best Regards
Alessio

ipa-cmh, 2022-08-17:
Yes, your device seems to be only configurable via lss - many other drives have some switch to do that manually.
If you store the configuration, it should be persistent, so you only need to do it once.

By the way if you figure out how to exactly configure you device with lss and python canopen, you could write a short summary and add it to our documentation ;-).

AlessioMosca, 2022-08-23:
Hi @ipa-cmh 
many thanks for your help.

I did what you have suggested and I have configured the drive to have the node_id = 2 and the baudrate = 1 MBit/sec

Since I have noticed that you have update the package I have updated my ros package as you can see in the attached zip (I have followed the new instruction on the site to make the launch, cmake file etc)

However, when i run the launch file i get the following output:

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-08-23-19-03-24-930057-dti-12659-229096
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/drfly/dev_ws/install/everest/share/everest/config/FR_bus/bus.yml
[INFO] [launch.user]: /home/drfly/dev_ws/install/everest/share/everest/config/FR_bus/master.dcf
[INFO] [launch.user]: /home/drfly/dev_ws/install/everest/share/everest/config/FR_bus/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [229107]
[device_container_node-1] [INFO] [1661274205.020811343] [device_container_node]: Found 3 devices
[device_container_node-1] [INFO] [1661274205.020872039] [device_container_node]: Found Master.
[device_container_node-1] [INFO] [1661274205.020972766] [device_container_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_container_node-1] [INFO] [1661274205.022263245] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_container_node-1] [INFO] [1661274205.022281301] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_container_node-1] [INFO] [1661274205.023951895] [device_container_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_container_node-1] [INFO] [1661274205.023969632] [device_container_node]: Adding master with node id 1
[device_container_node-1] NMT: entering reset application state
[device_container_node-1] NMT: entering reset communication state
[device_container_node-1] NMT: running as master
[device_container_node-1] NMT: entering pre-operational state
[device_container_node-1] NMT: entering operational state
[device_container_node-1] [INFO] [1661274205.040265796] [device_container_node]: Found Driver ros2_canopen::MotionControllerDriver with lazy_load false
[device_container_node-1] [INFO] [1661274205.040330168] [device_container_node]: Load Library: /home/drfly/dev_ws/install/canopen_402_driver/lib/libcanopen_402_driver.so
[device_container_node-1] [INFO] [1661274205.042779670] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_container_node-1] [INFO] [1661274205.042816890] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_container_node-1] [INFO] [1661274205.044725352] [device_container_node]: Added node of type ros2_canopen::MotionControllerDriver with name "canopen_402_driver" for node_id 2 to executor.
[device_container_node-1] [INFO] [1661274205.044744166] [device_container_node]: Adding ros2_canopen::MotionControllerDriver for node id 2 to master loop.
[device_container_node-1] bad conversion
[device_container_node-1] bad conversion
[device_container_node-1] bad conversion
[device_container_node-1] bad conversion
[device_container_node-1] [ERROR] [1661274205.049932064] [device_container_node]: Error: Bus Configuration has uncomplete configuration for options
[device_container_node-1] [INFO] [1661274205.049948144] [device_container_node]: Initialisation successful.
[device_container_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_container_node-1] [INFO] [1661274207.049979739] [canopen_402_driver]: Intitialising Device and Objects
[device_container_node-1] Initialised object : 6041  0 RPDO: yes TPDO: no 
[device_container_node-1] Initialised object : 6040  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6061  0 RPDO: yes TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 

```
while by running the candump:
```
candump can0
  can0  701   [1]  00
  can0  000   [2]  82 00
  can0  080   [0] 
  can0  602   [8]  40 00 10 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  602   [8]  80 00 10 00 00 00 04 05
  can0  602   [8]  40 00 10 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  602   [8]  80 00 10 00 00 00 04 05
[... 506 lines cut ...]
  can0  080   [0] 
  can0  602   [8]  2B 40 60 00 00 01 00 00
  can0  582   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  602   [8]  2B 40 60 00 00 01 00 00
  can0  582   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 
  can0  080   [0] 
  can0  602   [8]  2B 40 60 00 00 01 00 00
  can0  582   [8]  60 40 60 00 00 00 00 00
  can0  080   [0] 

```
As you can see from the launch terminal output, I get some bad conversation anc SDO error. Moreover, the launch terminal stopped to print out after the initialised object and on the can0 looks like that the same frame is sent.

what am i doing wrong?

Many thanks for your support

Best regards 
Alessio

[everest_test.zip](https://github.com/ros-industrial/ros2_canopen/files/9407607/everest_test.zip)

AlessioMosca, 2022-08-26:
Hi there,
I solved one error

`[device_container_node-1] [ERROR] [1661274205.049932064] [device_container_node]: Error: Bus Configuration has uncomplete configuration for options`
 
Now the output from the launch file is the following:

```
ros2 launch everest eve.launch.py 
[INFO] [launch]: All log files can be found below /home/drfly/.ros/log/2022-08-26-16-51-21-516036-dti-12659-6756
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/drfly/dev_ws/install/everest/share/everest/config/FR_bus/bus.yml
[INFO] [launch.user]: /home/drfly/dev_ws/install/everest/share/everest/config/FR_bus/master.dcf
[INFO] [launch.user]: /home/drfly/dev_ws/install/everest/share/everest/config/FR_bus/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [6767]
[device_container_node-1] [INFO] [1661525481.676812220] [device_container_node]: Found 2 devices
[device_container_node-1] [INFO] [1661525481.676873116] [device_container_node]: Found Master.
[device_container_node-1] [INFO] [1661525481.677345654] [device_container_node]: Load Library: /home/drfly/dev_ws/install/canopen_core/lib/libmaster_node.so
[device_container_node-1] [INFO] [1661525481.684951749] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_container_node-1] [INFO] [1661525481.684969964] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_container_node-1] [INFO] [1661525481.686691160] [device_container_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_container_node-1] [INFO] [1661525481.686713548] [device_container_node]: Adding master with node id 1
[device_container_node-1] NMT: entering reset application state
[device_container_node-1] NMT: entering reset communication state
[device_container_node-1] NMT: running as master
[device_container_node-1] NMT: entering pre-operational state
[device_container_node-1] NMT: entering operational state
[device_container_node-1] [INFO] [1661525481.708587647] [device_container_node]: Found Driver ros2_canopen::MotionControllerDriver with lazy_load false
[device_container_node-1] [INFO] [1661525481.708750051] [device_container_node]: Load Library: /home/drfly/dev_ws/install/canopen_402_driver/lib/libcanopen_402_driver.so
[device_container_node-1] [INFO] [1661525481.728079555] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_container_node-1] [INFO] [1661525481.728098655] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_container_node-1] [INFO] [1661525481.730583580] [device_container_node]: Added node of type ros2_canopen::MotionControllerDriver with name "canopen_402_driver" for node_id 2 to executor.
[device_container_node-1] [INFO] [1661525481.730601883] [device_container_node]: Adding ros2_canopen::MotionControllerDriver for node id 2 to master loop.
[device_container_node-1] [INFO] [1661525481.736706806] [device_container_node]: List of active components:
[device_container_node-1] [INFO] [1661525481.736725516] [device_container_node]: 1 : /master
[device_container_node-1] [INFO] [1661525481.736732569] [device_container_node]: 2 : /canopen_402_driver
[device_container_node-1] [INFO] [1661525481.736738748] [device_container_node]: Initialisation successful.
[device_container_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_container_node-1] [INFO] [1661525483.736980137] [canopen_402_driver]: Intitialising Device and Objects
[device_container_node-1] Initialised object : 6041  0 RPDO: yes TPDO: no 
[device_container_node-1] Initialised object : 6040  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6061  0 RPDO: yes TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 
```

The can frames sent on can0 are the following:

```
  can0  701   [1]  00  // NMT error control
  can0  000   [2]  82 00 //NMT reset com
  
  can0  602   [8]  40 00 10 00 00 00 00 00 		// master to slave read obj 1000
  can0  080   [0] 					// sync mes

  can0  080   [0] 
  can0  602   [8]  80 00 10 00 00 00 04 05 		// master abort time out sdo
  can0  602   [8]  40 00 10 00 00 00 00 00 		// master to slave read obj 1000
  can0  080   [0] 

  can0  080   [0] 
  can0  602   [8]  80 00 10 00 00 00 04 05 		// master abort time out sdo
  can0  602   [8]  40 00 10 00 00 00 00 00 		// master to slave read obj 1000
  can0  080   [0] 

  can0  080   [0] 
  can0  602   [8]  80 00 10 00 00 00 04 05 		// master abort time out sdo
  can0  602   [8]  40 00 10 00 00 00 00 00 		// master to slave read obj 1000
  can0  080   [0] 
[... 34 lines cut ...]
  can0  080   [0] 
  can0  602   [8]  2B 40 60 00 01 25 00 00		// master to slave write in obj 6040 the value 00002501 (????????)
  can0  582   [8]  60 40 60 00 00 00 00 00		// slave to master confirms tha in obj 6040 the value is 0  ??????
  can0  080   [0] 

  can0  602   [8]  2B 40 60 00 01 25 00 00		// master to slave write in obj 6040 the value 00002501 (????????)
  can0  582   [8]  60 40 60 00 00 00 00 00		// slave to master confirms tha in obj 6040 the value is 0  ??????
[... cut ...]
As it can be noticed, it looks like that the master send a request to write in the object 6040 the value 0x2501 but the drive confirms that in the object 6040 the value is 0.

Why the master sent the value 0x2501? it seems that the drive does not know the meaning (https://doc.ingeniamc.com/summit/manuals/summit-reference-manual/summit-register-dictionary/ethercat-canopen-registers/0x6040-control-word)

Also the status word looks strange (https://doc.ingeniamc.com/summit/manuals/summit-reference-manual/summit-register-dictionary/ethercat-canopen-registers/0x6041-status-word)

Do you have suggestion ?

Best regrads 
Alessio

ipa-cmh, 2022-08-29:
Hi Allessio,

First thing, this now looks good on ROS side.
CAN side: I noticed is that your pdos are not periodically send, this is probably related to #39.
Please add ```cob_id: "auto"``` to each of your pdo statements in bus.yml, it seems your pdos are disabled in the eds. 

Also in the bus.yml I saw that you did not configure control word as pdo, which I would suggest.

Regarding the command_word, I don't see the status_word, so its only guessing...

AlessioMosca, 2022-08-29:
Hi @ipa-cmh ,
many thanks.

I have added as you suggested the `cob_id: "auto"` and the control word as pdo.

```
master:
  node_id: 1
  sync_period: 20000
  driver: "ros2_canopen::MasterNode"
  package: "canopen_core"

canopen_402_driver:
  node_id: 2
  dcf: "eve.eds"
  dcf_path: "install/everest/share/everest/config/FR_bus"
  driver: "ros2_canopen::MotionControllerDriver"
  package: "canopen_402_driver"
  enable_lazy_load: false
  period: 50

  #sdo: # SDO executed during config
  #- {index: 0x6040, sub_index: 0, value: 06} # shutdown
  #- {index: 0x6040, sub_index: 0, value: 07} # switchon
  #- {index: 0x6040, sub_index: 0, value: 15} # enable op

  tpdo:
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0}
        - {index: 0x6061, sub_index: 0}
        - {index: 0x6064, sub_index: 0}
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0}
        - {index: 0x606c, sub_index: 0}
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6040, sub_index: 0}

  rpdo:
    1:
      enabled: true
      cob_id: "auto"
      mapping:
        - {index: 0x607a, sub_index: 0}
        - {index: 0x6081, sub_index: 0}

  heartbeat_producer: 1000 # Heartbeat every 1000 ms
```

what do you mean with "Regarding the command_word, I don't see the status_word, so its only guessing..."
?  what is it ?  maybe is the control word ? if so, it is not configured by the first tpdo mapping ? is not the 0x6040 register ?

Best regards  and many thanks for your help

Alessio

AlessioMosca, 2022-08-29:
Hi @ipa-cmh,

so I have tried to call a service manually (/canopen_402_driver/nmt_start_node) and after I have called it the internal state 402 is changed form Not_Ready_To_Switch_On to Switch_On_Disabled. After this. I have called the service (/canopen_402_driver/init) and the InternalState State402 is changed form Switch_On_Disabled -> Ready_To_Switch_On -> Switched_On -> Operation_Enable (as you can see from the terminal output ).  However, I got the following error `EMCY: 8220 11` and I do not understand why.

```

[device_container_node-1] Switch_On_Disabled
[device_container_node-1] Switch_On_Disabled
[device_container_node-1] Switch_On_Disabled 
[device_container_node-1] [INFO] [1661782918.763290436] [canopen_402_driver]: Init: Read State
[device_container_node-1] Switch_On_Disabled 
[device_container_node-1] [INFO] [1661782918.764597223] [canopen_402_driver]: Init: Enable
[device_container_node-1] Switch_On_Disabled 
[device_container_node-1] [INFO] [1661782918.790052845] [canopen_402_driver]: Fault reset
[device_container_node-1] EMCY: 8220 11
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] Ready_To_Switch_On
[device_container_node-1] EMCY: 8220 11
[device_container_node-1] Switched_On
[device_container_node-1] Switched_On
[device_container_node-1] EMCY: 8220 11
[device_container_node-1] Operation_Enable
[device_container_node-1] [INFO] [1661782919.388626949] [canopen_402_driver]: Init: Switch to homing
[device_container_node-1] Operation_Enable
[device_container_node-1] [INFO] [1661782919.440063250] [canopen_402_driver]: Init: Execute homing
[device_container_node-1] [INFO] [1661782919.442849671] [canopen_402_driver]: Init: Switch no mode
[device_container_node-1] Initialised object : 6041  17975 RPDO: yes TPDO: no   --> STATUS WORD 17975 = Operation enable
[device_container_node-1] Initialised object : 6040  271 RPDO: yes TPDO: no       -->  CONTROL WORD = op enable
[device_container_node-1] EMCY: 8220 11                                                              --> error PDO length exceeded 
[device_container_node-1] Initialised object : 6061  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6071  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 60ff  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 607a  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 60c1  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6098  0 RPDO: no TPDO: no 
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] EMCY: 8220 11
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable

--- many lines of operation_enable

[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] Operation_Enable
[device_container_node-1] EMCY: 8220 11
[device_container_node-1] Fault_Reaction_Active
[device_container_node-1] Fault_Reaction_Active
[device_container_node-1] EMCY: 8220 11
[device_container_node-1] Fault
[device_container_node-1] Fault
[device_container_node-1] Fault
[device_container_node-1] Fault
[device_container_node-1] Fault

```

Moreover, 
I did not set all the following PDOs, is it possible that are default PDO in the canopen standard?

```
[device_container_node-1] Initialised object : 6061  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6071  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 60ff  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 607a  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 60c1  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6098  0 RPDO: no TPDO: no 
```
 
Then, is it correct that I have to call manually or from the code the following services:  canopen_402_driver/init and /canopen_402_driver/nmt_start_node? 

Why does the internal state 402 change into [device_container_node-1] Fault_Reaction_Active ? 

Which is the correct behavior of the library ? should I call the service or the master should do it automatically ?  Do you have any example ?

How can I test the code? do you have any suggestion ?

Many thanks for your time 

Alessio

ipa-cmh, 2022-09-02:
Hi @AlessioMosca,

The intended behaviour of the service-based interface is that you call service init, then trigger the Operation Mode (ie. position_mode) and then set targets via service. A ros2_control interface is under development see #33. 

> Moreover,
I did not set all the following PDOs, is it possible that are default PDO in the canopen standard?
```
[device_container_node-1] Initialised object : 6061  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6071  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 60ff  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 607a  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 60c1  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6098  0 RPDO: no TPDO: no 
```
This is verbose output. These are the objects needed by the canopen_402_driver and which transmission type it has available (e.g. pdo or sdo). If an object doesn't say RPDO or TPDO, then SDO is used for both directions. So nothing wrong with this. Only 607a is mapped to a PDO as you specified in your yaml.

> Why does the master ask for the 6061 obj if i have commented in the yaml ?

This is the operation mode - it is necessary for operating the drive with cia402 protocol.

> I have just checked If on the canopne_402 topics are published or not. So, on topics: actual_speed and actual_position I am able to see the data. However, on the remaning topics (nmt_state, rpdo, tpdo) no data are published. If I have understand well, I should see here the data coming from the rpdo and tpdo, or am i wrong?

When using the canopen_402_driver rpdo is not published. Tpdo is only subscribed to.

> Why does the internal state 402 change into [device_container_node-1] Fault_Reaction_Active ?

I cannot say. Best read out the error/fault register of the drive via sdo. You can use the provided service sdo read. This can be anything from not connected voltage or some unconfigured parameter. This is probably not due to the stack but either a misconfiguration or problem with the mechanical setup.

**General recommendation**:
I would recommend the following PDO config:
* status word
* control word
* mode of operation
* mode of operation display
* actual position
* actual velocity
* and the target object of your operation mode (ie. target position)

AlessioMosca, 2022-09-02:
Hi @ipa-cmh 
many thanks or your help.

> The intended behaviour of the service-based interface is that you call service init, then trigger the Operation Mode (ie. position_mode) and then set targets via service. A ros2_control interface is under development see https://github.com/ros-industrial/ros2_canopen/pull/33.

Thus, is it correct that first I have to sent a service /canopen_402_driver/nmt_start_node and then /canopen_402_driver/init  to turn in enable operation the drive ?

> When using the canopen_402_driver rpdo is not published. Tpdo is only subscribed to.

Where can I see the tpdo that I have defined in the bus.yml file ? if i defined as tpdo  for example the status word where can i read it ?
And if I defined a rpdo where can i write the value to be sent to the slave?

Can you help me to understand the following tag ?
- period (under the canopen_402_driver)
- trasmission (under tpdo e rpdo)
- sync_period (under master)

is there any correletion among the above tag?

I have modified the bus.yml file as you suggest an i have add in the first rpdo the request to sent to slave the control word (0x6040). However, with this yml file I am not anymore able to enable the operation of the drive.  If I comment out the 0x6040 as you can see below I return able to put in operation enable the drive. what am i doing wrong ?

```
master:
  node_id: 1
  sync_period: 20000  #
  driver: "ros2_canopen::MasterNode"
  package: "canopen_core"
  start_all_nodes: true

canopen_402_driver:
  node_id: 2
  dcf: "eve.eds"
  dcf_path: "install/everest/share/everest/config/FR_bus"
  driver: "ros2_canopen::MotionControllerDriver"
  package: "canopen_402_driver"
  enable_lazy_load: false
  period: 50

  #sdo: # SDO executed during config
  #- {index: 0x6040, sub_index: 0, value: 06} # shutdown
  #- {index: 0x6040, sub_index: 0, value: 07} # switchon
  #- {index: 0x6040, sub_index: 0, value: 15} # enable op

  # data sent from slave to master dal pdv master
  tpdo:
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # Status Word
        - {index: 0x2015, sub_index: 0} # Operation mode display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0} # Actual position
        - {index: 0x2031, sub_index: 0} # Actual velocity
    3:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x2029, sub_index: 0} # Actual torque

  # data sent from master to slave
  rpdo:
    1:
      enabled: true
      cob_id: "auto"
      mapping:
#        - {index: 0x6040, sub_index: 0} # Control Word   ----------> makes no possible to switch in operation enable
        - {index: 0x2014, sub_index: 0} # Operation Mode
    2:
      enabled: true
      cob_id: "auto"
      mapping:
        - {index: 0x607A, sub_index: 0} # Position set-point
        - {index: 0x60FF, sub_index: 0} # Velocity set-point
    3:
      enabled: true
      cob_id: "auto"
      mapping:
        - {index: 0x2022, sub_index: 0} # Torque set-point#

  heartbeat_producer: 1000 # Heartbeat every 1000 ms
```

Many thanks for your help

Best Regards
Alessio

ipa-cmh, 2022-09-13:
Hi Alessio,

Some general explanation, we are managing two lifecycles for the canopen nodes.

CIA301 lifecycle: The canopen_master takes care of starting the node via cia302 and the necessary nmt commands. If you specify commands in sdo in the yaml, the are executed in pre-operational mode. Once the master is ready with configuration, all your canopen nodes should be in operational, you can see this in candump in the heartbeat messages.

CIA402 lifecycle: This is managed by the canopen_402_driver if you register it for a device in bus.yml. You can now use the service init which will try to bring the device into enabled state as well as execute the default homing method of your drive. This will automatically emit the necessary 402 command_words to reach enabled state.

CIA402 control mode: You can chose a control mode via the offered services of the ros node. This will emit a message to mode_of_operation.

Once you have chosen a control mode you send a target via the target service.

Regarding the tpdo and rpdo: If you have configured a canopen_402_driver for your device in bus.yml, there is no proxy for rpdo and tpdo to ros messages. If you configure canopen_proxy_driver, rpdo and tpdo can be accessed through the related topic/service.

If you simply want to check what is in an object, you can use the sdo services, these also work with canopen_402_driver.

AlessioMosca, 2022-09-19:
Hi @ipa-cmh,
many thanks for the answer.

do you have any tutorial or example for configuring the proxy ? should I add it in the bus.yaml and in the launch file ?

If I understand well, it is correct to run  a service /canopen_402_driver/nmt_start_node and then /canopen_402_driver/init to turn in enable operation the drive , am I wrong?

What is the difference between the ros2_canopen::MasterNode or ros2_canopen::LifecycleMasterNode ? when it is appropriate to use MasterNode or LifecycleMasterNode, may I ask an example ?

Many thanks for your time.
Best Regards

Alessio

AlessioMosca, 2022-09-19:
Hi @ipa-cmh,

many thanks for your answer.

If I call only the init service what I get from the terminal where I have launched the launch file is:

```
[INFO] [launch]: All log files can be found below /home/drfly/.ros/log/2022-09-19-15-18-45-256145-dti-12659-60546
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/drfly/grace_ws/install/everest/share/everest/config/FR_bus/bus.yml
[INFO] [launch.user]: /home/drfly/grace_ws/install/everest/share/everest/config/FR_bus/master.dcf
[INFO] [launch.user]: /home/drfly/grace_ws/install/everest/share/everest/config/FR_bus/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [60557]
[device_container_node-1] [INFO] [1663593525.350503905] [device_container_node]: Found 2 devices
[device_container_node-1] [INFO] [1663593525.350563805] [device_container_node]: Found Master.
[device_container_node-1] [INFO] [1663593525.350667848] [device_container_node]: Load Library: /home/drfly/grace_ws/install/canopen_core/lib/libmaster_node.so
[device_container_node-1] [INFO] [1663593525.352072800] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_container_node-1] [INFO] [1663593525.352089854] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterNode>
[device_container_node-1] [INFO] [1663593525.354174655] [device_container_node]: Added node of type ros2_canopen::MasterNode with name "master" for node_id 1 to executor.
[device_container_node-1] [INFO] [1663593525.354196310] [device_container_node]: Adding master with node id 1
[device_container_node-1] NMT: entering reset application state
[device_container_node-1] NMT: entering reset communication state
[device_container_node-1] NMT: running as master
[device_container_node-1] NMT: entering pre-operational state
[device_container_node-1] NMT: entering operational state
[device_container_node-1] [INFO] [1663593525.370680266] [device_container_node]: Found Driver ros2_canopen::MotionControllerDriver with lazy_load false
[device_container_node-1] [INFO] [1663593525.370743261] [device_container_node]: Load Library: /home/drfly/grace_ws/install/canopen_402_driver/lib/libcanopen_402_driver.so
[device_container_node-1] [INFO] [1663593525.373150657] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_container_node-1] [INFO] [1663593525.373168478] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MotionControllerDriver>
[device_container_node-1] [INFO] [1663593525.374801388] [device_container_node]: Added node of type ros2_canopen::MotionControllerDriver with name "canopen_402_driver" for node_id 2 to executor.
[device_container_node-1] [INFO] [1663593525.374819357] [device_container_node]: Adding ros2_canopen::MotionControllerDriver for node id 2 to master loop.
[device_container_node-1] [INFO] [1663593525.379820305] [device_container_node]: List of active components:
[device_container_node-1] [INFO] [1663593525.379840284] [device_container_node]: 1 : /master
[device_container_node-1] [INFO] [1663593525.379853982] [device_container_node]: 2 : /canopen_402_driver
[device_container_node-1] [INFO] [1663593525.379862148] [device_container_node]: Initialisation successful.
[device_container_node-1] error: SDO abort code 05040000 received on upload request of object 1000 (Device type) to node 02: SDO protocol timed out
[device_container_node-1] [INFO] [1663593527.380141754] [canopen_402_driver]: Intitialising Device and Objects
[device_container_node-1] Initialised object : 6041  0 RPDO: yes TPDO: no 
[device_container_node-1] Initialised object : 6040  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 6061  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 
[device_container_node-1] [INFO] [1663593527.390372913] [canopen_402_driver]: Fault reset
[device_container_node-1] [INFO] [1663593534.714999204] [canopen_402_driver]: Init: Read State
[device_container_node-1] [INFO] [1663593534.716601498] [canopen_402_driver]: Init: Enable
[device_container_node-1] [INFO] [1663593534.740183732] [canopen_402_driver]: Fault reset
[device_container_node-1] [INFO] [1663593539.716915713] [canopen_402_driver]: Transition timed out.
[device_container_node-1] Could not enable motor
[device_container_node-1] Initialised object : 6041  0 RPDO: yes TPDO: no 
[device_container_node-1] Initialised object : 6040  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 6061  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6060  255 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6502  997 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6071  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 60ff  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 607a  0 RPDO: no TPDO: yes 
[device_container_node-1] Initialised object : 60c1  0 RPDO: no TPDO: no 
[device_container_node-1] Initialised object : 6098  0 RPDO: no TPDO: no 
```
Moreover, If I modified the bus.yaml file as follows:

```
master:
  node_id: 1
  sync_period: 20000  #
  driver: "ros2_canopen::MasterNode"
  package: "canopen_core"
  start_all_nodes: true

[... cut ...]
[device_container_node-1] [INFO] [1663597471.276553400] [device_container_node]: Load Library: /home/drfly/grace_ws/install/canopen_proxy_driver/lib/libcanopen_proxy_driver.so
[ERROR] [device_container_node-1]: process has died [pid 78203, exit code -11, cmd '/home/drfly/grace_ws/install/canopen_core/lib/canopen_core/device_container_node --ros-args -r __node:=device_container_node -r __ns:=/ --params-file /tmp/launch_params_z5a256wc --params-file /tmp/launch_params_syjwia2d --params-file /tmp/launch_params_rua31ld0 --params-file /tmp/launch_params_bntbwfdb'].

```

Do you have any suggestion ?

Best regard
Alessio

CityAplons, 2022-09-21:
> Hi @ipa-cmh,
>
> so I have tried to call a service manually (/canopen_402_driver/nmt_start_node) and after I have called it the internal state 402 is changed form Not_Ready_To_Switch_On to [...]

Hello, you had there a non-volatile emergency error.
How did you resolve it?

BR

ipa-cmh, 2022-09-22:
@CityAplons 
This error usuallly occurs with lely canopen library, when there is a mismatch between the .eds you have in the bus.yml for driver and master configuraiton and what is actually stored on the device.
This happens for example if you set a rpdo/tpdo configuration in bus.yml, but the setting them by the master failed. In general, we are not throwing errors during configuring of devices done by the master. I will add this as an issue.

CityAplons, 2022-09-22:
Thank you for clarifying this information. At least now I know in what direction to search for :)

AlessioMosca, 2022-09-23:
> 

Dear @ipa-cmh,
do you have any suggestion?

Best regards 

Alessio

AlessioMosca, 2022-09-23:
Hi @ipa-cmh,

I just tried to figure out why I am not able to switch the motor by using the `/canopen_402_driver/init`.
I have tried to switch the state of the machine by using the sdo request and by doing this I am able to enable the motor, set the profile mode and the setpoint.

What am I doing wrong for the `/canopen_402_driver/init` ? 

Moreover, by adding in the bus.yaml the following lines:

```
proxy_device_1:
  node_id: 4
  dcf: "eve.eds"
  dcf_path: "install/everest/share/everest/config/FR_bus"
  driver: "ros2_canopen::ProxyDriver"
  package: "canopen_proxy_driver"
  enable_lazy_load: false
```
when i run the launch i get the following error:

`[ERROR] [device_container_node-1]: process has died [pid 19524, exit code -11, cmd '/home/drfly/test_ws/install/canopen_core/lib/canopen_core/device_container_node --ros-args -r __node:=device_container_node -r __ns:=/ --params-file /tmp/launch_params_1k0xojee --params-file /tmp/launch_params_w5mnvixq --params-file /tmp/launch_params_q2m7pl4c --params-file /tmp/launch_params_2p6a6y4w'].
`

do you have any suggestion?

Many thanks for you help

Best regards 
Alessio

ipa-cmh, 2022-10-11:
Hi Alessio,

Proxydriver should replace the MotionControllerDriver not be an additional node. You simply use ProxyDriver instead of the MotionControllerDriver.

Just change package and driver under canopen_402_driver, do not add an additional node that has no corresponding device on the bus.

The proxydriver does not have the capabilities of the MotionControllerDriver. It only gives you the possiblity to interact with the CANopenDevice using sdo and pdo manually. If you want to move the motor, you'll need to send command words manually via the provided sdo service or pdo topic.

Regarding the error:
The initialisation failed, when the driver tried to put the device in enabled state.
The statusword indicates, your device is "not ready to switch on". That is not good as it should switch to "switch on disabled" automatically (according to cia402).

To make sure, you can read status_word with the sdo_read service.

```

ros2 service call <device_name>/sdo_read canopen_interfaces/srv/CORead "{index: 0x6041, subindex: 0, type: 16}"

```

If this returns 0, the device is not ready to switch on.

Check in the devices documentation for preconditions for the transition "not ready to switch" -> "switch on disabled". Maybe as well check the devices error register.

```

ros2 service call <device_name>/sdo_read canopen_interfaces/srv/CORead "{index: 0x1001, subindex: 0, type: 8}"

```

Best

Christoph

closed 2023-04-24 by ipa-cmh