ros-industrial/ros2_canopen #55: compilation error

AlessioMosca, 2022-09-20:
Hi there,
I  downloaded he last version available on the mastar branch and when I run `colcon build ` I got the following compilation error:

```
Starting >>> canopen_interfaces
Starting >>> lely_core_libraries
Finished <<< canopen_interfaces [5.99s]                                                                         
Starting >>> canopen_utils
Finished <<< canopen_utils [0.87s]                                                              
--- stderr: lely_core_libraries                                 
Cloning into 'upstr_lely_core_libraries'...
Note: switching to 'ac31cd5cdf2ded4b2a5ba4a94520f388450614ff'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at ac31cd5c release v2.3.1
configure.ac:17: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'
configure.ac:606: installing './tap-driver.sh'
src/can/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
libtool: warning: relinking 'liblely-tap.la'
libtool: warning: relinking 'liblely-util.la'
libtool: warning: relinking 'liblely-can.la'
libtool: warning: relinking 'liblely-co.la'
libtool: warning: relinking 'liblely-io.la'
libtool: warning: relinking 'liblely-ev.la'
libtool: warning: relinking 'liblely-io2.la'
libtool: warning: relinking 'liblely-coapp.la'
---
Finished <<< lely_core_libraries [17.5s]
Starting >>> canopen_core
Starting >>> canopen_mock_slave
Finished <<< canopen_mock_slave [3.85s]                                                                   
--- stderr: canopen_core                               
In file included from /home/drfly/test_ws/src/ros2_canopen/canopen_core/src/device_container_node.cpp:16:
/home/drfly/test_ws/src/ros2_canopen/canopen_core/include/canopen_core/device_container_node.hpp:68:5: error: ‘virtual void DeviceContainerNode::on_list_nodes(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<composition_interfaces::srv::ListNodes_Request_<std::allocator<void> > >, std::shared_ptr<composition_interfaces::srv::ListNodes_Response_<std::allocator<void> > >)’ marked ‘override’, but does not override
   68 |     on_list_nodes(
      |     ^~~~~~~~~~~~~
/home/drfly/test_ws/src/ros2_canopen/canopen_core/src/device_container_node.cpp: In member function ‘bool DeviceContainerNode::init()’:
/home/drfly/test_ws/src/ros2_canopen/canopen_core/src/device_container_node.cpp:211:11: error: ‘rclcpp::Service<composition_interfaces::srv::LoadNode>::SharedPtr rclcpp_components::ComponentManager::loadNode_srv_’ is private within this context
  211 |     this->loadNode_srv_.reset();
      |           ^~~~~~~~~~~~~
In file included from /home/drfly/test_ws/src/ros2_canopen/canopen_core/include/canopen_core/device_container_node.hpp:22,
                 from /home/drfly/test_ws/src/ros2_canopen/canopen_core/src/device_container_node.cpp:16:
/opt/ros/galactic/include/rclcpp_components/component_manager.hpp:190:40: note: declared private here
  190 |   rclcpp::Service<LoadNode>::SharedPtr loadNode_srv_;
      |                                        ^~~~~~~~~~~~~
/home/drfly/test_ws/src/ros2_canopen/canopen_core/src/device_container_node.cpp:212:11: error: ‘rclcpp::Service<composition_interfaces::srv::UnloadNode>::SharedPtr rclcpp_components::ComponentManager::unloadNode_srv_’ is private within this context
  212 |     this->unloadNode_srv_.reset();
      |           ^~~~~~~~~~~~~~~
In file included from /home/drfly/test_ws/src/ros2_canopen/canopen_core/include/canopen_core/device_container_node.hpp:22,
                 from /home/drfly/test_ws/src/ros2_canopen/canopen_core/src/device_container_node.cpp:16:
/opt/ros/galactic/include/rclcpp_components/component_manager.hpp:191:42: note: declared private here
  191 |   rclcpp::Service<UnloadNode>::SharedPtr unloadNode_srv_;
      |                                          ^~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/device_container.dir/build.make:63: CMakeFiles/device_container.dir/src/device_container_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:171: CMakeFiles/device_container.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< canopen_core [5.32s, exited with code 2]

Summary: 4 packages finished [23.1s]
  1 package failed: canopen_core
  2 packages had stderr output: canopen_core lely_core_libraries
  6 packages not processed

```

I have also tried to download the last release version on a new ros2 workspace, but i got the same error.

Best Regards
Alessio

ipa-cmh, 2022-09-20:
You are using galactic or lower - master only builds for humble onwards, as we are merging in some features (ros2_control) that do not have the necessary feature set in galactic and earlier.

AlessioMosca, 2022-09-21:
Hi @ipa-cmh,
you are right =D i used galatic.

so in order to use ros2_control i should update humble ?

Bets regards
Alessio

closed 2022-10-13 by ipa-cmh