ros-industrial/ros2_canopen #80: Implement CiA 301 Multiple Device Profiles

ciandonovan, 2022-10-25:
**Is your feature request related to a problem? Please describe.**

The ROS2 CANopen Stack supports only a single CiA 301 device profile for each node, with CiA 402 being the current reference implementation, allowing for motion control of a single axle.

Many CiA 402 compliant motor controller drivers though, allow for control of more than a single axle, by using the CiA 301's Multiple Device Modules. The ability to control multiple axles from a single driver is of particular interest to the mobile robotics community, as
it allows lower costs through hardware consolidation and reduces synchronization issues between independent drivers.

Implicitly, the existing ROS2 CANopen stack implements all functionality in the zeroth device profile segment, indices 6000h-67FFh.
However, there are up to 8 segments available (0-7), with the object indices of these subsequent device profile segments offset each by 800h. Listed below is an excerpt from the CiA 301 standard.

[image]

Listed below here is an excerpt from the CiA 402 standard, showing how both CiA 402 motor control profiles and CiA 401 I/O profiles can both be implemented and addressed on a single node.

[image]

Both these specifications can be downloaded in full at https://www.can-cia.org/groups/specations/

The ROS2 CANopen Stack currently hardcodes the object indices for addressing the CiA 402 profile, as shown below.

https://github.com/ros-industrial/ros2_canopen/blob/1b12b2c72e5dee515679bff12234a8aaa3bcfa1d/canopen_402_driver/include/canopen_402_driver/motor.hpp#L525-L529

https://github.com/ros-industrial/ros2_canopen/blob/1b12b2c72e5dee515679bff12234a8aaa3bcfa1d/canopen_402_driver/include/canopen_402_driver/motor.hpp#L243-L249

This makes it impossible to address device profiles in different segments.

**Describe the solution you'd like**
Instead of hardcoding the object indices, the base indices can be mapped to any device profile segment by `Index + (800h * X)`, where X is the device profile segment number (0-7).

ipa-cmh, 2022-11-02:
@ciandonovan sounds like a good new feature.

However, I am not sure if it would be as easy as indicated in the solution. You can add adtitional operation modes in the current driver, but you cannot add additional drivers. This would probably need a little deeper digging than indicated, but it is not impossible. Currently, the 402 driver is a more or less direct port from the ros_canopen stack to reduce efforts and ros_canopen did not support this. 

The good news is, with the lely_canopen stack in the ros2_canopen stack we have the possibility to create logical drivers for devices, which would facilitate the described feature. Logical drivers are registered for a certain address range of a device.

The bad news is, that this would require some restrucutring of lely_motion_controller_bridge.hpp and the 402 driver itself.

I will investigate this a little bit more.

nkmoduga, 2023-08-10:
Hi @ipa-cmh, any update on this feature?

ipa-cmh, 2023-08-11:
@nkmoduga, I did a bit of work on the cia402_driver - splitting it up etc. However, for full integration this will be quite some work and will probably mean writing a cia402_multi_driver. I currently do not have a drive that supports this, no use case and no funding to do the development.

I could coach on the development though if you want to take it up. 
In case you need the feature and have funding available for a drive and ~1 week dev&test time, I can probably get this done as well.

jclinton830, 2024-05-21:
It would be great to have this feature, to be able to use a dual channel motor controller to control two motors in either CIA402 or CIA301 mode. I can dedicate 4 weeks of development for this feature if @hellantos can provide some guidance.

millejade, 2024-08-02:
Hello @hellantos!

This is a clarification about this issue. I apologize if I didn't understand it clearly as I'm not well-versed yet with CiA402.

Does this mean the CiA402 Driver can't be used for motor controllers with multiple axes (or supports multiple motors in a single controller)?

I'm trying to use this package to control at most 6 stepper motors with a single motor controller only. Would that still be possible?

Thanks!

hellantos, 2024-08-02:
Right now no. But you could try and reach @ipa-vsp. We had working driver package implementation for this at Fraunhofer, but it wasn't yet ready to open-source when I left. Maybe it is now...

millejade, 2024-08-06:
> Right now no. But you could try and reach @ipa-vsp. We had working driver package implementation for this at Fraunhofer, but it wasn't yet ready to open-source when I left. Maybe it is now...

This was in ROS2 also, right?

ipa-vsp, 2024-08-15:
@millejade May be soon! Currently under negotiation to open source it.

millejade, 2024-08-16:
Hi @ipa-vsp !

Can you give me some details on it?
Is there an estimated date and scope (like what it can and can't do) for this new feature?

Thank you~

rishavbpm, 2024-08-30:
> @nkmoduga, I did a bit of work on the cia402_driver - splitting it up etc. However, for full integration this will be quite some work and will probably mean writing a cia402_multi_driver. I [...]

> @nkmoduga, I did a bit of work on the cia402_driver - splitting it up etc. However, for full integration this will be quite some work and will probably mean writing a cia402_multi_driver. I [...]

What is the status on this. I have a use case for multi axis driver, I am willing to work on it for 2 weeks, if you can guide me @hellantos .

millejade, 2024-09-03:
Hello, @hellantos and @ipa-vsp!

Hope you're doing well.

I just wanted to confirm that this Multiple Axis support is only specific to CIA402 Driver, is that right?
I might develop my own CiA 402 driver with multiple-axis support as I need more features due to my application aside from what the CIA402 driver has now.
If I create my own CIA402 driver, I can use the proxy driver and master driver, is that right? Please enlighten me if I am in the wrong path hehe.

Thanks~

millejade, 2024-10-07:
Hello @hellantos ~

This is regarding the multiple axes.

About the suggestion mentioned by @ciandonovan, it seems like it will be only possible if you use SDO to drive the motor as you can specifically indicate the `Index` and `Sub-Index` for each motor axle. But `ros2_canopen` uses PDO so it would not be possible to use these Index and Sub-Index. I don't know if my point was correct but that's how I understand it.
With that, the only way to support multiple axes and drive the motor using PDO is to set the COB-ID for each axis, which I think will exhaust the COB-IDs (and if the motor driver product supports this feature).
Would you mind giving your insights about this? (I might have missed the important point in this discussion or I might have misunderstood something, so please enlighten me).

Thank you~

nishalangovender, 2025-03-10:
Hi everyone.

I just wanted to follow up on the state of this feature? I'm sure it would be really helpful to many developers.

Thanks!