ros-industrial/ros2_canopen #82: Master branch not building in ROS2 Humble

jclinton830, 2022-11-22:
**Describe the bug**
I cloned the master branch of this repo. Executed rodep install and tried to colcon build and get the following error. 

I thought this branch builds fine in Humble. Is this not correct? Am I missing something?

See below for the log

**Logs**

```
jerome@jjustin:~/test_ws$ colcon build
Starting >>> canopen_interfaces
Starting >>> lely_core_libraries
Finished <<< canopen_interfaces [7.96s]                                      
Starting >>> canopen_utils
--- stderr: canopen_utils                                                    
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< canopen_utils [0.90s]
--- stderr: lely_core_libraries                                
Cloning into 'upstr_lely_core_libraries'...
HEAD is now at ac31cd5c release v2.3.1
configure.ac:17: warning: The macro `AC_PROG_CC_STDC' is obsolete.
configure.ac:17: You should run autoupdate.
./lib/autoconf/c.m4:1666: AC_PROG_CC_STDC is expanded from...
configure.ac:17: the top level
configure.ac:81: warning: $as_echo is obsolete; use AS_ECHO(["message"]) instead
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/ax_pthread.m4:88: AX_PTHREAD is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
configure.ac:81: the top level
configure.ac:434: warning: AC_CHECK_HEADERS($ixxat_header): you should use literals
./lib/autoconf/headers.m4:217: AC_CHECK_HEADERS is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
configure.ac:434: the top level
configure.ac:17: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'
configure.ac:606: installing './tap-driver.sh'
src/can/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
libtool: warning: relinking 'liblely-tap.la'
libtool: warning: relinking 'liblely-util.la'
libtool: warning: relinking 'liblely-can.la'
libtool: warning: relinking 'liblely-co.la'
libtool: warning: relinking 'liblely-io.la'
libtool: warning: relinking 'liblely-ev.la'
libtool: warning: relinking 'liblely-io2.la'
libtool: warning: relinking 'liblely-coapp.la'
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< lely_core_libraries [26.2s]
Starting >>> canopen_core
Starting >>> canopen_fake_slaves
--- stderr: canopen_fake_slaves                                        
In file included from /home/jerome/test_ws/src/ros2_canopen/canopen_fake_slaves/include/canopen_fake_slaves/basic_slave.hpp:14,
                 from /home/jerome/test_ws/src/ros2_canopen/canopen_fake_slaves/src/basic_slave.cpp:2:
/home/jerome/test_ws/src/ros2_canopen/canopen_fake_slaves/include/canopen_fake_slaves/base_slave.hpp: In member function ‘virtual rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn ros2_canopen::BaseSlave::on_shutdown(const rclcpp_lifecycle::State&)’:
[... 55 lines cut ...]
   17 |     delete(interface);
      |     ^~~~~~~~~~~~~~~~~
---
Finished <<< canopen_master_driver [18.3s]
Finished <<< canopen_proxy_driver [19.3s]                             
Starting >>> canopen_402_driver
Finished <<< canopen_402_driver [20.3s]                             
Starting >>> canopen_tests
Starting >>> canopen
Starting >>> canopen_ros2_controllers
Finished <<< canopen [0.56s]                                             
Finished <<< canopen_tests [2.27s]                                          
Starting >>> canopen_ros2_control
Finished <<< canopen_ros2_control [10.7s]    
Finished <<< canopen_ros2_controllers [24.8s]                             

Summary: 13 packages finished [2min 7s]
  6 packages had stderr output: canopen_base_driver canopen_core canopen_fake_slaves canopen_master_driver canopen_utils lely_core_libraries

```

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: Master

ipa-cmh, 2022-11-28:
Looks like it builds just not cleanly. There are still some warnings which we need to remove.
You do not need to care about these for now.

jclinton830, 2022-11-29:
Apologies. Yes, you are right it does build. My custom software packages are failing for some other reason. I posted the build log of just the ros2_canopen package for readability. Will close this now.

closed 2022-11-29 by jclinton830