ros-industrial/ros2_canopen #85: Interpolated position mode

gseqBE, 2023-01-11:
This is more of a question if the package supports it.
One of our applications has multiple CANopen motion drives connected over the CANBus.

Would we be able to run the drives in Interpolated position mode? Basically send PVT messages at set frequency, with one of the drives programmed to be the SYNC generator.

Thank you

ipa-cmh, 2023-01-12:
> 1. Would we be able to run the drives in Interpolated position mode? Basically send PVT messages at set frequency.
>    The table below from the [documentation](https://ros-industrial.github.io/ros2_canopen/manual/user-guide/cia402-driver.html) does not seem to list this mode.

Right we currently do not expose that mode, but it should be possible to expose it.

> 2. Can one of the drives be programmed as the SYNC generator.
>     This would be an eds or a config via an SDO.

I have not tried this myself. SYNC is handled on lely canopen level. You could preset on the slave that should generate, there is no option in the config for doing this. You would need to refrain from adding the sync_period etc. in the master's config section.

> 3\. Does the service interface allow us to send data at a (high) frequency. We are looking at around 100Hz PVT data for 5-6 drives.

I would not recommend the service interface for this, as each pvt-point send requires a handshake. This could work but would probably require some testing and maybe tuning of qos and linux priorities. I'd rather recommend using the ros2_control interface for this.

ipa-cmh, 2023-01-20:
@gseqBE Checked a bit deeper.

It seems there are 3 submodes for interpolated position mode: linear, PV and PVT interpolation.
> 1. Linear interpolation as described in the CiA 402 standard (when object 208Eh bit8=1); This mode is almost
> identical with Cyclic Synchronous Position mode, only that it receives its position data into 60C1h sub-index 01
> instead of object 607Ah. No interpolation point buffer will be used.
> 2. PT (Position – Time) linear interpolation (legacy) (when object 208Eh bit8=0)
> 3. PVT (Position – Velocity – Time) cubic interpolation (legacy) (when object 208Eh bit8=0). 

For us it would currently be simple to support mode 1 mentioned here, as the current 402 driver does not support writing 3 values as a target.

gseqBE, 2023-01-20:
I see. Thanks for the update.

For our application, we do need 3, which requires buffer support. We will revisit the stack when 3 is implemented.
For now, we will use lely-core directly.

 Here is a reference from Copley drives, for PVT which we are trying to use(section 12.1.4):
[image]

Thank you. This may close if you are ok, as it got marked as an enhancement.

ipa-cmh, 2023-01-20:
@gseqBE ok.
#90 adds the support for the linear mode, not cubic interpolation.