ros-industrial/ros2_canopen #93: CAN frame: Operation canceled

jclinton830, 2023-03-01:
**Describe the bug**

After the node has been configured basic_slave node is reporting ` warning: error writing CAN frame: Operation canceled`

Do you have any idea what would be causing this issue? I have submitted the log after launching [proxy_lifecycle_setup.launch.py](http://proxy_lifecycle_setup.launch.py/)  and also the bus.yaml file

**Logs**

```
groundhog@groundhogPC:~/gh_ws$ ros2 launch groundhog_canopen [proxy_lifecycle_setup.launch.py](http://proxy_lifecycle_setup.launch.py/)
[INFO] [launch]: All log files can be found below /home/groundhog/.ros/log/2023-03-01-15-05-48-779937-groundhogPC-121543
[INFO] [launch]: Default logging verbosity is set to INFO
/home/groundhog/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/proxy_write_sdo/master.dcf
[INFO] [launch.user]: /home/groundhog/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/bus.yml
[INFO] [launch.user]: /home/groundhog/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.dcf
[INFO] [launch.user]:
[INFO] [launch.user]: can0
[INFO] [basic_slave_node-1]: process started with pid [121555]
[INFO] [device_container_node-2]: process started with pid [121557]
[device_container_node-2] [INFO] [1677643551.581533659] [device_container_node]: Starting Device Container with:
[device_container_node-2] [INFO] [1677643551.583381255] [device_container_node]:     master_config /home/groundhog/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.dcf
[device_container_node-2] [INFO] [1677643551.584828511] [device_container_node]:     bus_config /home/groundhog/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/bus.yml
[device_container_node-2] [INFO] [1677643551.586292279] [device_container_node]:     can_interface_name can0
[device_container_node-2] [INFO] [1677643551.594472759] [device_container_node]: Loading Master Configuration.
[device_container_node-2] [INFO] [1677643551.597940061] [device_container_node]: Load Library: /home/groundhog/gh_ws/install/canopen_master_driver/lib/liblifecycle_master_driver.so
[device_container_node-2] [INFO] [1677643551.631148562] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-2] [INFO] [1677643551.632752080] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-2] [INFO] [1677643551.673329425] [master]: NodeCanopenBasicMaster
[device_container_node-2] [INFO] [1677643551.674027649] [device_container_node]: Load master component.
[device_container_node-2] [INFO] [1677643551.674263459] [device_container_node]: Added /master to executor
[device_container_node-2] [INFO] [1677643551.683782439] [device_container_node]: Loading Driver Configuration.
[device_container_node-2] [INFO] [1677643551.684021177] [device_container_node]: Found device hetronic_rc with driver ros2_canopen::LifecycleProxyDriver
[device_container_node-2] [INFO] [1677643551.685647591] [device_container_node]: Load Library: /home/groundhog/gh_ws/install/canopen_proxy_driver/lib/liblifecycle_proxy_driver.so
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[basic_slave_node-1] [INFO] [1677643551.692696231] [hetronic_can_node]: Reaching inactive state.
[device_container_node-2] [INFO] [1677643551.698330374] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleProxyDriver>
[device_container_node-2] [INFO] [1677643551.698468120] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleProxyDriver>
[basic_slave_node-1] [INFO] [1677643551.714879490] [hetronic_can_node]: Reaching active state.
[device_container_node-2] [INFO] [1677643551.734151497] [device_container_node]: Load driver component.
[device_container_node-2] [INFO] [1677643551.734437094] [device_container_node]: Added /hetronic_rc to executor
[basic_slave_node-1] NMT: entering reset application state
[basic_slave_node-1] NMT: entering reset communication state
[basic_slave_node-1] NMT: running as slave
[basic_slave_node-1] NMT: entering pre-operational state
[basic_slave_node-1] NMT: entering operational state
[basic_slave_node-1] [INFO] [1677643551.755971389] [hetronic_can_node]: Created slave for node_id 100.
[device_container_node-2] [INFO] [1677643551.780409953] [device_container_node]: Loading Manager Configuration.
[device_container_node-2] [INFO] [1677643551.821794468] [device_container_node]: Added /lifecycle_manager to executor
[basic_slave_node-1] warning: error writing CAN frame: Operation canceled
```

Here is the bus.yaml

```
master:
  node_id: 1
  driver: ros2_canopen::LifecycleMasterDriver
  package: canopen_master_driver

hetronic_rc:
  node_id: 100
  dcf: "RX_ES-CAN-HL_line0.eds"
  dcf_path: "install/groundhog_canopen/share/groundhog_canopen/config/hetronic_rc"
  driver: "ros2_canopen::LifecycleProxyDriver"
  package: "canopen_proxy_driver"
  tpdo:
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0xF
      event_timer: 50
      mapping:
        - {index: 0x5000, sub_index: 0}
        - {index: 0x5000, sub_index: 1}
        - {index: 0x5000, sub_index: 2}
        - {index: 0x5000, sub_index: 3}
        - {index: 0x5000, sub_index: 4}
        - {index: 0x5000, sub_index: 5}
        - {index: 0x5000, sub_index: 6}
        - {index: 0x5000, sub_index: 7}
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0xF
      event_timer: 50
      mapping:
        - {index: 0x5001, sub_index: 0}
        - {index: 0x5001, sub_index: 1}
        - {index: 0x5001, sub_index: 2}
        - {index: 0x5001, sub_index: 3}
        - {index: 0x5001, sub_index: 4}
        - {index: 0x5001, sub_index: 5}
        - {index: 0x5001, sub_index: 6}
        - {index: 0x5001, sub_index: 7}
    3:
      enabled: false
    4:
      enabled: false
```

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble

jclinton830, 2023-03-20:
@ipa-cmh sorry to tag you here, but some help here would be greatly appreciated!

ipa-cmh, 2023-03-21:
@jclinton830 
Sorry for the late reply, I have been a bit under water the last weeks.
This seems to be an issue with the fake slave only (probably caused by some issue with the dcf configuration passed to it).
As you are launching the lifecycle version, the only thing that is currently active and doing something on the bus is the fake slave.
I do not have access to the launch files you use, so can't say much about where the problem is - but probably it is the eds bassed to the slave. 

Could you try running:
```bash
ros2 launch basic_slave.launch.py  node_id:=<YourID> node_name:=slave_node_1 slave_config:"<path/to/the/eds>"
ros2 lifecycle set /slave_node_1 configure
ros2 lifecycle set /slave_node_1 activate
```

Does the same error occur?

ipa-vsp, 2023-03-21:
There is also another hint, could it be that the PDO you are trying to configure are disabled in your eds? Then you need to enable them in the EDS. For some reason, lely libs does not pick that up.

jclinton830, 2023-03-22:
> There is also another hint, could it be that the PDO you are trying to configure are disabled in your eds? Then you need to enable them in the EDS. For some reason, lely libs does not pick that up.

The PDO details from the EDS looks like this for the COB IDs I am trying to set PDOs for. 

```
[5000]
SubNumber=9
ParameterName=TxPDO1
ObjectType=0x08

[5000sub0]
ParameterName=Number of Entries
ObjectType=0x07
DataType=0x0005
AccessType=ro
DefaultValue=0x08
PDOMapping=0

[5000sub1]
ParameterName=TxPDO1_1
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub2]
ParameterName=TxPDO1_2
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub3]
ParameterName=TxPDO1_3
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub4]
ParameterName=TxPDO1_4
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub5]
ParameterName=TxPDO1_5
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub6]
ParameterName=TxPDO1_6
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub7]
ParameterName=TxPDO1_7
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5000sub8]
ParameterName=TxPDO1_8
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001]
SubNumber=9
ParameterName=TxPDO2
ObjectType=0x08

[5001sub0]
ParameterName=Number of Entries
ObjectType=0x07
DataType=0x0005
AccessType=ro
DefaultValue=0x08
PDOMapping=0

[5001sub1]
ParameterName=TxPDO2_1
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub2]
ParameterName=TxPDO2_2
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub3]
ParameterName=TxPDO2_3
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub4]
ParameterName=TxPDO2_4
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub5]
ParameterName=TxPDO2_5
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub6]
ParameterName=TxPDO2_6
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub7]
ParameterName=TxPDO2_7
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

[5001sub8]
ParameterName=TxPDO2_8
ObjectType=0x07
DataType=0x0005
AccessType=rwr
DefaultValue=0x0
PDOMapping=1

```

jclinton830, 2023-03-22:
@ipa-cmh Like you suggested I ran this command 

`ros2 launch groundhog_canopen basic_slave.launch.py  node_id:=100 node_name:=hetronic_can_node slave_config:="src/groundhog/groundhog_canopen/config/bus_two/RX_ES-CAN-HL_line0.eds"`

And this is what I got 

```
groundhog@groundhogPC:~/gh_ws$ ros2 launch groundhog_canopen basic_slave.launch.py  node_id:=100 node_name:=hetronic_can_node slave_config:="src/groundhog/groundhog_canopen/config/bus_two/RX_ES-CAN-HL_line0.eds"
[INFO] [launch]: All log files can be found below /home/groundhog/.ros/log/2023-03-22-12-02-33-209878-groundhogPC-7116
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [basic_slave_node-1]: process started with pid [7128]
[basic_slave_node-1] [INFO] [1679446955.394721635] [hetronic_can_node]: Reaching inactive state.
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[basic_slave_node-1] [INFO] [1679446955.409459902] [hetronic_can_node]: Reaching active state.
[basic_slave_node-1] NMT: entering reset application state
[basic_slave_node-1] NMT: entering reset communication state
[basic_slave_node-1] NMT: running as slave
[basic_slave_node-1] NMT: entering pre-operational state
[basic_slave_node-1] NMT: entering operational state
[basic_slave_node-1] [INFO] [1679446955.435622866] [hetronic_can_node]: Created slave for node_id 100.
[basic_slave_node-1] warning: error writing CAN frame: Operation canceled

```

Then I ran this 

`ros2 lifecycle set /hetronic_can_node configure` and I got this response

```
groundhog@groundhogPC:~/gh_ws$ ros2 lifecycle set /hetronic_can_node configure
Unknown transition requested, available ones are:
- deactivate [4]
- shutdown [7]
```

So yeah still the same issue. 

Here is the launch file I am using 

```
import os
from ament_index_python import get_package_share_directory
from launch import LaunchDescription
import launch
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    slave_eds_path = os.path.join(
        get_package_share_directory("groundhog_canopen"),
        "config",
        "bus_two",
        "RX_ES-CAN-HL_line0.eds",
    )
    hetronic_can_node = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(
                    get_package_share_directory("groundhog_canopen"), "launch"
                ),
                "/basic_slave.launch.py",
            ]
        ),
        launch_arguments={
            "node_id": "100",
            "node_name": "hetronic_can_node",
            "slave_config": slave_eds_path,
        }.items(),
    )

    print(
        os.path.join(
            get_package_share_directory("groundhog_canopen"),
            "config",
            "proxy_write_sdo",
            "master.dcf",
        )
    )

    device_container = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(get_package_share_directory("canopen_core"), "launch"),
                "/canopen.launch.py",
            ]
        ),
        launch_arguments={
            "master_config": os.path.join(
                get_package_share_directory("groundhog_canopen"),
                "config",
                "bus_two",
                "master.dcf",
            ),
            "master_bin": "",
            "bus_config": os.path.join(
                get_package_share_directory("groundhog_canopen"),
                "config",
                "bus_two",
                "bus.yml",
            ),
            "can_interface_name": "can0",
        }.items(),
    )

    return LaunchDescription([hetronic_can_node, device_container])

```

ipa-cmh, 2023-03-22:
You are running the fake_slave on can0, so a real network - I have never tested that (try with a virtual can).
Otherwise, if you do have the real hardware connected, you do not need to launch the fake slave.
The fake slave is only for testing purposes.

jclinton830, 2023-04-11:
> You are running the fake_slave on can0, so a real network - I have never tested that (try with a virtual can). Otherwise, if you do have the real hardware connected, you do not need to launch the fake slave. The fake slave is only for testing purposes.

I have tried to run it with vcan0 and here is the output I get when running my launch file 

```
jerome@jjustin:~/gh_ws$ ros2 launch groundhog_canopen proxy_lifecycle_setup.launch.py 
[INFO] [launch]: All log files can be found below /home/jerome/.ros/log/2023-04-11-11-50-17-821117-jjustin-99998
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/bus.yml
[INFO] [launch.user]: /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.dcf
[INFO] [launch.user]: 
[INFO] [launch.user]: vcan0
[INFO] [basic_slave_node-1]: process started with pid [100010]
[INFO] [device_container_node-2]: process started with pid [100012]
[device_container_node-2] [INFO] [1681177818.392130521] [device_container_node]: Starting Device Container with:
[device_container_node-2] [INFO] [1681177818.392325661] [device_container_node]:         master_config /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.dcf
[device_container_node-2] [INFO] [1681177818.392346294] [device_container_node]:         bus_config /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/bus.yml
[device_container_node-2] [INFO] [1681177818.392367490] [device_container_node]:         can_interface_name vcan0
[device_container_node-2] [INFO] [1681177818.393228312] [device_container_node]: Loading Master Configuration.
[device_container_node-2] [INFO] [1681177818.394218781] [device_container_node]: Load Library: /home/jerome/gh_ws/install/canopen_master_driver/lib/liblifecycle_master_driver.so
[device_container_node-2] [INFO] [1681177818.398645157] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-2] [INFO] [1681177818.398710834] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-2] [INFO] [1681177818.406989622] [master]: NodeCanopenBasicMaster
[device_container_node-2] [INFO] [1681177818.407304624] [device_container_node]: Load master component.
[device_container_node-2] [INFO] [1681177818.407442639] [device_container_node]: Added /master to executor
[device_container_node-2] [INFO] [1681177818.410391861] [device_container_node]: Loading Driver Configuration.
[device_container_node-2] [INFO] [1681177818.410587976] [device_container_node]: Found device hetronic_rc with driver ros2_canopen::LifecycleProxyDriver
[device_container_node-2] [INFO] [1681177818.411228148] [device_container_node]: Load Library: /home/jerome/gh_ws/install/canopen_proxy_driver/lib/liblifecycle_proxy_driver.so
[device_container_node-2] [INFO] [1681177818.415641149] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleProxyDriver>
[device_container_node-2] [INFO] [1681177818.415673189] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleProxyDriver>
[device_container_node-2] [INFO] [1681177818.426932838] [device_container_node]: Load driver component.
[device_container_node-2] [INFO] [1681177818.427314559] [device_container_node]: Added /hetronic_rc to executor
[device_container_node-2] [INFO] [1681177818.455246557] [device_container_node]: Loading Manager Configuration.
[device_container_node-2] [INFO] [1681177818.470507531] [device_container_node]: Added /lifecycle_manager to executor
[basic_slave_node-1] [INFO] [1681177818.615125055] [hetronic_can_node]: Reaching inactive state.
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[basic_slave_node-1] [INFO] [1681177818.624466692] [hetronic_can_node]: Reaching active state.
[basic_slave_node-1] terminate called after throwing an instance of 'std::system_error'
[basic_slave_node-1]   what():  CanController: No such device
[ERROR] [basic_slave_node-1]: process has died [pid 100010, exit code -6, cmd '/home/jerome/gh_ws/install/canopen_fake_slaves/lib/canopen_fake_slaves/basic_slave_node --ros-args -r __node:=hetronic_can_node -r __ns:=/ --params-file /tmp/launch_params_j6ng2cwf'].
```

ipa-cmh, 2023-04-11:
The can controller was not activated. Please run

```
sudo modprobe vcan
sudo ip link add dev vcan0 type vcan
sudo ip link set vcan0 txqueuelen 1000
sudo ip link set up vcan0
```

To activate and configure the virtual can network.

jclinton830, 2023-04-11:
> The can controller was not activated. Please run
> 
> ```
> sudo modprobe vcan
> sudo ip link add dev vcan0 type vcan
> sudo ip link set vcan0 txqueuelen 1000
> sudo ip link set up vcan0
> ```
> 
> To activate and configure the virtual can network.

Yes I have already done that before running the launch file

ipa-cmh, 2023-04-11:
In that case, vcan0 might not be passed to the slave device?
The error message indicates that the hectronic_can slave does not find the canController, which is the socketcan interface.
This is either because a wrong name was passed in or the interface does not exist.

closed 2023-04-11 by jclinton830

ipa-cmh, 2023-04-11:
No worries!