ros-industrial/ros2_canopen #95: There are a few lines executed in the `update`-loop that are rather part of the activation procedure

destogl, 2023-04-14:
Resetting and starting of nodes is rather a part of the activation process, so it shouldn't be done/checked all the time in the update loop. Changing this would lead to clearer code and behavior, but it has some strong drawbacks:

1. Needs error management inside HW interface. ros2_control supports that (see [this PR](https://github.com/ros-controls/ros2_control/pull/805)), but then also all the controllers get deactivated – and they have to manually activated again.
2. The first point clearly disturbs the prototyping and makes it much harder is someone doesn't understand all the internal mechanisms from ros2_control.
3. There is currently not possibility to automatically re-activate hardware and controllers which probably is desired scenario is many applications to handle bus errors better, e.g., bus errors caused by use of non-realtime kernel.

https://github.com/ros-industrial/ros2_canopen/blob/1ac8251125cfd3869ede1d5d2321158e593e98cf/canopen_ros2_control/src/canopen_system.cpp#L270-L280

P.S. I will also extend this in the future with possible solutions when different options are tested in real-world scenarios.

ipa-cmh, 2023-06-05:
@destogl any updates here or is this closable?

destogl, 2023-06-14:
I didn't do enough tests yet. So I would keep this open, but it will take time to get it fixed / closed.