ros-industrial/ros2_canopen #99: Stack smashing detected

james-ward, 2023-04-26:
**Describe the bug**
Similar to #86 - the master crashes when trying to bring up a slave node with the error:
`[device_container_node-1] *** stack smashing detected ***: terminated`

**To Reproduce**
* Add `heartbeat_producer: 500` to either of the nodes in `canopen_tests/config/simple/bus.yml`
* Compile
* `ros2 launch canopen_tests proxy_setup.launch.py`

**Expected behavior**
All nodes brought up and producing heartbeats.

**Logs**
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-04-26-10-41-28-417737-feather-14143
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/underlay_ws/install/canopen_tests/share/canopen_tests/config/proxy_write_sdo/master.dcf
[INFO] [launch.user]: /opt/underlay_ws/install/canopen_tests/share/canopen_tests/config/simple/bus.yml
[INFO] [launch.user]: /opt/underlay_ws/install/canopen_tests/share/canopen_tests/config/simple/master.dcf
[INFO] [launch.user]: 
[INFO] [launch.user]: vcan0
[INFO] [basic_slave_node-1]: process started with pid [14160]
[INFO] [basic_slave_node-2]: process started with pid [14162]
[INFO] [device_container_node-3]: process started with pid [14164]
[device_container_node-3] [INFO] [1682505689.137535473] [device_container_node]: Starting Device Container with:
[device_container_node-3] [INFO] [1682505689.137878680] [device_container_node]: 	 master_config /opt/underlay_ws/install/canopen_tests/share/canopen_tests/config/simple/master.dcf
[device_container_node-3] [INFO] [1682505689.137910592] [device_container_node]: 	 bus_config /opt/underlay_ws/install/canopen_tests/share/canopen_tests/config/simple/bus.yml
[device_container_node-3] [INFO] [1682505689.137920857] [device_container_node]: 	 can_interface_name vcan0
[device_container_node-3] [INFO] [1682505689.138356329] [device_container_node]: Loading Master Configuration.
[device_container_node-3] [INFO] [1682505689.138907891] [device_container_node]: Load Library: /opt/underlay_ws/install/canopen_master_driver/lib/libmaster_driver.so
[device_container_node-3] [INFO] [1682505689.142654231] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-3] [INFO] [1682505689.142698212] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-3] [INFO] [1682505689.150617656] [master]: NodeCanopenBasicMaster
[device_container_node-3] [INFO] [1682505689.150663132] [device_container_node]: Load master component.
[device_container_node-3] [INFO] [1682505689.150692411] [device_container_node]: Added /master to executor
[device_container_node-3] NMT: entering reset application state
[device_container_node-3] NMT: entering reset communication state
[device_container_node-3] NMT: running as master
[device_container_node-3] NMT: entering pre-operational state
[device_container_node-3] NMT: entering operational state
[device_container_node-3] [INFO] [1682505689.188611419] [device_container_node]: Loading Driver Configuration.
[device_container_node-3] [INFO] [1682505689.188690707] [device_container_node]: Found device proxy_device_1 with driver ros2_canopen::ProxyDriver
[device_container_node-3] [INFO] [1682505689.189064212] [device_container_node]: Load Library: /opt/underlay_ws/install/canopen_proxy_driver/lib/libproxy_driver.so
[device_container_node-3] [INFO] [1682505689.195532526] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[device_container_node-3] [INFO] [1682505689.195581692] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[device_container_node-3] [INFO] [1682505689.205458349] [device_container_node]: Load driver component.
[device_container_node-3] [INFO] [1682505689.205518158] [device_container_node]: Added /proxy_device_1 to executor
[device_container_node-3] [WARN] [1682505689.214935944] [proxy_device_1]: Wait for device to boot.
[basic_slave_node-1] [INFO] [1682505689.299788677] [slave_node_1]: Reaching inactive state.
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[basic_slave_node-2] [INFO] [1682505689.301485169] [slave_node_2]: Reaching inactive state.
[basic_slave_node-1] [INFO] [1682505689.311529428] [slave_node_1]: Reaching active state.
[basic_slave_node-2] [INFO] [1682505689.311751743] [slave_node_2]: Reaching active state.
[basic_slave_node-1] NMT: entering reset application state
[basic_slave_node-1] NMT: entering reset communication state
[basic_slave_node-2] NMT: entering reset application state
[basic_slave_node-2] NMT: entering reset communication state
[basic_slave_node-1] NMT: running as slave
[basic_slave_node-1] NMT: entering pre-operational state
[basic_slave_node-1] NMT: entering operational state
[basic_slave_node-1] [INFO] [1682505689.330502406] [slave_node_1]: Created slave for node_id 2.
[basic_slave_node-2] NMT: running as slave
[basic_slave_node-2] NMT: entering pre-operational state
[basic_slave_node-2] NMT: entering operational state
[basic_slave_node-2] [INFO] [1682505689.330392091] [slave_node_2]: Created slave for node_id 3.
[device_container_node-3] *** stack smashing detected ***: terminated
[ERROR] [device_container_node-3]: process has died [pid 14164, exit code -6, cmd '/opt/underlay_ws/install/canopen_core/lib/canopen_core/device_container_node --ros-args -r __node:=device_container_node -r __ns:=/ --params-file /tmp/launch_params_yzyx9hqq --params-file /tmp/launch_params_9w2ul1ja --params-file /tmp/launch_params_5gzf2tkq --params-file /tmp/launch_params_cr_9hrus'].
```

**Setup:**
 - Device:
 - OS: Ubuntu 22.04
 - ROS-Distro: rolling
 - Branch/Commit: master

**Additional context**
N/A

ipa-cmh, 2023-04-27:
Can you clean ws and rebuild completely to see if the issue still persists?

james-ward, 2023-04-28:
Yes, this is with a complete rebuild - everything built from scratch each time.

Interestingly, if you don't turn on `heartbeat_producer` it works fine.
And making the master send heartbeats is also okay. From what I can see it seems like there is something going on when the SDO is sent to configure the slave.

ipa-cmh, 2023-04-28:
@james-ward
Just tried to reproduce with the follwoing setup:
* Latest Humble Docker in privilidged mode
* commit ecd79d0 on ros2_canopen/master

**bus.yml**
```
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 1
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"

proxy_device_1:
  node_id: 2
  dcf: "simple.eds"
  driver: "ros2_canopen::ProxyDriver"
  package: "canopen_proxy_driver"
  #heartbeat_producer: 500

proxy_device_2:
  node_id: 3
  dcf: "simple.eds"
  driver: "ros2_canopen::ProxyDriver"
  package: "canopen_proxy_driver"
  heartbeat_producer: 500
```

For me it does what it should see log below. Is aboves bus.yml the config you use?

```
[INFO] [launch]: All log files can be found below /home/christoph/.ros/log/2023-04-28-08-05-02-285348-christoph-ThinkPad-T490-54651
[INFO] [launch]: Default logging verbosity is set to INFO
/home/ws/install/canopen_tests/share/canopen_tests/config/proxy_write_sdo/master.dcf
[INFO] [launch.user]: /home/ws/install/canopen_tests/share/canopen_tests/config/simple/bus.yml
[INFO] [launch.user]: /home/ws/install/canopen_tests/share/canopen_tests/config/simple/master.dcf
[INFO] [launch.user]: 
[INFO] [launch.user]: vcan0
[INFO] [basic_slave_node-1]: process started with pid [54665]
[INFO] [basic_slave_node-2]: process started with pid [54667]
[INFO] [device_container_node-3]: process started with pid [54669]
[device_container_node-3] [INFO] [1682669102.473699680] [device_container_node]: Starting Device Container with:
[device_container_node-3] [INFO] [1682669102.473805439] [device_container_node]:         master_config /home/ws/install/canopen_tests/share/canopen_tests/config/simple/master.dcf
[device_container_node-3] [INFO] [1682669102.473817178] [device_container_node]:         bus_config /home/ws/install/canopen_tests/share/canopen_tests/config/simple/bus.yml
[device_container_node-3] [INFO] [1682669102.473822290] [device_container_node]:         can_interface_name vcan0
[device_container_node-3] [INFO] [1682669102.474228641] [device_container_node]: Loading Master Configuration.
[device_container_node-3] [INFO] [1682669102.474715395] [device_container_node]: Load Library: /home/ws/install/canopen_master_driver/lib/libmaster_driver.so
[device_container_node-3] [INFO] [1682669102.477699648] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-3] [INFO] [1682669102.477741866] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-3] [INFO] [1682669102.481434514] [master]: NodeCanopenBasicMaster
[device_container_node-3] [INFO] [1682669102.481560640] [device_container_node]: Load master component.
[device_container_node-3] [INFO] [1682669102.481621131] [device_container_node]: Added /master to executor
[device_container_node-3] NMT: entering reset application state
[device_container_node-3] NMT: entering reset communication state
[device_container_node-3] NMT: running as master
[device_container_node-3] NMT: entering pre-operational state
[device_container_node-3] NMT: entering operational state
[device_container_node-3] [INFO] [1682669102.502875181] [device_container_node]: Loading Driver Configuration.
[device_container_node-3] [INFO] [1682669102.502954081] [device_container_node]: Found device proxy_device_1 with driver ros2_canopen::ProxyDriver
[device_container_node-3] [INFO] [1682669102.503101027] [device_container_node]: Load Library: /home/ws/install/canopen_proxy_driver/lib/libproxy_driver.so
[device_container_node-3] [INFO] [1682669102.506984518] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[device_container_node-3] [INFO] [1682669102.507025807] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[device_container_node-3] [INFO] [1682669102.510562053] [device_container_node]: Load driver component.
[device_container_node-3] [INFO] [1682669102.510652222] [device_container_node]: Added /proxy_device_1 to executor
[device_container_node-3] [WARN] [1682669102.519275856] [proxy_device_1]: Wait for device to boot.
[basic_slave_node-1] [INFO] [1682669102.694433343] [slave_node_1]: Reaching inactive state.
[basic_slave_node-2] [INFO] [1682669102.696255236] [slave_node_2]: Reaching inactive state.
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[... 3 lines cut ...]
[basic_slave_node-2] NMT: entering reset application state
[basic_slave_node-2] NMT: entering reset communication state
[basic_slave_node-2] NMT: running as slave
[basic_slave_node-1] NMT: entering reset application state
[basic_slave_node-1] NMT: entering reset communication state
[basic_slave_node-1] NMT: running as slave
[basic_slave_node-1] NMT: entering pre-operational state
[basic_slave_node-1] NMT: entering operational state
[basic_slave_node-2] NMT: entering pre-operational state
[basic_slave_node-2] NMT: entering operational state
[basic_slave_node-2] [INFO] [1682669102.735530582] [slave_node_2]: Created slave for node_id 3.
[basic_slave_node-1] [INFO] [1682669102.735853689] [slave_node_1]: Created slave for node_id 2.
[device_container_node-3] [INFO] [1682669102.738679735] [proxy_device_1]: Driver booted and ready.
[... cut ...]
[device_container_node-3] [INFO] [1682669102.750953999] [device_container_node]: Added /proxy_device_2 to executor
[device_container_node-3] [WARN] [1682669102.761358024] [proxy_device_2]: Wait for device to boot.
[device_container_node-3] [INFO] [1682669103.237673182] [proxy_device_2]: Driver booted and ready.
[device_container_node-3] [INFO] [1682669103.237804010] [proxy_device_2]: Starting RPDO Listener
[device_container_node-3] [INFO] [1682669103.237842440] [proxy_device_2]: Starting EMCY Listener
```

james-ward, 2023-05-01:
Okay, I have confirmed what is causing the problem. I can reproduce it reliably now.
It is not related to the heartbeat or configuring via SDO.
I was able to run without crashing on the native OS and using Docker (humble and rolling).

The crash occurs when you build in release mode using the colcon mixin.

Steps to reproduce:
1. Enable standard mixins: `colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml`
2. `colcon mixin update`
3. `colcon build --mixin release`
4. `source install/setup.bash`
5. `ros2 launch canopen_tests proxy_setup.launch.py`

ipa-cmh, 2023-05-03:
Thanks, great job at making it reproducable. It will now as well appear for me.

Ok, I now narrowed it down, I believe.
When running with -fno-stack-protector, everything seems to work fine.

ipa-cmh, 2023-05-03:
For me #103 seems to fix this issue.
Can you confirm?

james-ward, 2023-05-04:
> For me #103 seems to fix this issue. Can you confirm?

Yes, I agree. #103 fixes it. Thanks!!

closed 2023-05-04 by ipa-cmh