ros-industrial/ros2_canopen #105: Segmentation Fault after driver activated

jclinton830, 2023-05-05:
**Describe the bug**
Hi, 

I have a custom driver that I have written inheriting the proxy driver. When I launch the my canopen configuration everything seems to work fine. Then when I configure and then activate it, it say that the driver booted. However, after that the software crashes. See below log.

I am not sure if I am missing something or if this is a bug. Any help would be appreciated!

**Logs**
```
jerome@jjustin:~/gh_ws$ ros2 launch groundhog_canopen hetronic_lifecycle_setup.launch.py 
[INFO] [launch]: All log files can be found below /home/jerome/.ros/log/2023-05-04-16-18-52-337749-jjustin-141855
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/bus.yml
[INFO] [launch.user]: /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.dcf
[INFO] [launch.user]: /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [141867]
[device_container_node-1] [INFO] [1683181132.461812327] [device_container_node]: Starting Device Container with:
[device_container_node-1] [INFO] [1683181132.461968434] [device_container_node]:         master_config /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/master.dcf
[device_container_node-1] [INFO] [1683181132.461980390] [device_container_node]:         bus_config /home/jerome/gh_ws/install/groundhog_canopen/share/groundhog_canopen/config/bus_two/bus.yml
[device_container_node-1] [INFO] [1683181132.461988993] [device_container_node]:         can_interface_name can0
[device_container_node-1] [INFO] [1683181132.462514468] [device_container_node]: Loading Master Configuration.
[device_container_node-1] [INFO] [1683181132.463186408] [device_container_node]: Load Library: /home/jerome/gh_ws/install/canopen_master_driver/lib/liblifecycle_master_driver.so
[device_container_node-1] [INFO] [1683181132.465420160] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-1] [INFO] [1683181132.465477073] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-1] [INFO] [1683181132.470055615] [master]: NodeCanopenBasicMaster
[device_container_node-1] [INFO] [1683181132.470286658] [device_container_node]: Load master component.
[device_container_node-1] [INFO] [1683181132.470374402] [device_container_node]: Added /master to executor
[device_container_node-1] [INFO] [1683181132.472936289] [device_container_node]: Loading Driver Configuration.
[device_container_node-1] [INFO] [1683181132.473021459] [device_container_node]: Found device hetronic_rc with driver ros2_hetronic::LifecycleHetronicDriver
[device_container_node-1] [INFO] [1683181132.473220670] [device_container_node]: Load Library: /home/jerome/gh_ws/install/ros2_hetronic_driver/lib/liblifecycle_hetronic_driver.so
[device_container_node-1] [INFO] [1683181132.477058718] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_hetronic::LifecycleHetronicDriver>
[device_container_node-1] [INFO] [1683181132.477105099] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_hetronic::LifecycleHetronicDriver>
[device_container_node-1] [INFO] [1683181132.480066419] [device_container_node]: Load driver component.
[device_container_node-1] [INFO] [1683181132.480157456] [device_container_node]: Added /hetronic_rc to executor
[device_container_node-1] [INFO] [1683181132.481234850] [device_container_node]: Loading Manager Configuration.
[device_container_node-1] [INFO] [1683181132.484891380] [device_container_node]: Added /lifecycle_manager to executor
[device_container_node-1] [INFO] [1683181136.179801169] [lifecycle_manager]: Configuring for 2 devices.
[device_container_node-1] [INFO] [1683181136.180001900] [lifecycle_manager]: Found hetronic_rc (node_id=100)
[device_container_node-1] [INFO] [1683181136.182523906] [lifecycle_manager]: Found master (node_id=1)
[device_container_node-1] NMT: entering reset application state
[device_container_node-1] NMT: entering reset communication state
[device_container_node-1] NMT: running as master
[device_container_node-1] NMT: entering pre-operational state
[device_container_node-1] NMT: entering operational state
[device_container_node-1] [INFO] [1683181151.074156495] [hetronic_rc]: Driver booted and ready.
[device_container_node-1] [INFO] [1683181151.074540619] [hetronic_rc]: Starting RPDO Listener
[ERROR] [device_container_node-1]: process has died [pid 141867, exit code -11, cmd '/home/jerome/gh_ws/install/canopen_core/lib/canopen_core/device_container_node --ros-args -r __node:=device_container_node -r __ns:=/ --params-file /tmp/launch_params_2tvu7_i7 --params-file /tmp/launch_params_vn7o9r5y --params-file /tmp/launch_params_ptlggfjs --params-file /tmp/launch_params_bp0tt4bq'].
Exception ignored in: <function Executor.__del__ at 0x7ffaf09d1e10>
Traceback (most recent call last):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 229, in __del__
    self._sigint_gc.destroy()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/signals.py", line 65, in destroy
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
```

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: master

**Additional context**
Here is the link to my [project](https://github.com/acfr/groundhog/tree/canopen/groundhog_canopen)

Here is the [link](https://github.com/acfr/ros2_hetronic/tree/1f831e9a6a485bd10bca2fb67b7786d6e20cfeb8) to the custom driver I am developing

ipa-cmh, 2023-05-08:
Can't say much without the code. It seems to be failing when starting RPDO listener or shortly after. You could add a few more logging points for the start of the different functions you implemented in your driver

jclinton830, 2023-05-10:
Ok so what happened is that my add_to_master() was empty and only had 

```
RCLCPP_INFO(this->node_->get_logger(), "Driver booted and ready.");
```

closed 2023-05-10 by jclinton830