ros-industrial/ros2_canopen #107: Segmentation fault (Address not mapped to object [0x398])

albert285957606, 2023-05-10:
**Describe the bug**
I uninstalled all the unnecessary components on the control launch, but it still gave me this error and I couldn't get rid of it.

**Logs**
[ros2_control_node-1] [INFO] [1683709909.075067756] [Cia402System]: Number of registered drivers: '1'
[ros2_control_node-1] [INFO] [1683709909.075209664] [Cia402System]: 
[ros2_control_node-1] Registered driver:
[ros2_control_node-1]     name: 'kinco_node_1'
[ros2_control_node-1]     node_id: '10'
[ros2_control_node-1] [INFO] [1683709909.075234619] [device_container]: Initialisation successful.
[ros2_control_node-1] [INFO] [1683709909.075468356] [resource_manager]: Successful 'configure' of hardware 'canopen_test_system'
[ros2_control_node-1] [INFO] [1683709909.075514248] [resource_manager]: 'activate' hardware 'canopen_test_system' 
[ros2_control_node-1] [INFO] [1683709909.075530046] [resource_manager]: Successful 'activate' of hardware 'canopen_test_system'
[ros2_control_node-1] [INFO] [1683709909.078441846] [kinco_node_1]: Starting EMCY Listener
[ros2_control_node-1] [INFO] [1683709909.086642043] [controller_manager]: update rate is 10 Hz
[ros2_control_node-1] [WARN] [1683709909.086912809] [controller_manager]: Could not enable FIFO RT scheduling policy
[ros2_control_node-1] [ERROR] [1683709909.087055454] [kinco_node_1]: ---------------------------------打印代码执行----------------------------------
[ros2_control_node-1] Stack trace (most recent call last) in thread 41744:
[ros2_control_node-1] [INFO] [1683709909.088014763] [kinco_node_1]: Starting RPDO Listener
[ros2_control_node-1] #11   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-1] #10   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f02eb3749ff]
[ros2_control_node-1] #9    Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f02eb2e2b42]
[ros2_control_node-1] #8    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f02eb5742b2, in 
[ros2_control_node-1] #7    Source "./src/ros2_control_node.cpp", line 80, in operator() [0x561a1b681810]
[ros2_control_node-1] #6    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f02eb17a401, in hardware_interface::ResourceManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-1] #5    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f02eb1a17c4, in hardware_interface::System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-1] #4    Object "/home/test/Desktop/controller/demo5/install/canopen_ros2_control/lib/libcanopen_ros2_control.so", at 0x7f02e42e2efe, in canopen_ros2_control::Cia402System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-1] #3    Object "/home/test/Desktop/controller/demo5/install/canopen_ros2_control/lib/libcanopen_ros2_control.so", at 0x7f02e42e4965, in ros2_canopen::Cia402Driver::get_position()
[ros2_control_node-1] #2    Object "/home/test/Desktop/controller/demo5/install/canopen_ros2_control/lib/libcanopen_ros2_control.so", at 0x7f02e42e4ca7, in std::__shared_ptr_access<ros2_canopen::node_interfaces::NodeCanopen402Driver<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2, false, false>::operator->() const
[ros2_control_node-1] #1    Object "/home/test/Desktop/controller/demo5/install/canopen_ros2_control/lib/libcanopen_ros2_control.so", at 0x7f02e42e5453, in std::__shared_ptr_access<ros2_canopen::node_interfaces::NodeCanopen402Driver<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2, false, false>::_M_get() const
[ros2_control_node-1] #0    Object "/home/test/Desktop/controller/demo5/install/canopen_ros2_control/lib/libcanopen_ros2_control.so", at 0x7f02e42e5d80, in std::__shared_ptr<ros2_canopen::node_interfaces::NodeCanopen402Driver<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2>::get() const
[ros2_control_node-1] Segmentation fault (Address not mapped to object [0x398])
[ERROR] [ros2_control_node-1]: process has died [pid 41684, exit code -11, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_rh1l5w25 --params-file /home/test/Desktop/controller/demo5/install/canopen_ros2_control/share/canopen_ros2_control/config/cia402_ros2_control.yaml'].
[spawner-3] [INFO] [1683709911.295526322] [spawner_cia402_device_1_controller]: Waiting for '/controller_manager' node to exist
[spawner-3] [INFO] [1683709913.315490602] [spawner_cia402_device_1_controller]: Waiting for '/controller_manager' node to exist

**Setup:**
 - Device: can0
 - OS: ubuntu22.04
 - ROS-Distro: master
 - Branch/Commit:

**Additional context**
Add any other context about the problem here.

ipa-cmh, 2023-05-10:
I haven't seen this one yet. Can you give a bit more context for reproducing?

albert285957606, 2023-05-11:
running command :ros2 launch canopen_ros2_control cia402_system.launch.py

cia402_ros2_control.yaml
`controller_manager:
  ros__parameters:
    update_rate: 10  # Hz
    cia402_device_1_controller:
      type: canopen_ros2_controllers/Cia402DeviceController
cia402_device_1_controller:
  ros__parameters:
    joint: node_1`

cia402_system.launch.py
`from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import OpaqueFunction
from launch.substitutions import Command, FindExecutable, LaunchConfiguration, PathJoinSubstitution
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def launch_setup(context, *args, **kwargs):

    name = LaunchConfiguration("name")
    prefix = LaunchConfiguration("prefix")

    # bus configuration
    bus_config_package = LaunchConfiguration("bus_config_package")
    bus_config_directory = LaunchConfiguration("bus_config_directory")
    bus_config_file = LaunchConfiguration("bus_config_file")
    # bus configuration file full path
    bus_config = PathJoinSubstitution(
        [FindPackageShare(bus_config_package), bus_config_directory, bus_config_file]
    )

    # master configuration
    master_config_package = LaunchConfiguration("master_config_package")
    master_config_directory = LaunchConfiguration("master_config_directory")
    master_config_file = LaunchConfiguration("master_config_file")
    # master configuration file full path
    master_config = PathJoinSubstitution(
        [FindPackageShare(master_config_package), master_config_directory, master_config_file]
    )

    # can interface name
    can_interface = LaunchConfiguration("can_interface")

    # robot description stuff
    description_package = LaunchConfiguration("description_package")
    description_file = LaunchConfiguration("description_file")
    robot_description_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution(
                [FindPackageShare(description_package), "urdf", description_file]
            ),
            " ",
            "name:=",
            name,
            " ",
            "prefix:=",
            prefix,
            " ",
            "bus_config:=",
            bus_config,
[... 147 lines cut ...]
    declared_arguments.append(
        DeclareLaunchArgument(
            "master_config_directory",
            default_value="config/cia402",
            description="Path to master configuration file (*.dcf)",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "master_config_file",
            default_value="master.dcf",
            description="Path to master configuration file (*.dcf)",
        )
    )

    declared_arguments.append(
        DeclareLaunchArgument(
            "can_interface",
            default_value="can0",
            description="Interface name for can",
        )
    )

    return LaunchDescription(declared_arguments + [OpaqueFunction(function=launch_setup)])
`

ipa-cmh, 2023-06-08:
@albert285957606 
Can you check if you can launch the following without problems:

```bash
ros2 launch canopen_tests robot_control.launch.py
```

This uses a different hardware interface.

closed 2023-06-09 by ipa-cmh

gsalinas, 2023-09-08:
Hi! I'm getting the same segfault when I run on my actual hardware, but `robot_control_setup.launch.py` works for me: @albert285957606 did you ever figure out what your issue was? I can't tell how the merged PR addresses whatever this problem is.

gsalinas, 2023-09-12:
For anyone else who runs into this in the future, my issue turned out to be that my URDF files were incorrect: the *.urdf.xacro file has some references to the bus and master configuration that need to be updated, and the *.ros2_control.xacro file needs the node IDs for each joint.