ros-industrial/ros2_canopen #114: Confusion with master driver

leonce-m, 2023-05-15:
I am trying to set up the maxon epos4 50/15 CAN controller with a Jetson AGX Orin running Foxy. (Humble won't be available until October with JetPack 6.)

Reading through the documentation, source code on the foxy branch and doxygen I am a little confused what driver I should use for the master.

according to read the docs:
`ros2_canopen::MasterDriver` or `ros2_canopen::LifecycleMasterDriver`
according to [doxygen](https://ros-industrial.github.io/ros2_canopen/api/classros2__canopen_1_1LifecycleCanopenMaster.html): 
`ros2_canopen::CanopenMaster` or `ros2_canopen::LifecycleCanopenMaster`
according to [foxy source](https://github.com/ros-industrial/ros2_canopen/blob/foxy/canopen_core/include/canopen_core/master_node.hpp):
`ros2_canopen::MasterNode` or `ros2_canopen::LifecycleMasterNode`

ipa-cmh, 2023-05-15:
Hi, that is mainly because the ros2 control part will only work in galactic / humble onwards. Consequently, the foxy branch currently has an old state, whereas the documentation is built of master.

ipa-cmh, 2023-06-08:
Seems to be stale.

closed 2023-06-08 by ipa-cmh