ros-industrial/ros2_canopen #118: Problem With canopen 402 driver

charanhs123, 2023-06-05:
I am trying to set up the maxon epos2 50/5 CAN controller with a Jetson AGX Orin running Humble. (Through Docker Image.)

The CANopen 402 driver is reporting a failed error while trying to build the package, this is when cloned from the Master branch.
While the Humble branch reportedly didn't report any error but not sure of using which driver as a Master.

Need help with this issue.

Thank you

ipa-cmh, 2023-06-05:
Can you try this commit: 9637faf108330f6262a2a23eba1af31c72217716 (it three merges back).

I am in the process of merging a bunch of PRs right now, so there could be an issue. Also can you provide the error message?

charanhs123, 2023-06-06:
1.The Error while using Humble branch is " Could not find the requested resource in ament index" , which is due to lack of master driver but when Cia 402 driver was used the initialisation wwas successful but master failed.
2. The error while using Master branch was with the Cpp file from node interface folder 

I observe there s a update in the Humble Branch , i will test it and update here regarding the progress.

Thank you

ipa-cmh, 2023-06-06:
1. Thanks for the issue

2. You can use the master branch for humble - we develop on humble.

3. The humble branch is only for release purposes. We are currently in process of preparing the first beta release, so no guarantees on the humble branch right now.

4. I can't solve, reproduce or help without knowing what you launched. In case the error persists please provide launch command and logs. In case it is a build probem please provide colcon command and build logs or at least the error log.

charanhs123, 2023-06-07:
Hii ,
i am trying to launch a file to control joints of the Exoskeleton, i have got this error msg after trying the humble branch.

robot@jetson-desktop:~/ros2_ws$ ros2 launch exomotor exo.launch.py
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2023-06-07-12-15-53-594919-jetson-desktop-1000
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/robot/ros2_ws/install/exomotor/share/exomotor/config/exocanbus/bus.yml
[INFO] [launch.user]: /home/robot/ros2_ws/install/exomotor/share/exomotor/config/exocanbus/master.dcf
[INFO] [launch.user]: /home/robot/ros2_ws/install/exomotor/share/exomotor/config/exocanbus/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [1011]
[device_container_node-1] [INFO] [1686140153.952840771] [device_container_node]: Starting Device Container with:
[device_container_node-1] [INFO] [1686140153.953069732] [device_container_node]: 	 master_config /home/robot/ros2_ws/install/exomotor/share/exomotor/config/exocanbus/master.dcf
[device_container_node-1] [INFO] [1686140153.953099588] [device_container_node]: 	 bus_config /home/robot/ros2_ws/install/exomotor/share/exomotor/config/exocanbus/bus.yml
[device_container_node-1] [INFO] [1686140153.953113892] [device_container_node]: 	 can_interface_name can0
[device_container_node-1] [INFO] [1686140153.954673583] [device_container_node]: Loading Master Configuration.
[device_container_node-1] [INFO] [1686140153.955825303] [device_container_node]: Load Library: /home/robot/ros2_ws/install/canopen_master_driver/lib/libmaster_driver.so
[device_container_node-1] [INFO] [1686140153.960929625] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1686140153.961013274] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1686140153.965182102] [master]: NodeCanopenBasicMaster
[device_container_node-1] [INFO] [1686140153.965475064] [device_container_node]: Load master component.
[device_container_node-1] [INFO] [1686140153.965624665] [device_container_node]: Added /master to executor
[device_container_node-1] NMT: entering reset application state
[device_container_node-1] NMT: entering reset communication state
[device_container_node-1] NMT: running as master
[device_container_node-1] NMT: entering pre-operational state
[device_container_node-1] NMT: entering operational state
[device_container_node-1] [INFO] [1686140154.001395497] [device_container_node]: Loading Driver Configuration.
[device_container_node-1] [INFO] [1686140154.001599978] [device_container_node]: Found device slave1 with driver ros2_canopen::Cia402Driver
[device_container_node-1] [ERROR] [1686140154.003221941] [device_container_node]: Loading driver failed: node_id(3), node_name(slave1), driver(ros2_canopen::Cia402Driver), driver_package(canopen_402_driver)
[device_container_node-1] terminate called after throwing an instance of 'ros2_canopen::DeviceContainerException'
[device_container_node-1]   what():  std::exception
[ERROR] [device_container_node-1]: process has died [pid 1011, exit code -6, cmd '/home/robot/ros2_ws/install/canopen_core/lib/canopen_core/device_container_node --ros-args -r __node:=device_container_node -r __ns:=/ --params-file /tmp/launch_params_cy85th5l --params-file /tmp/launch_params_21ztqtq0 --params-file /tmp/launch_params_jqrz3gj0 --params-file /tmp/launch_params_cucwtl6o'].
 
Thank you

charanhs123, 2023-06-07:
Hello the Canopen Error is repeating again unfortunately.

-- stderr: canopen_402_driver
In file included from /home/robot/dependencies_ws/src/ros2_canopen/canopen_402_driver/src/node_interfaces/node_canopen_402_driver.cpp:2:
/home/robot/dependencies_ws/src/ros2_canopen/canopen_402_driver/include/canopen_402_driver/node_interfaces/node_canopen_402_driver_impl.hpp: In member function 'void ros2_canopen::node_interfaces::NodeCanopen402Driver<NODETYPE>::init(bool) [with NODETYPE = rclcpp::Node]':
/home/robot/dependencies_ws/src/ros2_canopen/canopen_402_driver/include/canopen_402_driver/node_interfaces/node_canopen_402_driver_impl.hpp:32:71: error: reference to '_1' is ambiguous
   32 |     std::bind(&NodeCanopen402Driver<rclcpp::Node>::handle_init, this, _1, _2));
      |                                                                       ^~
In file included from /usr/include/c++/9/pstl/glue_algorithm_defs.h:13,
                 from /usr/include/c++/9/algorithm:71,
                 from /home/robot/dependencies_ws/src/ros2_canopen/canopen_402_driver/include/canopen_402_driver/motor.hpp:4,
                 from /home/robot/dependencies_ws/src/ros2_canopen/canopen_402_driver/include/canopen_402_driver/node_interfaces/node_canopen_402_driver.hpp:4,
                 from /home/robot/dependencies_ws/src/ros2_canopen/canopen_402_driver/src/node_interfaces/node_canopen_402_driver.cpp:1:
/usr/include/c++/9/functional:211:34: note: candidates are: 'const std::_Placeholder<1> std::placeholders::_1'
  211 |     extern const _Placeholder<1> _1;
      |                                  ^~
In file included from /usr/include/boost/bind/bind.hpp:2356,....

/usr/include/boost/bind/placeholders.hpp:47:38: note:                 'constexpr const boost::arg<2> boost::placeholders::_2'
   47 | BOOST_STATIC_CONSTEXPR boost::arg<2> _2;
      |                                      ^~
make[2]: *** [CMakeFiles/node_canopen_cia402_driver.dir/build.make:76: CMakeFiles/node_canopen_cia402_driver.dir/src/node_interfaces/node_canopen_402_driver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:204: CMakeFiles/node_canopen_cia402_driver.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< canopen_402_driver [1min 31s, exited with code 2]

Thank you

ipa-cmh, 2023-06-08:
The second error is certainly solvable. I'll see if I can create a PR for this tonight. Seems to be a thing with the ARM platform, we haven't tested on ARM. Can you also check the boost version you have installed?

The first one I can't tell yet. Can you aswell post the bus.yml?

ipa-cmh, 2023-06-08:
@charanhs123
It seems you are using an older version of boost, which will define boost::bind::placeholders in the golbal namespace. These will the make the usage std::placeholders difficult.

1. Can you run to fetch your boost version?
```bash
dpkg -s libboost-dev | grep 'Version'
```

2. And can you check if https://github.com/ros-industrial/ros2_canopen/pull/130 helps with the build problem? Simply checkout ``branch fix-placeholders-issue``.

charanhs123, 2023-06-12:
> The second error is certainly solvable. I'll see if I can create a PR for this tonight. Seems to be a thing with the ARM platform, we haven't tested on ARM. Can you also check the boost version you have installed?
> 
> The first one I can't tell yet. Can you aswell post the bus.yml?

Hello
1. The boot version is 
jetson@jetson-desktop:~$ dpkg -s libboost-dev | grep 'Version'
Version: 1.71.0.0ubuntu2
2.The Bus.yml is below
  
master: 
  node_id: 100
  package: canopen_master_driver
  driver: ros2_canopen::MasterDriver
  baudrate: 500000 
  heartbeat_consumer: false 
  sync_period: 20000 
   
   
slave1: 
  node_id: 7 
  package: canopen_402_driver 
  driver: ros2_canopen::Cia402Driver
  dcf: maxon_motor_EPOS2_2126h_6322h_0000h_0000h.eds
  period: 20 
  enable_lazy_load: false 
  sdo:
    - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
    - {index: 0x6081, sub_index: 0, value: 1000}
    - {index: 0x6083, sub_index: 0, value: 2000}
  tpdo: # TPDO needed statusword, actual velocity, actual position, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6061, sub_index: 0} # mode of operation display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0} # position actual value
        - {index: 0x606c, sub_index: 0} # velocity actual position
    3:
      enabled: false
    4:
      enabled: false
  rpdo: # RPDO needed controlword, target position, target velocity, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6060, sub_index: 0, value: 0x03} # mode of operation
      - {index: 0x607F, sub_index: 0, value: 25000} # max. Profile Velocity
      - {index: 0x6083, sub_index: 0, value: 10000} # profile acceleration
      - {index: 0x6040, sub_index: 0, value: 10000} # profile deceleration
      - {index: 0x6040, sub_index: 0, value: 0x000F} # controlword
      - {index: 0x60FF, sub_index: 0, value: 0x03} # target velocity
      - {index: 0x6040, sub_index: 0, value: 0x03} # controlword
    
    2:
      enabled: true
      cob_id: "auto"
      mapping:
      #- {index: 0x607A, sub_index: 0} # target position

charanhs123, 2023-06-12:
hello i tried with this branch "fix-placeholders-issue."
but the error is now with the ros2 controllers

--- stderr: canopen_ros2_controllers
CMake Error at CMakeLists.txt:24 (find_package):
  By not providing "Findgenerate_parameter_library.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "generate_parameter_library", but CMake did
  not find one.

  Could not find a package configuration file provided by
  "generate_parameter_library" with any of the following names:

    generate_parameter_libraryConfig.cmake
    generate_parameter_library-config.cmake

  Add the installation prefix of "generate_parameter_library" to
  CMAKE_PREFIX_PATH or set "generate_parameter_library_DIR" to a directory
  containing one of the above files.  If "generate_parameter_library"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< canopen_ros2_controllers [31.6s, exited with code 1]

charanhs123, 2023-06-13:
> Can you run the following before build in the root folder of your ROS workspace?
> 
> ```shell
> rosdep update
> rosdep install --from-paths src/ros2_canopen --ignore-src
> ```
> 
> This will install missing dependencies.

We have built the Docker image such that , it clones some repo s ,builds and source it directly to the pathway ,
So the commands you mentioned are used by me after the packages get built successfully.

I have a doubt if I may ask , i am getting error with the docker now unfortunately, it's showing access denied and image can't be found locally.
I tried docker login but didn't work either
What could be this issue ?

ipa-cmh, 2023-06-13:
@charanhs123 
Sorry, we are following a process here, fixes are done on a seperate branch and only merged into master once tested and reviewed. The version branches (humble, iron, ...) are only synced for releases. This is to make sure people get no to frequent changes that might break their apps.

You are using an outdated version of boost. All currently supported OS (currently ubuntu 22.04 and  debian bullseye) come with boost 1.7.4+, you use 1.7.1 which causes the build problem. I'll not setup a testsystem for non supported libraries. Consequently, if you want to have this fixed, please give me the build output, or use boost 1.7.4+.

charanhs123, 2023-06-15:
Hello I am trying every possible way to update the boost version but my jetson is not at all reponsive with updating the boost
it s giving he output like below

[image]

and for the build output you metioned i am attaching the screen shot below.
[image]
[image]

charanhs123, 2023-06-19:
Hello, also i have one more question is this problem due to OS of Jetson (stuck in 20.04 focal )  or with ROS2 Framework ?

ipa-cmh, 2023-06-20:
Hi @charanhs123,

I have now merged the fix at least for the ambiguous placeholders build issue.
It is now available in master. You were mentioning you as well had a build issue in canopen_ros2_controllers?
If you give me that build output, I can possibly fix it.

I do not know if it is anyone's fault. There are just facts...
* ROS 2 distros support only certain linux distribution versions (see https://www.ros.org/reps/rep-2000.html).
* Nvidia is a bit slow with bringing out latest OS for their Jetsons (Ubuntu 22.04 and ROS 2 Humble are around since April 2022)

In the end the problem is that creating complex systems such as an OS or ROS with thousands of packages is pretty difficult. Making sure that all packages work is only possible when you can be certain that all dependencies are at specific versions.

charanhs123, 2023-06-21:
Hello @ipa-cmh 
I tried with the **Master** branch today,the output is given below,
[image]

ipa-cmh, 2023-06-22:
@charanhs123 sorry forgot your on humble. Please use humble branch.

charanhs123, 2023-06-22:
Hello @ipa-cmh ,
Humble resuleted in same error like mentioned above in Screen shot, unfortunately

ipa-cmh, 2023-06-22:
@charanhs123
Did you git checkout and pull humble branch? It should not result in the QoS error, as there is no `rclcpp::QoS(10)` in humble branches source code.

charanhs123, 2023-07-06:
Hello @ipa-cmh , I had tested this branch but the error was same as mentioned above for humble

ipa-vsp, 2024-02-13:
@charanhs123 Is there any update? 

I will close this issue for now. If you have any updates, then you can reopen it.

closed 2024-02-13 by ipa-vsp