ros-industrial/ros2_canopen #140: Problem when launching cia402_lifecycle

MinElite, 2023-06-14:
Hello, I'm working on control of a differential drive using ros2_canopen. When running the following command 

ros2 launch canopen_tests ci402_lifecycle_setup.launch.py
ros2 lifecycle set /lifecycle_manager configure
ros2 lifecycle set /lifecycle_manager activate

i got the following error message (the last few lines):

[INFO] [launch]: All log files can be found below /home/yan/.ros/log/2023-06-14-10-57-54-049561-yan-10307
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/yan/ros/canopenoffical/install/canopen_tests/share/canopen_tests/config/cia402_lifecycle/bus.yml
[INFO] [launch.user]: /home/yan/ros/canopenoffical/install/canopen_tests/share/canopen_tests/config/cia402_lifecycle/master.dcf
[INFO] [launch.user]: 
[INFO] [launch.user]: can0
[INFO] [cia402_slave_node-1]: process started with pid [10319]
[INFO] [device_container_node-2]: process started with pid [10321]
[device_container_node-2] [INFO] [1686711474.165393736] [device_container_node]: Starting Device Container with:
[device_container_node-2] [INFO] [1686711474.165446486] [device_container_node]: 	 master_config /home/yan/ros/canopenoffical/install/canopen_tests/share/canopen_tests/config/cia402_lifecycle/master.dcf
[device_container_node-2] [INFO] [1686711474.165451599] [device_container_node]: 	 bus_config /home/yan/ros/canopenoffical/install/canopen_tests/share/canopen_tests/config/cia402_lifecycle/bus.yml
[device_container_node-2] [INFO] [1686711474.165455107] [device_container_node]: 	 can_interface_name can0
[device_container_node-2] [INFO] [1686711474.165892487] [device_container_node]: Loading Master Configuration.
[device_container_node-2] [INFO] [1686711474.166103997] [device_container_node]: Load Library: /home/yan/ros/canopenoffical/install/canopen_master_driver/lib/liblifecycle_master_driver.so
[device_container_node-2] [INFO] [1686711474.167545284] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-2] [INFO] [1686711474.167555501] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleMasterDriver>
[device_container_node-2] [INFO] [1686711474.169541532] [master]: NodeCanopenBasicMaster
[device_container_node-2] [INFO] [1686711474.169623003] [device_container_node]: Load master component.
[device_container_node-2] [INFO] [1686711474.169656551] [device_container_node]: Added /master to executor
[device_container_node-2] [INFO] [1686711474.170389650] [device_container_node]: Loading Driver Configuration.
[device_container_node-2] [INFO] [1686711474.170415915] [device_container_node]: Found device cia402_device_1 with driver ros2_canopen::LifecycleCia402Driver
[device_container_node-2] [INFO] [1686711474.170590834] [device_container_node]: Load Library: /home/yan/ros/canopenoffical/install/canopen_402_driver/lib/liblifecycle_cia402_driver.so
[device_container_node-2] [INFO] [1686711474.173800859] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleCia402Driver>
[device_container_node-2] [INFO] [1686711474.173820933] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::LifecycleCia402Driver>
[device_container_node-2] [INFO] [1686711474.181155612] [device_container_node]: Load driver component.
[device_container_node-2] [INFO] [1686711474.181211845] [device_container_node]: Added /cia402_device_1 to executor
[device_container_node-2] [INFO] [1686711474.186581407] [device_container_node]: Loading Manager Configuration.
[device_container_node-2] [INFO] [1686711474.188399685] [device_container_node]: Added /lifecycle_manager to executor
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[cia402_slave_node-1] [INFO] [1686711474.400581615] [cia402_node_1]: Reaching inactive state.
[cia402_slave_node-1] [INFO] [1686711474.407299042] [cia402_node_1]: Reaching active state.
[cia402_slave_node-1] NMT: entering reset application state
[cia402_slave_node-1] NMT: entering reset communication state
[cia402_slave_node-1] NMT: running as slave
[cia402_slave_node-1] NMT: entering pre-operational state
[cia402_slave_node-1] NMT: entering operational state
[cia402_slave_node-1] [INFO] [1686711474.413394002] [cia402_node_1]: Created cia402 slave for node_id 2.
[device_container_node-2] [INFO] [1686711571.144998907] [lifecycle_manager]: Configuring for 2 devices.
[device_container_node-2] [INFO] [1686711571.145094618] [lifecycle_manager]: Found cia402_device_1 (node_id=2)
[device_container_node-2] [INFO] [1686711571.146118023] [lifecycle_manager]: Found master (node_id=1)
[device_container_node-2] [ERROR] [1686711609.221874195] []: Caught exception in callback for transition 13
[device_container_node-2] [ERROR] [1686711609.221915688] []: Original error: CanController: Operation not permitted
[device_container_node-2] [WARN] [1686711609.221939444] []: Error occurred while doing error handling.
[device_container_node-2] [WARN] [1686711609.222226943] [lifecycle_manager]: Failed to trigger transition 3
[device_container_node-2] [ERROR] [1686711609.222283343] [lifecycle_manager]: Failed to bring up master. Activate Transition failed.
[device_container_node-2] [WARN] [1686711609.222298741] []: Error occurred while doing error handling.

System: Ubuntu 22.04 with ROS2 humble

Any idea on what happened here?

Thanks a lot.

Greetings
Min

ipa-cmh, 2023-06-14:
You are running a file that is thought be run on vcan0 on can0. This will not work. The tests only work with virtual can device.
They start specific slaves that mock the behaviour of an actual can device. This is only possible on vcan0.

MinElite, 2023-06-14:
Thank you for your reply! Since I am new with CANOpen, could you give me some hint on how to implement the ros2_canopen package on can0? Should I create my own configuration package?

ipa-cmh, 2023-06-15:
Checkout this example here: https://ros-industrial.github.io/ros2_canopen/manual/rolling/application/trinamic.html

It describes how to create a configuration package for a single motor.
If you want to do it for your device, make sure you have the eds file from the manufacturer and that it is cia402 capabable if it is a motion contorller.

MinElite, 2023-06-15:
Thanks! It would be very helpful. 
Just one more question to close this issue :-p Does the beta release 0.2.0 work for humble? I had some building problems. If the answer is yes I'd like to open a new issue for that.

cheers

closed 2023-06-20 by ipa-cmh