ros-industrial/ros2_canopen #169: Failure to build Trinamic example

r9-pena, 2023-07-06:
Implemented the Trinamic example as per the manual. However, I get the following error. I'm not sure if there additional steps are needed to build this package.

[image]

**Setup:**
 - Device: Mac
 - OS: Ubuntu 20.04
 - ROS-Distro: Foxy
 - Branch/Commit: Foxy

ipa-cmh, 2023-07-07:
Did you build ros2_canopen? Did you source your workspace?

r9-pena, 2023-07-07:
Yes to both questions. 
I would like to note that when only building ros2_canopen, every package within ros2_canopen builds with the only issue occurs with the lely_core_libraries, as shown in the image below. Although, I reviewed the some of the latest CI workflows, and this seems to be a normal occurrence.
[image]

When I include the Trinamic package, a few of them complete building, but the following errors persist. Note the the "cogen_dcf" error.
[image]

ipa-cmh, 2023-07-07:
Those are "normal" warnings from libtool. Super annoying, have already spent hours trying to get rid of them...

**Regarding the real error**
It could be you are missing a ``find_package(lely_core_libraries REQUIRED)`` in your CMakeLists.txt. The cmake command is exported by that package and can't be found if the package is not found.

If this is not the case, please check you have cloned the latest master. We added cogen_def a few weeks back.
Also try building https://github.com/ipa-cmh/trinamic_pd42_can/tree/master (it is still the old yaml format).

r9-pena, 2023-07-07:
I used [https://github.com/ipa-cmh/trinamic_pd42_can/tree/master](https://github.com/ipa-cmh/trinamic_pd42_can/tree/master) (old yaml format) and it worked. Thank you so much for the help.

I will close this issue now.

closed 2023-07-07 by r9-pena