ros-industrial/ros2_canopen #171: Difference between Launch file behaviors

NB-ENG23, 2023-07-13:
**Difference between Launch file behaviors **
 
Are the different behaviors between the test launch files cia402_setup.launch.py and proxy_setup.launch.py (i.e., the canopen messages that I get when I run the candump command)  based only on the EDS, which is responsible for triigireng-specific functions, or is there another high-level code responsible for these behaviors ? 

**For cia402_setup.launch.py**
   vcan0  080   [0] 
  vcan0  182   [3]  40 00 00
  vcan0  282   [8]  00 00 00 00 00 00 00 00
  vcan0  202   [3]  00 01 00
  vcan0  080   [0]

**For proxy_setup.launch.py**
  vcan0  080   [0] 
  vcan0  182   [3]  40 00 00
  vcan0  282   [8]  00 00 00 00 00 00 00 00
  vcan0  202   [3]  00 01 00
  vcan0  080   [0]

ipa-cmh, 2023-07-18:
The bus.yml file is responsible for defining the behavior and which eds and driver to use for each device. Please read the documentation, especially the part on configuration.

NB-ENG23, 2023-07-18:
Yes, this is clear. Thank you for your answer.
But, I was talking about the types of messages and their behavior in responding to such specific messages as SYNC messages, and I found out that they are related to EDS content: PDOs Communication Parameters.

closed 2024-03-11 by ipa-vsp