ros-industrial/ros2_canopen #172: Build Error on Humble

alberthli, 2023-07-13:
**Describe the bug**
When following the build instructions in the manual, I run into the following error:
```
Finished <<< canopen_fake_slaves [6.34s]                                                                    
--- stderr: canopen_core                                
/root/dev_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function 'void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp::Node]':
/root/dev_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:30:32: error: cannot convert 'rclcpp::QoS' to 'const rmw_qos_profile_t&' {aka 'const rmw_qos_profile_s&'}
   30 |     init_service_name, rclcpp::QoS(10), client_cbg_);
      |                                ^~~~~~~
      |                                |
      |                                rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /root/dev_ws/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:33,
                 from /root/dev_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of 'typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]'
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/root/dev_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function 'void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp_lifecycle::LifecycleNode]':
/root/dev_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:73:32: error: cannot convert 'rclcpp::QoS' to 'const rmw_qos_profile_t&' {aka 'const rmw_qos_profile_s&'}
   73 |     init_service_name, rclcpp::QoS(10), client_cbg_);
      |                                ^~~~~~~
      |                                |
      |                                rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:996,
                 from /root/dev_ws/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:34,
                 from /root/dev_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node_impl.hpp:103:29: note:   initializing argument 2 of 'typename rclcpp::Client<ServiceT>::SharedPtr rclcpp_lifecycle::LifecycleNode::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]'
  103 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/node_canopen_driver.dir/build.make:76: CMakeFiles/node_canopen_driver.dir/src/node_interfaces/node_canopen_driver.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:211: CMakeFiles/node_canopen_driver.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< canopen_core [7.43s, exited with code 2]

Summary: 19 packages finished [37.1s]
  1 package failed: canopen_core
  3 packages had stderr output: canopen_core franka_startup lely_core_libraries
  8 packages not processed
```

**To Reproduce**
Minimal reproducible example uses this Dockerfile:
```
# syntax=docker/dockerfile:1

FROM osrf/ros:humble-desktop-full-jammy
SHELL ["/bin/bash", "-c"]
ENV TZ=America/Los_Angeles
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

# installing essential dependencies
RUN apt-get update -y
RUN apt-get install -y \
    build-essential \
    cmake

# clone and build
WORKDIR /root/dev_ws/src
RUN git clone https://github.com/ros-industrial/ros2_canopen.git
WORKDIR /root/dev_ws
RUN rosdep install --from-paths src/ros2_canopen --ignore-src -r -y
RUN source /opt/ros/humble/setup.bash && colcon build --parallel-workers $(nproc)

# source ros2
RUN echo "source /opt/ros/humble/setup.bash; source /root/dev_ws/install/setup.sh" >> ~/.bashrc

# setting up Peak CANController
RUN modprobe peak_usb
RUN ip link set can0 up type can bitrate 1000000
RUN ip link set can0 txqueuelen 1000
RUN ip link set up can0
```

**Expected behavior**
The build should complete without error.

**Setup:**
In the Docker container, we are running Ubuntu 22.04 and ROS2 Humble. The machine itself has the following specs:
 - Device: Nuc 10
 - OS: Ubuntu 22.04
 - ROS-Distro: N/A (only running in Docker)
 - Branch/Commit: Master (commit faca729)

**Additional context**
If we switch to the Humble branch significantly behind `master`, we can build successfully with some warnings. 
```
libtool: warning: relinking 'liblely-tap.la'                                                                                                                                                               
libtool: warning: relinking 'liblely-util.la'                                                                                                                                                              
libtool: warning: relinking 'liblely-can.la'                                                                                                                                                               
libtool: warning: relinking 'liblely-co.la'                                                                                                                                                                
libtool: warning: relinking 'liblely-io.la'                                                                                                                                                                
libtool: warning: relinking 'liblely-ev.la'                                                                                                                                                                
libtool: warning: relinking 'liblely-io2.la'                                                                                                                                                               
libtool: warning: relinking 'liblely-coapp.la'  
```
It would be helpful to release pre-built binaries as is done with Rolling.

ipa-vsp, 2023-07-14:
Hi @alberthli,

The `rclcpp::QoS` error is expected if you use the master branch in ROS2 Humble. Therefore please use the humble branch.  

In ROS2 Rolling `rmw_qos_profile_t`is [deprecated](https://docs.ros.org/en/ros2_packages/rolling/api/rclcpp/generated/page_deprecated.html). 

We are planning the release pre-built binaries for Humble, but there is no fixed timeline yet.

ipa-vsp, 2023-07-14:
I have synched the latest changes with the humble branch.

alberthli, 2023-07-17:
Thanks, this solved the problem - I would suggest updating the instructions in the manual to include specifying the branch in the git commands depending on the ROS2 distro.

closed 2023-07-17 by alberthli

sandropp, 2023-10-28:
hi, i also have this question, how to use the humble branch?  with this command: git checkout -b humble. but error still up.