ros-industrial/ros2_canopen #175: Ros2_Canopen w/ Nanotec Controller

r9-pena, 2023-07-18:
**Describe the bug**
I am trying to attach a nanotec controller (CL3-E-2-0F) to operate a BLDC motor (DB42M03). I created a single-motor package configuration using the Canopen tutorial and the Trinamic Stepper Motor example. However, I get the errors shown in the attached images. I would greatly appreciate any help in resolving this issue and getting the motor to operate w/ ROS2.

**Logs**
Add build or execution log for context.

**Setup:**
 - Device: Jetson AGX Orin
 - OS: Linux
 - ROS-Distro: Foxy
 - Branch/Commit: Foxy

**Additional context**
I have also attached a link to the package repository, for any clarification on the code.
[https://github.com/r9-pena/drive_motor.git](https://github.com/r9-pena/drive_motor.git)

[image]

ipa-cmh, 2023-07-18:
Looks like you do find_package(ros2_canopen) in CmakeLists.txt. ros2_canopen is not a package but a repository containing packages. You probably want find_package(canopen). Also we do not support foxy and the branch is super outdated...

ipa-vsp, 2023-07-18:
#176

ipa-vsp, 2023-07-18:
#176 Fixed the package creation command in the documentation.

ipa-cmh, 2023-09-05:
This seems to be stale. If you need more support, please feel free to reopen.

closed 2023-09-05 by ipa-cmh