ros-industrial/ros2_canopen #177: Problem building lely_core_libraries

jonaskonig, 2023-07-19:
**Describe the bug**
When building via `colcon build` the following output is generated: 
```
Starting >>> lely_core_libraries
Starting >>> canopen_interfaces
Starting >>> roboter_sensor_controller
Starting >>> rplidar_ros
--- stderr: lely_core_libraries                                                                                                                                                
fatal: not a git repository (or any of the parent directories): .git
gmake[2]: *** [CMakeFiles/upstr_lely_core_libraries.dir/build.make:125: upstr_lely_core_libraries-prefix/src/upstr_lely_core_libraries-stamp/upstr_lely_core_libraries-update] Error 128
gmake[1]: *** [CMakeFiles/Makefile2:143: CMakeFiles/upstr_lely_core_libraries.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< lely_core_libraries [0.32s, exited with code 2]
```
I don't understand what that means and how to fix it.

**Setup:**
 - Device: intel Nuc
 - OS: Ubunut 22.04
 - ROS-Distro: Humble
 - Branch/Commit: Humble

jonaskonig, 2023-07-20:
No, there is a working internet connection. It can't find the reference where to pull from. I re-cloned the repo, but this did not change the behavior.

jonaskonig, 2023-07-21:
I am trying to figure out how and why. But after completely uninstalling ros2 and reinstalling it, it now works again.

closed 2023-07-21 by jonaskonig

ipa-cmh, 2023-07-21:
Interesting. But great that it now works!

jclinton830, 2023-07-25:
I have the same exact problem. Is there any other way to fix this without having to reinstall ROS2?