ros-industrial/ros2_canopen #181: Error installing ros2_canopen - "cannot convert 'rclcpp::QoS' to 'const rmw_qos_profile_t&'"

Lau86, 2023-08-03:
Hello,
I am currently installing ROS 2 CANopen on my system, following the recommended procedure on the official GitHub repository. However, I am encountering an error when running the colcon build command. The specific error I get is as follows:

"error: cannot convert 'rclcpp::QoS' to 'const rmw_qos_profile_t&'"{aka 'const rmw_qos_profile_s&'}
30 |        init_service_name, rclcpp::QoS(10), client_cbg_);
      |                                                     ^~~~~~~~
      |                                                     |
      |                                                     rclcpp::QoS
   
package failed: canopen_core

I have checked that I am using the latest version of ROS 2 and that I have correctly installed all the dependencies required for ROS 2 CANopen. However, the error persists.

I've also checked the README.md file in the repository, but I can't find any specific information about this error.

Have any of you already encountered this error when installing ROS 2 CANopen? If so, could you help me to understand the underlying cause of this error and to find a solution?

I'm new to ROS so any help or suggestions would be greatly appreciated. 
Thank you in advance for your support!

gavanderhoorn, 2023-08-03:
Probably duplicate of #172.

ipa-cmh, 2023-08-03:
yep, @Lau86 we haven't gotten to update the docs yet. If you use ros2 humble, please use the humble branch.

Lau86, 2023-08-03:
Thank you very much for your quick reply. I was able to solve the problem using the Humble branch as you suggested. 
Thanks again for your help !

kku-hydrive, 2024-03-15:
Hello,
I am using ROS Humble and I tried using the Humble branch but I see the same issue.