ros-industrial/ros2_canopen #182: Robot system with ros2_control

Lau86, 2023-08-04:
Hello,
I'm currently working on creating a robot system with ros2_control. I have followed the instructions in the manual. However, I'm having problems building my package. More specifically, I'm getting the following errors:

Traceback (most recent call last):
  File "/home/ros2_humble/install/lely_core_libraries/bin/cogen", line 33, in <module>
    sys.exit(load_entry_point('cogen==0.2.7', 'console_scripts', 'cogen')())
  File "/home/ros2_humble/install/lely_core_libraries/local/lib/python3.10/dist-packages/cogen/cogen.py", line 59, in main
    nodes[node_name][entry_name] = defaults[entry_name]
TypeError: list indices must be integers or slices, not dict
gmake[2]: *** [CMakeFiles/robot_control.dir/build.make:73: robot_control] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/robot_control.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< canopen_robot_control_example [0.66s, exited with code 2]

(ros2_humble is the name of my workspace)

Have any of you already encountered this error when building this package? If so, could you help me understand the underlying cause of this error and find a solution?

Any help or suggestions would be greatly appreciated. Thank you in advance for your support!

ipa-cmh, 2023-08-08:
Looks like there is an issue with the nodes section. It seems to be yaml issue in the config file. Can't say more without the config file though.

flippybit, 2023-08-09:
Hi @ipa-cmh  im having the same error and using the exact config files from the tutorial 
"How to create a robot system with ros2_control"
(using the humble [readthedocs)](https://ros-industrial.github.io/ros2_canopen/manual/humble/user-guide/how-to-create-a-robot-system.html)
<details>
<summary> Files used following the tutorials  </summary>

#### ros2_controllers.yaml
``` yaml
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    forward_position_controller:
      type: forward_command_controller/ForwardCommandController

forward_position_controller:
  ros__parameters:
    joints:
      - joint1
      - joint2
    interface_name: position
```
#### bus.yml

``` yaml
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 1
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  sync_period: 10000

defaults:
  dcf: "cia402_slave.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  period: 10
  position_mode: 1
  revision_number: 0
  sdo:
    - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
    - {index: 0x6081, sub_index: 0, value: 1000}
    - {index: 0x6083, sub_index: 0, value: 2000}
    - {index: 0x6060, sub_index: 0, value: 7}
  tpdo: # TPDO needed statusword, actual velocity, actual position, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6061, sub_index: 0} # mode of operation display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0} # position actual value
        - {index: 0x606c, sub_index: 0} # velocity actual position
  rpdo: # RPDO needed controlword, target position, target velocity, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0} # controlword
[... 149 lines cut ...]
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< canopen_robot_control_example [0.52s, exited with code 2]

Summary: 0 packages finished [0.75s]
  1 package failed: canopen_robot_control_example
  1 package had stderr output: canopen_robot_control_example

```
</details>

### Conclusion
Any help would be apreciated, 
I'm trying to use this package to communicate with my robots servo motors "the proper way" but i guess i am condemned to use arduino code like a noob 4eva 

``` ino
  // Set synchronous velocity profile
    if (input == "set\n"){
        Serial.print("received:");
        Serial.print(input);
        CAN0.sendMsgBuf(0x601, 0, 8, set_sync_control);
        CAN0.sendMsgBuf(0x601, 0, 8, set_profile_vel_mode); 
```
BTW you guys are the best ! :heart: 
would love to contribute to this repo even if only to correct some of the typos or outdated docs (but to scared to do so) :fearful:

ipa-vsp, 2023-08-09:
Hi @flippybit and @Lau86,

To create a package, please use the following command instead of the one in the documentation: 

`ros2 pkg create --dependencies canopen lely_core_libraries --build-type ament_cmake {package_name}`

Also, to remove `TypeError: list indices must be integers or slices, not dict` change the following ins bus.yml.

```
nodes:
    joint_1:
      node_id: 2
    joint_2:
      node_id: 3
```
This will be fixed in #176

Lau86, 2023-08-10:
Thank you very much for your help !

flippybit, 2023-08-10:
Hi @ipa-vsp and @Lau86,
> First of all thanx @ipa-vsp  for pointing my in the right direction and @Lau86 for starting this thread.
> I dont know if im breaching github etiquette with the following post please correct me if i should open a new issue instead.

I have continued with the "How to create a robot system with ros2_control"
(using the humble [readthedocs)](https://ros-industrial.github.io/ros2_canopen/manual/humble/user-guide/how-to-create-a-robot-system.html)
And have found other potential pitfalls for others following this guide.

## *Point 11. Start your launch file* 
### FileNotFoundError: [Errno 2] No such file or directory:
At this point you should have built your package and sourced setup.bash file and be ready to launch your newly built package, if following the guide should be called **canopen_robot_control_example**

The command to do this should look like this:
```bash
$ ros2 launch canopen_robot_control_example robot_control.launch.py 
```
<details>
<summary>But i got the following error output:</summary>

```bash 
[INFO] [launch]: All log files can be found below /home/water/.ros/log/2023-08-10-18-58-35-285345-water-tender-18171
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /home/water/ROS2PORT/canopen_ws/install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro
Captured stderr output: Traceback (most recent call last):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1012, in parse
    inp = f = open(filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/water/ROS2PORT/canopen_ws/install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1119, in _process
    doc = process_file(input_file_name, **opts)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1096, in process_file
    doc = parse(None, input_file_name)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1016, in parse
    raise XacroException(e.strerror + ": " + e.filename, exc=e)
xacro.XacroException: No such file or directory: /home/water/ROS2PORT/canopen_ws/install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro [Errno 2] No such file or directory: '/home/water/ROS2PORT/canopen_ws/install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/humble/bin/xacro", line 33, in <module>
    sys.exit(load_entry_point('xacro==2.0.8', 'console_scripts', 'xacro')())
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1171, in main
    _process(input_file_name, vars(opts))
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1124, in _process
    except xml.parsers.expat.ExpatError as e:
AttributeError: module 'xml' has no attribute 'parsers'

```
</details>

It seems like there is a FileNotFoundError:
```bash
FileNotFoundError: [Errno 2] No such file or directory: '/home/water/ROS2PORT/canopen_ws/install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro'

```
(my path and workspace name will not be the same as yours, in my case my workspace name is called **canopen_ws** )

~~`/home/water/ROS2PORT/canopen_ws/`~~`install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro`

### What is think is happening
<details>
<summary> Click to expand </summary>

**robot_control.launch.py** file is trying to access the **robot_controller.urdf.xacro** file in this path:

`canopen_ws/install/canopen_tests/share/canopen_tests/robot_controller/robot_controller.urdf.xacro`

[... 108 lines cut ...]
            "slave_config": slave_config,
        }.items(),
    )

    slave_node_2 = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(slave_launch),
        launch_arguments={
            "node_id": "3",
            "node_name": "slave_node_2",
            "slave_config": slave_config,
        }.items(),
    )

    nodes_to_start = [
        control_node,
        joint_state_broadcaster_spawner,
        forward_position_controller_spawner,
        robot_state_publisher_node,
        slave_node_1,
        slave_node_2
    ]

    return LaunchDescription(nodes_to_start)
```
</details>

ipa-cmh, 2023-08-11:
@flippybit 
Many thanks for the review! 
I would really appreciate a PR if you can do that?

flippybit, 2023-08-11:
@ipa-cmh 
Thanx to you! for your awesome ROS2  package
I would be honored to do so! 
But im not sure if ill do it right. But **ill give it my best try.** :sweat_smile:

ipa-cmh, 2023-09-05:
@flippybit thanks for your help, we @ipa-vsp merged the your changes with #176. I will close this.

closed 2023-09-05 by ipa-cmh