ros-industrial/ros2_canopen #183: how to use this ros2_canopen?    what():  Device: No such file or directory

zouzhe1, 2023-08-07:
1，Firstly, for beginners, following your DOC document webpage is not successful.

2，
[https://ros-industrial.github.io/ros2_canopen/manual/humble/application/trinamic.html](https://ros-industrial.github.io/ros2_canopen/manual/humble/application/trinamic.html)

I found an example at the end of the document. I downloaded the program, compiled it successfully, and run with the following error message.

```ros2 launch trinamic_pd42_can direct_launch.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-08-07-16-04-45-008809-B-ROS2-IGH-RT-63910
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/b/ros2wORKSPACE/ros2_canopen/install/trinamic_pd42_can/share/trinamic_pd42_can/config/single-pd42/bus.yml
[INFO] [launch.user]: /home/b/ros2wORKSPACE/ros2_canopen/install/trinamic_pd42_can/share/trinamic_pd42_can/config/single-pd42/master.dcf
[INFO] [launch.user]: /home/b/ros2wORKSPACE/ros2_canopen/install/trinamic_pd42_can/share/trinamic_pd42_can/config/single-pd42/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [63922]
[device_container_node-1] [INFO] [1691395485.128755885] [device_container_node]: Starting Device Container with:
[device_container_node-1] [INFO] [1691395485.128844374] [device_container_node]:         master_config /home/b/ros2wORKSPACE/ros2_canopen/install/trinamic_pd42_can/share/trinamic_pd42_can/config/single-pd42/master.dcf
[device_container_node-1] [INFO] [1691395485.128854198] [device_container_node]:         bus_config /home/b/ros2wORKSPACE/ros2_canopen/install/trinamic_pd42_can/share/trinamic_pd42_can/config/single-pd42/bus.yml
[device_container_node-1] [INFO] [1691395485.128861964] [device_container_node]:         can_interface_name can0
[device_container_node-1] [INFO] [1691395485.129336810] [device_container_node]: Loading Master Configuration.
[device_container_node-1] [INFO] [1691395485.129700090] [device_container_node]: Load Library: /home/b/ros2wORKSPACE/ros2_canopen/install/canopen_master_driver/lib/libmaster_driver.so
[device_container_node-1] [INFO] [1691395485.132414097] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1691395485.132438733] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1691395485.134720944] [master]: NodeCanopenBasicMaster
[device_container_node-1] [INFO] [1691395485.134827530] [device_container_node]: Load master component.
[device_container_node-1] [INFO] [1691395485.134878100] [device_container_node]: Added /master to executor
[device_container_node-1] terminate called after throwing an instance of 'std::system_error'
[device_container_node-1]   what():  Device: No such file or directory
[ERROR] [device_container_node-1]: process has died [pid 63922, exit code -6, cmd '/home/b/ros2wORKSPACE/ros2_canopen/install/canopen_core/lib/canopen_core/device_container_node --ros-args -r __node:=device_container_node -r __ns:=/ --params-file /tmp/launch_params_69tz4jmk --params-file /tmp/launch_params_u6wybrjs --params-file /tmp/launch_params_9m9i2dn9 --params-file /tmp/launch_params_sroinpxx'].
```

3,
I don't know if this is a permission issue, I made the same mistake when entering root #

4，
I am sure that my can device can communicate and use normally, and can linux(ros2+canopen)  send data to WIN10(USB CAN) normally.

ipa-cmh, 2023-08-07:
Seems you do not have the device can0 set on your linux machine. Usually you need to activate it first, depending on the CAN interface you have on your machine. A few examples are here: https://ros-industrial.github.io/ros2_canopen/manual/rolling/quickstart/operation.html

zouzhe1, 2023-08-07:
@ipa-cmh   i can see  “CAN0”，when i run ifconfig command.
and  i can use   can-utils  tool   cansend message!
and i can  receive message.
linux --->win10, is  ok. (can-utils)

qfwysw, 2023-08-07:
I have the same question.

ipa-cmh, 2023-08-11:
Can't reproduce this error here. Need more info about you system.

flippybit, 2023-08-12:
Hey guys,
I'm having the same issue but i have not got around to doing any investigations. 
:male_detective: could be something in the `DeviceContainer`
Just wanted to tell you guys that you are not alone in this fight  :muscle:
building robots aint easy! 

@zouzhe1 If you have any tips for others trying to navigate the DOCS you could share them with the rest of us ! 
i have made a controller using `python opencan` but its python :smile: and its just too slow you could give it a try! 

Anyways @ipa-cmh i appreciate the hardwork you put into this repo ! 

cya around :heart:

ipa-cmh, 2023-09-05:
@zouzhe1 and @flippybit this seems to be stale. I'll close it for now. Feel free to reopen.

closed 2023-09-05 by ipa-cmh