ros-industrial/ros2_canopen #185: When I run the example of a stepper motor, the master always automatically sends a PDO to 0x6040. How can I cancel this sending?

qfwysw, 2023-08-11:
**Describe the bug**
A clear and concise description of what the bug is.

**Logs**
Add build or execution log for context.

**Setup:**
 - Device:
 - OS:
 - ROS-Distro:
 - Branch/Commit:

**Additional context**
When I run the example of a stepper motor, the master always automatically sends a PDO to 0x6040. How can I cancel this sending?

ipa-cmh, 2023-08-11:
You can't if you use cia402 driver. It will always try to bring your device to operation enabled mode, and thus write to 0x6040. This happens after all configuration is done.

qfwysw, 2023-08-11:
Thank you for your generous response. If I annotate and send 0x6040 related code, will it affect other functions of your code?

ipa-cmh, 2023-08-11:
Honestly, I have never tried. When do you need to send commands to 0x6040? Before the driver does anything? In which NMT state of the device?

qfwysw, 2023-08-11:
I hope to enable the motor after sending the nmt and send a series of instructions to 0x6040 to enable the motor. Then, the cia402 driver sends the enabling data to 0x6040 from the beginning

ipa-cmh, 2023-08-11:
Currently, the master will start the initialization as soon as it receives the wake-up signal from the remote device. Sending some custom messages is not foreseen currently and you would need to patch the cia402_driver. If you simply want to do homeing, this is foreseen but not tested yet.

qfwysw, 2023-08-11:
Currently, the cia402 driver is sending some default values to 0x4060. How can I change these default values to match my own motor?

ipa-cmh, 2023-08-17:
@qfwysw you cannot, the whole purpose of having a standard profile such as CiA402 is to not have to change those values.
By sending things to 0x6040 you switch through the standardised state machine of the CiA402 state machine. The driver is developed to bring your device to the state operation enabled, based on that state machine. If your device does not follow that state machine, you will have to write your own driver.

qfwysw, 2023-08-17:
Thank you for your generous answer

closed 2023-08-17 by qfwysw