ros-industrial/ros2_canopen #189: Help creating a ros based slave.

sam-tj, 2023-08-30:
Hi Team,
I am finding some method to connect a PLC with ROS2 using CANopen. The main thing is that ROS must be as a slave device and PLC is the master device. Hence my plan was to make a EDS file for required topics from ROS and then use PLC to pull them.
As of what I understood from examples available on the website, ROS acts as master and devices like motor controller act as slave.

Please can someone help me with this.
Thanks in advance.

ipa-cmh, 2023-08-31:
It depends on what you want to do. Generally, for communication with a PLC I would recommend using opc-ua instead of CANopen. Most modern PLCs support that. If you really want to go with CANopen, then check the our fake slaves package for some hints how to use lely core library to create a slave.

sam-tj, 2023-08-31:
Thanks for your reply. 

I built master for humble but didn't work so used humble branch.
The launch file works and I can see that the device connected to the PLC. But how can I update / read the values from ROS side? I don't see any topics or services available. 

If possible, can you share some tutorials on how to use the fake slave package. :)

ipa-cmh, 2023-09-05:
Our fake slaves are not for what you seek. They do not provide any ROS topics. We use them for testing. They can be a reference (hint) for you to write a CANopen slave with lely core library that publishes the information you want. Sorry, but you will have to code that yourself.

sam-tj, 2023-09-05:
Thanks for the reply. 

I see there is a motion control file but it doesn't work as node so any tips how can I run that. Maybe that would work for my tests.

closed 2023-10-13 by ipa-cmh