ros-industrial/ros2_canopen #190: ros2_canopen - colcon build

zhiyiopp, 2023-08-31:
**Describe the bug**
after download the package, following the operation steps, in "colcon build" then:
Starting >>> lely_core_libraries
--- stderr: lely_core_libraries                                                                           
CMake Error at /opt/ros/foxy/share/ament_cmake_python/cmake/ament_python_install_package.cmake:34 (message):
  ament_python_install_package() called with unused arguments:
  SCRIPTS_DESTINATION;lib/cogen
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_python/cmake/ament_python_install_package.cmake:28 (_ament_cmake_python_install_package)
  CMakeLists.txt:41 (ament_python_install_package)
---
Failed   <<< lely_core_libraries [0.30s, exited with code 1]
Aborted  <<< canopen_interfaces [9.23s]                         

**Logs**
Add build or execution log for context.

**Setup:**
 - Device: VM 
 - OS: Ubuntu 20.04.6 LTS
 - ROS-Distro: ros2-foxy
 - Branch/Commit:

**Additional context**
Add any other context about the problem here.

ipa-cmh, 2023-08-31:
We don't support foxy anymore. It is end of life. Please use humble (humble branch) or iron or rolling.

zhiyiopp, 2023-08-31:
oh, i will upgrade it and try again.

sam-tj, 2023-08-31:
I had the same problem with humble, used humble branch to make it work.

ipa-cmh, 2023-09-05:
@sam-tj can you post the exact error?

sam-tj, 2023-09-05:
This is the error.

[image]

ipa-cmh, 2023-09-05:
This is because QoS initialization as it was done in humble is deprecated in Iron and Rolling and the new way does not work in humble. You currently use master branch of this repo but you have to use the Humble branch to work with humble.

sam-tj, 2023-09-05:
Yes, that's why I moved to humble branch and that works for me.

closed 2023-10-13 by ipa-cmh