ros-industrial/ros2_canopen #191: RPDO/TPDO configuration

gsalinas, 2023-08-31:
Firstly this is, AFAICT, not a bug, just something I'm unclear on, so apologies for that! If there's a forum the team uses please let me know and I can dig around / ask there!

My question is: are TPDOs and RPDOs in `bus.yml` files meant to:
a) describe the configuration of PDOs as they already exist on the device?
or
b) describe the configuration of PDOs that I want to set up on my device?

In case (a) I would expect to figure out from my device documentation what its PDOs are, and if they're not what I want, I would set up the writes to the appropriate registers in the SDO section of `bus.yml`, which would execute all those SDO writes during startup to configure the PDOs accordingly, then also write them in the TPDO and RPDO sections, whereas in case (b) all those SDO writes would be determined automatically from the `tpdo` and `rpdo` sections of `bus.yml` and still done during startup, but without me writing anything in the SDO section (other than any other one-time SDO writes I want for my situation.)

Apologies if this is described somewhere in the docs, I couldn't really see it anywhere! Thank you so much!

ipa-cmh, 2023-08-31:
It is b. The devices eds is read. Pdos set in eds will be known to the master. The bus.yml overwrites these. The pdos defined in bus.yml are compiled to sdo commands that then are sent during startup.

gsalinas, 2023-08-31:
Fantastic, thank you so much!

closed 2023-08-31 by gsalinas