ros-industrial/ros2_canopen #194: Change SDO timeout setting?

gsalinas, 2023-09-05:
Hi again!

I've run into an issue doing PDO configuration at the startup of my node where the SDO responses from my device to the configuration messages being received occasionally time out. This could of course be for a variety of reasons and may require further debugging, but in some cases my Wireshark traces show that the device does send a response, sometimes only ~10ms too late, at which point the SDO download has already failed, and the PDO configuration fails as well. Is there a way to configure the SDO timeout settings of my master or my devices?

Thank you so much for your help!

ipa-cmh, 2023-09-07:
There is a setting in the underlying lelycore library for this, we just currently do not expose it. I'll have to check on the work necessary to expose.

gsalinas, 2023-09-07:
Thank you for looking into it!

ipa-cmh, 2023-10-06:
@ipa-vsp will create a pr for this in the coming days. It will expose this timeout as setting to be set in bus.yml in the master section.

ipa-vsp, 2024-03-11:
#220

closed 2024-03-11 by ipa-vsp