ros-industrial/ros2_canopen #196: Configure homing timeout

lucasmluza, 2023-09-14:
I'm working in a project in which the homing procedure for a specific joint takes longer than 10 seconds. It then generates an error when executing the package, as the function that manages the homing has a fixed timeout (```finish_time```) of 10 seconds.

For reference, the timeout is hard coded here:

https://github.com/ros-industrial/ros2_canopen/blob/faca7297438bef5d7a986122959988d16a567246/canopen_402_driver/src/default_homing_mode.cpp#L89-L92

Considering that each joint can have a different timeout to be specified, it would be very interesting to expose this ```finish_time``` as a parameter in the bus configuration file.

ipa-cmh, 2023-10-06:
@lucasmluza I will see how we can do that.
I would make it an optional parameter where the default would be 10s.

lucasmluza, 2023-10-06:
@ipa-cmh, thank you for your attention.