ros-industrial/ros2_canopen #197: Number of RPDO callback calls per mapped TPDO object

gsalinas, 2023-09-18:
Hi! Thanks once again for this package and all the assistance.

The proxy driver publishes one message to `proxy_device_1/rpdo` for each TPDO mapped object on the corresponding device, so for example, if my device has 4 TPDOs of 8 objects each, all of which transmit every sync period, then each sync period the proxy driver publishes to that topic 32 times. This happens regardless of, and as far as I can tell, without any awareness of, the TPDO numbering assigned in the bus.yml file (see an example of my situation below.)

I'm writing a custom device driver for a sensor, some of whose measurements are split across multiple objects in its object dictionary. If I extend the ProxyDriver class, is it possible to register an RPDO callback based on all 8 objects of TPDO 1 being available? The grouping of the TPDOs ensures that all the relevant objects are sampled synchronously on the device, which appears to be information that is discarded by then calling the RPDO callback 8 separate times.

Thank you so much!

```
  tpdo:
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x2000, sub_index: 1}  
        - {index: 0x2000, sub_index: 2}  
        - {index: 0x2000, sub_index: 3}  
        - {index: 0x2000, sub_index: 4}  
        - {index: 0x2000, sub_index: 5}  
        - {index: 0x2000, sub_index: 6}  
        - {index: 0x2000, sub_index: 7}  
        - {index: 0x2000, sub_index: 8} 

    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x2000, sub_index: 9}  
        - {index: 0x2000, sub_index: 10} 
        - {index: 0x2000, sub_index: 11} 
        - {index: 0x2000, sub_index: 12} 
        - {index: 0x2000, sub_index: 13} 
        - {index: 0x2000, sub_index: 14} 
        - {index: 0x2000, sub_index: 15}
        - {index: 0x2000, sub_index: 16}
    3:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x2000, sub_index: 17} 
        - {index: 0x2000, sub_index: 18} 
        - {index: 0x2000, sub_index: 19} 
        - {index: 0x2000, sub_index: 20} 
        - {index: 0x2000, sub_index: 21} 
        - {index: 0x2000, sub_index: 22} 
        - {index: 0x2000, sub_index: 23} 
        - {index: 0x2000, sub_index: 24} 
    4:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x2000, sub_index: 25} 
        - {index: 0x2000, sub_index: 26} 
        - {index: 0x2000, sub_index: 27} 
        - {index: 0x2000, sub_index: 28} 
        - {index: 0x2000, sub_index: 29}  
        - {index: 0x2000, sub_index: 30}  
        - {index: 0x2000, sub_index: 31}  
        - {index: 0x2000, sub_index: 32} 
```

I'm implementing a custom device driver, and for my device, I'd like to

ipa-cmh, 2023-10-06:
@gsalinas that is interesting. If you write a custom driver you can probably put these together, lely core will create a callback for each object that is updated seperately though.

gsalinas, 2023-10-06:
@ipa-cmh yeah, for this case I ended up creating a custom driver still based on lely-core that builds an array from the 32 individual callbacks, packages and publishes it to my topic when it has all 32 messages, and clears on sync. This mostly works OK - I have some kind of problem where occasionally (~1 in 1e5 times) I get multiple callbacks for the same subindex before I've done all my callbacks for different subindices, and I publish something slightly incorrect, but I think I'll be able to solve that with a bit more digging in.

To get it working I did have to touch a few other things that might be worth thinking about for the base library: specifically, `poll_timer_callback` does a for loop over the RPDO queue where it tries to get an item from the queue, skips to the next iteration if it cannot, and calls `rpdo_cb_` and `on_rpdo` if it can. This loop is hardcoded to only do 10 iterations, which means it'll never process more than 10 PDOs and will miss the rest entirely. I changed this in my custom driver but it's a problem for the proxy driver and anything derived from it that wants more than 10 PDOs.

ipa-cmh, 2023-10-13:
> To get it working I did have to touch a few other things that might be worth thinking about for the base library: specifically, `poll_timer_callback` does a for loop over the RPDO queue where it [...]

So we should increase it to 32 that would the be the maximum for CANopen. Honestly, I just wanted to limit what is processed in the callback. Do you want to create a pr for this?