ros-industrial/ros2_canopen #199: Boot issues

janknipschild, 2023-09-22:
**Describe the bug**
I keep encountering boot issues. It seems that I cannot read the vendor ID. Wireshark shows that the SDO transmission is aborted ("command specifier unknown"). I also have to manually reset the node by sending the nmt message via `cansend can0 000#8105` to get a boot up message. Is there a configuration error? The services sdo_read and sdo_write are working.

**Logs**
```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/bus.yml
[INFO] [launch.user]: /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/master.dcf
[INFO] [launch.user]: /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/master.bin
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [63203]
[device_container_node-1] [INFO] [1695375612.787080333] [device_container_node]: Starting Device Container with:
[device_container_node-1] [INFO] [1695375612.787124750] [device_container_node]:         master_config /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/master.dcf
[device_container_node-1] [INFO] [1695375612.787128995] [device_container_node]:         bus_config /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/bus.yml
[device_container_node-1] [INFO] [1695375612.787133040] [device_container_node]:         can_interface_name can0
[device_container_node-1] [INFO] [1695375612.787512256] [device_container_node]: Loading Master Configuration.
[device_container_node-1] [INFO] [1695375612.787748421] [device_container_node]: Load Library: /home/entwicklung/ws_ros/install/canopen_master_driver/lib/libmaster_driver.so
[device_container_node-1] [INFO] [1695375612.789307070] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1695375612.789315602] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1695375612.790683164] [master]: NodeCanopenBasicMaster
[device_container_node-1] [INFO] [1695375612.790749669] [device_container_node]: Load master component.
[device_container_node-1] [INFO] [1695375612.790784572] [device_container_node]: Added /master to executor
[device_container_node-1] [INFO] [1695375612.799862776] [device_container_node]: Loading Driver Configuration.
[device_container_node-1] [INFO] [1695375612.799896859] [device_container_node]: Found device joint1 with driver ros2_canopen::ProxyDriver
[device_container_node-1] [INFO] [1695375612.800013739] [device_container_node]: Load Library: /home/entwicklung/ws_ros/install/canopen_proxy_driver/lib/libproxy_driver.so
[device_container_node-1] [INFO] [1695375612.802487164] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[device_container_node-1] [INFO] [1695375612.802496293] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[device_container_node-1] [INFO] [1695375612.803973021] [device_container_node]: Load driver component.
[device_container_node-1] [INFO] [1695375612.804016301] [device_container_node]: Added /joint1 to executor
[device_container_node-1] [ERROR] [1695375612.807274262] [joint1]: Could not read enable diagnostics from config, setting to false.
[device_container_node-1] [INFO] [1695375612.807999197] [joint1]: eds file /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/C5-E-2-09_auto_allign.eds
[device_container_node-1] [INFO] [1695375612.808009908] [joint1]: bin file /home/entwicklung/ws_ros/install/robot_ros2_canopen/share/robot_ros2_canopen/config/robot_bus/joint1.bin
[device_container_node-1] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-1] Found rpdo mapped object: index=6060 subindex=0
[device_container_node-1] Found rpdo mapped object: index=3202 subindex=0
[device_container_node-1] Found rpdo mapped object: index=607a subindex=0
[device_container_node-1] Found rpdo mapped object: index=6081 subindex=0
[device_container_node-1] Found rpdo mapped object: index=6042 subindex=0
[device_container_node-1] Found rpdo mapped object: index=60fe subindex=1
[device_container_node-1] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6061 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6064 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6044 subindex=0
[device_container_node-1] Found tpdo mapped object: index=60fd subindex=0
[device_container_node-1] Found tpdo mapped object: index=2039 subindex=2
[device_container_node-1] [WARN] [1695375612.814792347] [joint1]: Wait for device to boot.
[device_container_node-1] [ERROR] [1695375616.288240291] [joint1]: Boot Issue: Value of object 1018:01 from CANopen device is different to value in object 1F85 (Vendor-ID).
[device_container_node-1] [INFO] [1695375616.288368819] [joint1]: Driver booted and ready[device_container_node-1] [INFO] [1695375616.288929608] [joint1]: Starting with event mode.
```

wireshark log
```
No.     Time           Protocol Length ID         Data       Info
      1 0.000000000    CAN      32     1802 (0x70a) 00         ID: 1802 (0x70a), Length: 1

Frame 1: 32 bytes on wire (256 bits), 32 bytes captured (256 bits) on interface can0, id 0
Linux cooked capture v1
Controller Area Network, ID: 1802 (0x70a), Length: 1
Data (1 byte)

0000  00                                                .

No.     Time           Protocol Length ID         Data       Info
      2 3.278168975    AUTOSAR NM 32     0 (0x000)             NM (CBV: 0x82, SNI: 0x05)

Frame 2: 32 bytes on wire (256 bits), 32 bytes captured (256 bits) on interface can0, id 0
Linux cooked capture v1
Controller Area Network, ID: 0 (0x0), Length: 2
AUTOSAR Network Management, Control Bit Vector: 0x82, Source Node: 5

[... 166 lines cut ...]
AccessType=ro
DefaultValue=0x00000011
PDOMapping=0
ObjFlags=1

[1018sub3]
ParameterName=Revision number
ObjectType=0x7
DataType=0x0007
AccessType=ro
DefaultValue=0x06720000
PDOMapping=0
ObjFlags=1

[1018sub4]
ParameterName=Serial number
ObjectType=0x7
DataType=0x0007
AccessType=ro
DefaultValue=
PDOMapping=0
ObjFlags=1

...
```

ipa-cmh, 2023-09-22:
The vendor-id on the device and the one in the eds-file will probably not be the same.
Try to read out 0x1018sub1 to check. The master automatically checks this.

janknipschild, 2023-09-22:
> The vendor-id on the device and the one in the eds-file will probably not be the same. Try to read out 0x1018sub1 to check. The master automatically checks this.

The read out value is the same as in the eds-file:

candump
`can0  585   [8]  43 18 10 01 6C 02 00 00`

eds
`VendorNumber=0x26C`

ipa-cmh, 2023-09-22:
Hmm interesting, can you the check in the generated master.dcf (it is in the install/package_name/share/config folder) and search for 0x1F85?

janknipschild, 2023-09-22:
> Hmm interesting, can you the check in the generated master.dcf (it is in the install/package_name/share/config folder) and search for 0x1F85?

Looks right to me:

```
[1F85]
ParameterName=Vendor identification
ObjectType=0x08
DataType=0x0007
AccessType=rw
CompactSubObj=127

[1F85Value]
NrOfEntries=1
5=0x0000026C

```

janknipschild, 2023-09-25:
The problem was not with the  `ros2_canopen` package, but with the driver used for the USB-to-CAN adapter. (https://gitlab.com/lely_industries/lely-core/-/issues/173)
With the loopback activated the boot process works.

closed 2023-09-25 by janknipschild

LudoJ49, 2024-01-30:
I have the same issue. Can you be more precise on how you fixed it ?
thx

janknipschild, 2024-01-30:
The loopback is activated during the setup. Assuming that your interface is named can0:
`sudo ip link set can0 type can bitrate 1000000 loopback on`
`sudo ip link set can0 txqueuelen 1000`
`sudo ip link set can0 up`

LudoJ49, 2024-01-31:
thanks. I'll give it a try