ros-industrial/ros2_canopen #201: Trying to integrate a CANopen motor

VerlooyNick, 2023-10-05:
I'm trying to run a Nanotec motor through the ros2_canopen stack. 
Following the documentation and trinamic [tutorial](https://ros-industrial.github.io/ros2_canopen/manual/humble/application/trinamic.html), I'm able to init, start and stop the motor by using the service calls.
Now I'm looking for the next step: integrate the motor into my mobile robot. Therefore, I expected the /target [topic](https://ros-industrial.github.io/ros2_canopen/manual/humble/user-guide/cia402-driver.html) to be available to publish to. Unfortunate this is not the case, I'm not seeing the /target topic in the list. So currently I'm not able to publish targets. Any suggestions? 

**Logs**
`ros2 topic list
/device_container_node/transition_event
/motor_left/joint_states
/motor_left/nmt_state
/motor_left/rpdo
/motor_left/tpdo
/motor_right/joint_states
/motor_right/nmt_state
/motor_right/rpdo
/motor_right/tpdo
/parameter_events
/rosout`

bus.yml
`options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 10
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  sync_period: 1000

defaults:
  dcf: "PD4-CB59M024035-E-08.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  polling: true
  period: 10
  diagnostics:
    enable: false
  scale_pos_to_dev : 1.0
  scale_vel_to_dev : 1.0
  scale_pos_from_dev : 1.0
  scale_vel_from_dev : 1.0
  heartbeat_producer: 2000
  sdo: # SDO executed during config
    - {index: 0x6081, sub_index: 0, value: 1} # Set velocity
    - {index: 0x6083, sub_index: 0, value: 600} # Set acceleration
    - {index: 0x6083, sub_index: 0, value: 600} # Set deceleration
    - {index: 0x6085, sub_index: 0, value: 6000} # Set quickstop deceleration
    - {index: 0x6098, sub_index: 0, value: 35} # Set default homing mode to 35
    - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s

nodes:
  motor_left:
    node_id: 1

  motor_right:
    node_id: 2
`
launch file
`import os
import sys

sys.path.insert(0, os.path.join(os.path.dirname(__file__), ".."))  # noqa
sys.path.insert(0, os.path.join(os.path.dirname(__file__), "..", "..", "launch"))  # noqa

import launch
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from ament_index_python import get_package_share_directory
from launch import LaunchDescription

def generate_launch_description():
    ld = LaunchDescription()

    device_container = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(get_package_share_directory("canopen_core"), "launch"),
                "/canopen.launch.py",
            ]
        ),
        launch_arguments={
            "master_config": os.path.join(
                get_package_share_directory("nanotec_pd4_can"),
                "config",
                "double-pd4",
                "master.dcf",
            ),
            "master_bin": os.path.join(
                get_package_share_directory("nanotec_pd4_can"),
                "config",
                "double-pd4",
                "master.bin",
            ),
            "bus_config": os.path.join(
                get_package_share_directory("nanotec_pd4_can"),
                "config",
                "double-pd4",
                "bus.yml",
            ),
            "can_interface_name": "can0",
        }.items(),
    )

    ld.add_action(device_container)

    return ld
`

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Motor: PD4-CB59M024035-E-08

lucasmluza, 2023-10-05:
The "/target" is a service, just like the "/init", and "/halt".  You can check if it is available by running:

```bash
ros2 service list
```

If so, the "/target" may be set, for example, by:

```bash
ros2 service call /motor_left/target canopen_interfaces/srv/COTargetDouble "{ target: 10.0 }"
```

VerlooyNick, 2023-10-05:
Indeed, that works. But for integration into a robot system working with the service does not seem the right way. Image I have a joystick creating a velocity command which I want to apply to the motor. I’d assume I could publish to a certain topic to have a more real time behaviour instead of working with services.

ipa-cmh, 2023-10-06:
This is where you would use the ros2_control interface that provides you with all kind of streaming options.
Though I understand that the setup is a bit more difficult, we have a short but probably not optimal example here:
https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-robot-system.html

jclinton830, 2023-10-18:
> This is where you would use the ros2_control interface that provides you with all kind of streaming options. Though I understand that the setup is a bit more difficult, we have a short but probably [...]

@ipa-cmh Have you tested this robot system approach on real hardware?

VerlooyNick, 2023-10-18:
Right now I'm working with the proxy driver and writing my own Cia402 application layer. 
Is the [documentation for the Cia402](https://ros-industrial.github.io/ros2_canopen/manual/humble/user-guide/cia402-driver.html) driver correct? This states that the /target is a subscriber but it only appears as a service?

ipa-cmh, 2023-10-18:
@jclinton830 we are using it on the prbt robot (6 axis arm) with interpolated position mode. Also just created a system for a customer with Maxon hardware.

ipa-cmh, 2023-10-18:
> @ipa-cmh Have you tested this robot system approach on real hardware?

@jclinton830 we have the driver running on pilz prbt robot (6 axis robot arm). Apart from that we have just created a solution for a customer on Maxon hardware, this has pointed out some problems with the sdo timeouts.

millejade, 2024-08-02:
Hello!

I want to add some questions about that, I also saw in the documentation a subscriber `/target`.
May I know why it is in the documentation but not on the ROS2 package?
If this subscriber was in the ROS2 package before and was later removed, what implicates to take this action?

Thank you!