ros-industrial/ros2_canopen #203: rpdo can not be control

Mark9999999, 2023-10-12:
**Describe the bug**
I want to use ros2_canopen to maxon motor, when I set a rpdo mapping to index:6040 subindex:00  and set a value 0xF, It is not 0xF , but is 0x21BF in my motor drive.

**cia402_setup.launch.py**
```

import os
from ament_index_python import get_package_share_directory
from launch import LaunchDescription
import launch
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    ld = LaunchDescription()
    slave_eds_path = os.path.join(
        get_package_share_directory("canopen_tests"), "config", "cia402", "cia402_slave.eds"
    )

    slave_node_1 = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(get_package_share_directory("canopen_fake_slaves"), "launch"),
                "/cia402_slave.launch.py",
            ]
        ),
        launch_arguments={
            "node_id": "1",
            "node_name": "cia402_node_1",
            "slave_config": slave_eds_path,
        }.items(),
    )
    master_bin_path = os.path.join(
        get_package_share_directory("canopen_tests"),
        "config",
        "cia402_lifecycle",
        "master.bin",
    )
    if not os.path.exists(master_bin_path):
        master_bin_path = ""
    device_container = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(get_package_share_directory("canopen_core"), "launch"),
                "/canopen.launch.py",
            ]
        ),
        launch_arguments={
            "master_config": os.path.join(
                get_package_share_directory("canopen_tests"),
                "config",
                "cia402",
                "master.dcf",
            ),
            "master_bin": master_bin_path,
            "bus_config": os.path.join(
                get_package_share_directory("canopen_tests"),
                "config",
                "cia402",
                "bus.yml",
            ),
            "can_interface_name": "can0",
        }.items(),
    )

    ld.add_action(device_container)
    #ld.add_action(slave_node_1)
    return ld
```

**bus.yml**
 
options:
  dcf_path: "@BUS_CONFIG_PATH@"

```
master:
  node_id: 2
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  sync_period: 20000 #us

defaults:
  dcf: "cia402_slave.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  period: 20
  polling: false
  heartbeat_producer: 1000 # Heartbeat every 1000 ms控制701报文间隔
  sdo:
    - {index: 0x6060, sub_index: 0, value: 1} 
    - {index: 0x6040, sub_index: 0, value: 6} 
    - {index: 0x6040, sub_index: 0, value: 7}
    - {index: 0x6081, sub_index: 0, value: 1000}
    - {index: 0x6083, sub_index: 0, value: 5000}
    - {index: 0x6084, sub_index: 0, value: 10000}
    - {index: 0x6085, sub_index: 0, value: 10000}
    - {index: 0x6085, sub_index: 0, value: 10000}
    - {index: 0x6040, sub_index: 0, value: 15}
    - {index: 0x607a, sub_index: 0, value: 10000}
    #- {index: 0x6040, sub_index: 0, value: 95}
    #- {index: 0x6040, sub_index: 0, value: 15}
    #- {index: 0x6040, sub_index: 0, value: 95}
  tpdo:
    1:
      enabled: false
    2:
      enabled: false
    3:
      enabled: false
    4:
      enabled: false
  rpdo: 
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0, value: 15}
    2:
      enabled: false
    3:
      enabled: false
    4:
      enabled: false
     
nodes:
  cia402_device_1:
    node_id: 1
```

**sub1600 of rpdo1 in eds**
```

[1600sub0]
ParameterName=Number of mapped objects in RxPDO 1
ObjectType=0x7
DataType=0x5
AccessType=rw
DefaultValue=1
LowLimit=0
HighLimit=12
PDOMapping=0
ObjFlags=0

[1600sub1]
ParameterName=1st mapped object in RxPDO 1
ObjectType=0x7
DataType=0x7
AccessType=rw
DefaultValue=0x60400010
PDOMapping=0
ObjFlags=0
```

ipa-cmh, 2023-10-13:
@Mark9999999 the driver will deal with 6040 on its own. You do not need to set it. Otherwise if you want to set it use the proxy driver.

ZeroErrControl, 2025-01-25:
You can refer to [this project:https://github.com/ZeroErrControl/eRob_CANopen_Python](https://github.com/ZeroErrControl/eRob_CANopen_Python).