ros-industrial/ros2_canopen #204: Stuck on "Wait for device to boot."

jitaxis, 2023-10-12:
I am running a basic lunch file that launches the /canopen.launch.py file. I am using a KINCO motor driver with the .eds coming from the supplier.
When I run the file, the terminal gets stuck on:
 [kinco_124S]: Wait for device to boot.

If I try to use the nmt_start service I get the following:
[kinco_124S]: Could not start device via NMT because driver not activated.

I am using ROS2 Humble on Ubuntu 22.04

I feel like I must be making a basic mistake but I am just unsure of what I am doing wrong.
Any help is greatly appreciated, and let me know if there is anything required. Thank you.

My bus.yaml and .eds file:
[can_driver_config.zip](https://github.com/ros-industrial/ros2_canopen/files/12876525/can_driver_config.zip)

My launch file:
def generate_launch_description():
    ld = LaunchDescription()
    master_bin_path = os.path.join(
        get_package_share_directory("bhm_can_interface"),
        "config",
        "single-kinco-fd124s",
        "master.bin",
    )

    if not os.path.exists(master_bin_path):
        master_bin_path = ""

    device_container = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(get_package_share_directory("canopen_core"), "launch"),
                "/canopen.launch.py",
            ]
        ),
        launch_arguments={
            "master_config": os.path.join(
                get_package_share_directory("bhm_can_interface"),
                "config",
                "single-kinco-fd124s",
                "master.dcf",
            ),
            "master_bin": master_bin_path,
            "bus_config": os.path.join(
                get_package_share_directory("bhm_can_interface"),
                "config",
                "single-kinco-fd124s",
                "bus.yml",
            ),
            "can_interface_name": "can0",
        }.items(),
    )

    ld.add_action(device_container)

    return ld

ipa-cmh, 2023-10-12:
This could be related to a failed start-up due to missing sdo timeouts. We are currently working on a patch. Some CANopen devices do not respond within 100ms.

jitaxis, 2023-10-12:
Do you know how long until that patch is complete?
Also if I understand correctly the candump shows the following:
Master node Heartbeat
NMT reset communication
Slave Node Heartbeat
Slave Node RX
Slave Node TX
...
So it looks like there is communication between the driver and my PC.
Would it help if I provided a timestamped candump?

ipa-cmh, 2023-10-13:
@jitaxis is that everything or is there more communication Happening?

jitaxis, 2023-10-13:
Yeah thats all that happens.

yuriyfedyuk, 2023-11-24:
> This could be related to a failed start-up due to missing sdo timeouts. We are currently working on a patch. Some CANopen devices do not respond within 100ms.

We faced a similar issue. Could you please provide an estimation of when we can expect this patch? Or you can point us to a place where it should be implemented and we will be more than happy to fix that.

ipa-cmh, 2023-11-24:
#220 does add this feature. As you can see there are still a couple of changes to do.

lvjinxin-l, 2024-03-28:
Hello,where is the configuration of sdo and pdo in your bus.yml file?
How did you get these packets without configuration?
(000.002108) can0 601 [8] 40 00 10 00 00 00 00 00
(000.003869) can0 581 [8] 43 00 10 00 92 01 02 00
(000.004002) can0 601 [8] 40 18 10 01 00 00 00 00
(000.005613) can0 581 [8] 43 18 10 01 00 03 00 00
(000.005709) can0 601 [8] 40 18 10 02 00 00 00 00
(000.007415) can0 581 [8] 43 18 10 02 44 46 00 00
@jitaxis

remco-r, 2024-07-05:
I have a similar issue (humble, ubuntu22, 1dd817c67c6b1b1dee53e4851f1bc3c391d6ff43)

bus setup:

```
master:
  node_id: 1
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  # baudrate: The baudrate in kbit/s (default 1000)
  # vendor_id:The vendor-ID (default 0x00000000)
  # product_code:The product code (default 0x00000000)
  # revision_number: The revision number (default 0x00000000).
  # serial_number:  The serial number (default 0x00000000).
  # heartbeat_multiplier: The multiplication factor used to obtain the slave heartbeat consumer time from the master heartbeat producer time (default see options section).
  # heartbeat_consumer: Specifies whether the master should monitor the heartbeats of the slaves (default true).
  # heartbeat_producer: The heartbeat producer time in ms (default 0).
  # emcy_inhibit_time: The EMCY inhibit time in multiples of 100 μs (default 0, see object 1015).
  sync_period: 10000 # The SYNC interval in μs (default 0).
  sync_window: 10000 # The SYNC window length in μs (default 0, see object 1007).
  # sync_overflow: The SYNC counter overflow value (default 0, see object 1019).
  # error_behavior: A dictionary of error behaviors for different classes or errors (default {1: 0x00}, see object 1029).
  nmt_inhibit_time: 10000 # The NMT inhibit time in multiples of 100 μs (default 0, see object 102A).
  # start: Specifies whether the master shall switch into the NMT operational state by itself (default true, see bit 2 in object 1F80).
  # start_nodes: Specifies whether the master shall start the slaves (default true, see bit 3 in object 1F80).
  start_all_nodes: true # Specifies whether the master shall start all nodes simultaneously (default false, see bit 1 in object 1F80).
  # reset_all_nodes: Specifies whether all slaves shall be reset in case of an error event on a mandatory slave (default false, see bit 4 in object 1F80).
  # stop_all_nodes: Specifies whether all slaves shall be stopped in case of an error event on a mandatory slave (default false, see bit 6 in object 1F80).
  boot_time: 1000 # The timeout for booting mandatory slaves in ms (default 0, see object 1F89).
  boot_timeout: 2000 # The timeout for booting all slaves in ms (default 2000ms).

radar_left:
  node_id: 10
  dcf: "radar_sensor.eds"
  driver: "ros2_canopen::ProxyDriver"
  package: "canopen_proxy_driver"
```

Launch file output:
```
[device_container_node-1] [INFO] [1720181221.407411051] [radar_left]: eds file ***.eds
[device_container_node-1] [INFO] [1720181221.407448612] [radar_left]: bin file ***.bin
[device_container_node-1] Found tpdo mapped object: index=2000 subindex=1
[device_container_node-1] Found tpdo mapped object: index=2000 subindex=2
[device_container_node-1] Found tpdo mapped object: index=2000 subindex=3
[device_container_node-1] Found tpdo mapped object: index=2000 subindex=5
[device_container_node-1] [WARN] [1720181221.409308197] [radar_left]: Wait for device to boot.
```

On the canbus i see the master resetting the bus, the radar sensor replying with boot message, but there its stuck.
```
 (071.342994)  can0  TX - -  701   [1]  00
 (000.000048)  can0  TX - -  000   [2]  82 00
 (000.002481)  can0  RX - -  710   [1]  00
```

Notes:
* boot message comes within 2.5 ms as is visible.
* Manually posting node 10 boot message with cansend also does not trigger continuation.
* Using debug prints figured that the can driver does load the correct node id.
* Somehow the master does respond to boot messages of other nodes that i manually send using `cansend`.
  * `cansend can0 702#00`, and `cansend can0 703#00`
  * both node 2 and 3 are not part of my bus layout.
* with manual canopen commands i can fully operate the device
* when i configure the node (and bus layout) to have node id 2, the launch console is still stuck in boot, but candump comes further:
```
  (061.879969)  can0  TX - -  701   [1]  00
 (000.000024)  can0  TX - -  000   [2]  82 00
 (000.002672)  can0  RX - -  702   [1]  00
 (000.000142)  can0  TX - -  602   [8]  40 00 10 00 00 00 00 00
 (000.002866)  can0  RX - -  582   [8]  43 00 10 00 00 00 00 00
 (000.000068)  can0  TX - -  602   [8]  40 18 10 01 00 00 00 00
 (000.002934)  can0  RX - -  582   [8]  43 18 10 01 AD 00 00 00
```
* if i now manually send another startup `cansend can0 702#00` command, "driver booted and ready" is printed

chinaheyu, 2024-07-13:
I'm facing a similar problem with @remco-r.

I am running the latest branch on ubuntu 22.04, humble (https://github.com/ros-industrial/ros2_canopen/commit/1dd817c67c6b1b1dee53e4851f1bc3c391d6ff43), with the following `bus.yml`.

```yml
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 1
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"

defaults:
  dcf: "ZeroErr Driver_V1.5.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  enable_lazy_load: false
  sdo:
    - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
    - {index: 0x6081, sub_index: 0, value: 1000}
    - {index: 0x6083, sub_index: 0, value: 2000}
  tpdo: # TPDO needed statusword, actual velocity, actual position, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6061, sub_index: 0} # mode of operation display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0} # position actual value
        - {index: 0x606c, sub_index: 0} # velocity actual position
    3:
      enabled: false
    4:
      enabled: false
  rpdo: # RPDO needed controlword, target position, target velocity, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0} # controlword
      - {index: 0x6060, sub_index: 0} # mode of operation
    2:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x607A, sub_index: 0} # target position
      - {index: 0x60FF, sub_index: 0} # target velocity

nodes:
  cia402_device_1:
    node_id: 3
```

I made the following attempt to run the Cia402Driver.

1. Runs directly on the physical interface;
2. Runs on a virtual interface and emulates the device via cansend.

## 1. Running on candleLight USB-CAN Adapter (can0)

```
$ ros2 launch humanoid_canopen humanoid_canopen.launch.py 
[... 84 lines cut ...]
[device_container_node-1] [INFO] [1720861371.848897783] [canopen_402_driver]: Fault reset
```

The output of candump is as follows.

```
$ candump vcan0
  vcan0  701   [1]  00
  vcan0  000   [2]  82 00
  vcan0  703   [1]  00
  ↑ send 703#00 manually via cansend.
  vcan0  603   [8]  40 00 10 00 00 00 00 00
  vcan0  603   [8]  80 00 10 00 00 00 04 05
  vcan0  603   [8]  40 00 10 00 00 00 00 00
  vcan0  603   [8]  80 00 10 00 00 00 04 05
  vcan0  603   [8]  40 00 10 00 00 00 00 00
  vcan0  583   [8]  43 00 10 00 92 01 02 04
  ↑ repeat until recived 583#4300100092010204 from cansend.
  vcan0  203   [3]  56 65 00
  vcan0  203   [3]  D6 65 00
  vcan0  203   [3]  D6 65 00
  ↑ fast and repetitive output
```

The weird thing is that on the physical interface, even though the device sends 583#4300100092010204, the program is still stuck on "Wait for device to boot.", even though I send it again via cansend. But if I manually send the 703#00, the program continues to run, just like the virtual interface.