ros-industrial/ros2_canopen #206: System of mixed proxy controllers and CiA402 controllers

gsalinas, 2023-10-16:
Hi,

Is there currently a way to configure a bus that uses a mixture of ProxyDrivers (or, a custom driver I've written that's derived from a proxy driver) and Cia402Drivers? I have a system with some sensors that use a customized version of a proxy driver and some Cia402 motion devices, and I'm not sure what the intended way to use all of those together would be.

Thank you for all your help!

ipa-cmh, 2023-10-16:
If you use the service interface, it should work, though I haven't tested it extensively. ros2 control implementation not support it yet.

jclinton830, 2023-10-16:
This would be a great feature to have. I am currently working on a robot system where there are four motors that support CIA402 and four with CIA302 support only. 

@ipa-cmh is it possible to release a basic tutorial on how to do this?

phinners, 2023-11-14:
I just came across this question and was wondering if there has been some development in this direction in the meantime? @ipa-cmh @gsalinas @jclinton830

gsalinas, 2023-11-14:
For my project, it was easier to add a second physical CAN bus and run one master on can0 with all the 402 devices and a second master on can1 with everything else; obviously that may not be an option for you but good luck!

jclinton830, 2023-11-14:
For our project, we ended up just running all the motor controllers in 301 profile mode and using the proxy controller instead of the cia402 controller. That way we have some sort of consistency.

eholandcmu, 2024-09-06:
Hi, has there been any further work to support mixed proxy and cia402 drivers?