ros-industrial/ros2_canopen #209: What is the equivalent of socketCan_bridge in the ros1's ros_canopen package?

dayeongg084TW, 2023-10-17:
**Setup:**
 - Device: 
 - OS: ubuntu 22.04
 - ROS-Distro:  humble 
 - Branch/Commit: humble or master 

**Additional context**
I created and used a 'launch file' in the socketCan_bridge of ros_canopen package in ros1.  (I created a file to use can0 and can1)
The ros2_canopen driver appears to have a different structure from the existing ros1.
What acts as SocketCan_Bridge in ros2?
If there is no related package, where should I add the launch file?

Also what is the right branch for ros2 Humble?

thank you.

ipa-cmh, 2023-10-18:
This report only provides CANopen interface, no pure CAN interface. If you need the latter, please use https://github.com/autowarefoundation/ros2_socketcan.

leonce-m, 2023-10-22:
I also have a case where I need to control some actuator via canopen_ros2_control/Cia402System and other hardware via regular can messages. Is it possible to have canopen and ros2_socketcan access the same can interface?

Otherwise I would support the suggestion to have an equivalent of socketcan_bridge for ros2.

TZECHIN6, 2023-11-21:
I am very new to CANopen, as CANopen replies on CAN physical layer, whats the benefit of using this package, instead of using `ros2socketcan` to send out pure CAN frame base on the OD chart of the CANopen driver?

I am struggling to choose between ros2socketcan and ros2_can_open....