ros-industrial/ros2_canopen #210: Failed init services when running device_container_node in a two core computer

POtaoPereiraRosen, 2023-10-17:
**Describe the bug**
Failed init services when running device_container_node in a two core computer

**To Reproduce**
When launching device_container_node (canopen_core/launch/canopen.launch.py)
- in a two core computer
- with two motor nodes
- when calling the init service of these two nodes simultaneously
Then:
- both services fail (after a timeout of 5 seconds, they return success=false)
- the publishers of the topic joint_states, for both motors, do not publish any data for 5 seconds (after the call of the init services)

**Expected behavior**
This behaviour does not exist when device_container_node is launched in a virtual machine with 8 processors.
But our target platform is a computer with 2 cores, and there the faulty behaviour (described above) occurs.
The faulty behaviour also occurs, in a virtual machine setup with 2 processors.

**Setup:**
 - Device: Virtual machine with 2 processors or a computer with two cores
 - OS: Ubuntu 22
 - ROS-Distro: Humble

ipa-cmh, 2023-10-18:
@POtaoPereiraRosen this could be because three threads are necessary for this. Does it work if you initialise sequencially?

POtaoPereiraRosen, 2023-10-18:
@ipa-cmh Yes, when starting sequentially, it works (i.e., if I wait for the reply of the init-service of motor 1, and then call the init-service of motor 2, then it works). In your question, you referred to the need of three threads: I am guessing you meant hardware-threads/cores/CPUs/processors. Out of curiosity, why are three cores required?

ipa-cmh, 2023-11-22:
One thread runs the canopen loop, an one per driver.

closed 2023-11-28 by POtaoPereiraRosen