ros-industrial/ros2_canopen #212: How to use canopen_ros2_control and canopen_ros2_controllers

Mark9999999, 2023-10-18:
How to use canopen_ros2_control and canopen_ros2_controllers in real  motor such as maxon？
what is the step of configuration?

ipa-cmh, 2023-10-18:
Please refer to the example here:
https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-robot-system.html

And use:
https://github.com/ipa-cmh/prbt_robot

As guidance.

Mark9999999, 2023-10-26:
> Please refer to the example here: https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-robot-system.html
> 
> And use: https://github.com/ipa-cmh/prbt_robot
> 
> As guidance.
Now, my computer has setted socketcan in ubuntu 22.04.
How to write haraware_interface ,if my real motor use canopen by ros2_control.