ros-industrial/ros2_canopen #215: How to create a cia301 system that works with real hardware?

jclinton830, 2023-10-25:
**Describe the bug**
I have been following this tutorial https://ros-industrial.github.io/ros2_canopen/manual/rolling/software-tests/ros2_control_system-test.html after setting up a canopen cia301 system by following this tutorial https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-cia301-system.html

I am working with real hardware instead of using the virtual slave. My motor controller of choice is an MBL1660 from roboteq. 

I have the drivers booting up fine. See the below output.

```
jerome@jjustin:~/gh_ws$ ros2 launch canopen_tests canopen_system.launch.py 
[INFO] [launch]: All log files can be found below /home/jerome/.ros/log/2023-10-25-15-55-14-103207-jjustin-43872
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [43875]
[INFO] [robot_state_publisher-2]: process started with pid [43877]
[INFO] [spawner-3]: process started with pid [43879]
[INFO] [spawner-4]: process started with pid [43881]
[robot_state_publisher-2] [INFO] [1698209714.346875843] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1698209714.346977568] [robot_state_publisher]: got segment node_1_link
[robot_state_publisher-2] [INFO] [1698209714.346988652] [robot_state_publisher]: got segment node_2_link
[ros2_control_node-1] [INFO] [1698209714.347555741] [resource_manager]: Loading hardware 'canopen_test_system' 
[ros2_control_node-1] [INFO] [1698209714.355711862] [resource_manager]: Initialize hardware 'canopen_test_system' 
[ros2_control_node-1] [INFO] [1698209714.355931841] [CanopenSystem]: bus_config: '/home/jerome/gh_ws/install/canopen_tests/share/canopen_tests/config/canopen_system/bus.yml'
[ros2_control_node-1] [INFO] [1698209714.355938998] [CanopenSystem]: master_config: '/home/jerome/gh_ws/install/canopen_tests/share/canopen_tests/config/canopen_system/master.dcf'
[ros2_control_node-1] [INFO] [1698209714.355942311] [CanopenSystem]: can_interface_name: 'can0'
[ros2_control_node-1] [INFO] [1698209714.355945110] [CanopenSystem]: master_bin: '""'
[ros2_control_node-1] [INFO] [1698209714.355951352] [resource_manager]: Successful initialization of hardware 'canopen_test_system'
[ros2_control_node-1] [INFO] [1698209714.356157896] [resource_manager]: 'configure' hardware 'canopen_test_system' 
[ros2_control_node-1] [INFO] [1698209714.363262530] [device_container]: Starting Device Container with:
[ros2_control_node-1] [INFO] [1698209714.363397676] [device_container]:          can_interface_name can0
[ros2_control_node-1] [INFO] [1698209714.363413160] [device_container]:          master_config /home/jerome/gh_ws/install/canopen_tests/share/canopen_tests/config/canopen_system/master.dcf
[ros2_control_node-1] [INFO] [1698209714.363425275] [device_container]:          bus_config /home/jerome/gh_ws/install/canopen_tests/share/canopen_tests/config/canopen_system/bus.yml
[ros2_control_node-1] [INFO] [1698209714.364735438] [device_container]: Loading Master Configuration.
[ros2_control_node-1] [INFO] [1698209714.365772741] [device_container]: Load Library: /home/jerome/gh_ws/install/canopen_master_driver/lib/libmaster_driver.so
[ros2_control_node-1] [INFO] [1698209714.368552317] [device_container]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[ros2_control_node-1] [INFO] [1698209714.368620366] [device_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[ros2_control_node-1] [INFO] [1698209714.373982083] [master]: NodeCanopenBasicMaster
[ros2_control_node-1] [INFO] [1698209714.374199165] [device_container]: Load master component.
[ros2_control_node-1] [INFO] [1698209714.374434472] [device_container]: Added /master to executor
[ros2_control_node-1] [INFO] [1698209714.391554212] [device_container]: Loading Driver Configuration.
[ros2_control_node-1] [INFO] [1698209714.391665483] [device_container]: Found device proxy_device_1 with driver ros2_canopen::ProxyDriver
[ros2_control_node-1] [INFO] [1698209714.391866853] [device_container]: Load Library: /home/jerome/gh_ws/install/canopen_proxy_driver/lib/libproxy_driver.so
[ros2_control_node-1] [INFO] [1698209714.392575411] [device_container]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[ros2_control_node-1] [INFO] [1698209714.392589070] [device_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::ProxyDriver>
[ros2_control_node-1] [INFO] [1698209714.396614839] [device_container]: Load driver component.
[ros2_control_node-1] [INFO] [1698209714.396822619] [device_container]: Added /proxy_device_1 to executor
[ros2_control_node-1] [ERROR] [1698209714.405117995] [proxy_device_1]: Could not polling from config, setting to true.
[ros2_control_node-1] [ERROR] [1698209714.405265791] [proxy_device_1]: Could not read period from config, setting to 10ms
[ros2_control_node-1] [ERROR] [1698209714.405309590] [proxy_device_1]: Could not read enable diagnostics from config, setting to false.
[ros2_control_node-1] [INFO] [1698209714.407074469] [proxy_device_1]: eds file /home/jerome/gh_ws/install/canopen_tests/share/canopen_tests/config/canopen_system/roboteq_motor_controllers_v80.eds
[ros2_control_node-1] [INFO] [1698209714.407100394] [proxy_device_1]: bin file /home/jerome/gh_ws/install/canopen_tests/share/canopen_tests/config/canopen_system/proxy_device_1.bin
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=9
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=a
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=b
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=c
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=d
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=e
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=f
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=10
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=1
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=2
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=3
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=4
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=5
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=6
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=7
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=8
[ros2_control_node-1] [WARN] [1698209714.415358676] [proxy_device_1]: Wait for device to boot.
[ros2_control_node-1] [ERROR] [1698209714.527408651] [proxy_device_1]: Boot Issue: Node guarding event. No confirmation for guarding request received from CANopen device.
[ros2_control_node-1] [INFO] [1698209714.527482876] [proxy_device_1]: Driver booted and ready.
[ros2_control_node-1] [INFO] [1698209714.527759910] [proxy_device_1]: Starting with polling mode.
[ros2_control_node-1] [INFO] [1698209714.528155355] [CanopenSystem]: Number of registered drivers: '1'
[... 55 lines cut ...]
data: 0
---
index: 8454
subindex: 4
data: 0
---
index: 8454
subindex: 4
data: 0

```

My question is, have you tested this on real hardware? Am I missing something? I also see in my launch output the following warning 

`
[joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
`

Is there any additional setup one must do for this to work with real hardware?

**Setup:**
 - Device: Laptop
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: humble

ipa-cmh, 2023-10-27:
Can you check the candump? Is the tpdo sent out? 
Did you map 0x2005 as rpdo and tpdo on your device? Does 0x2005 exist on your device? Seems your device currently only has 0x2106 mapped as pdo.

jclinton830, 2023-11-06:
@ipa-cmh you are right. By default the device was mapping all the sub indexes of 0x2106 to all of its TPDOs and all the sub indexes fo object 0x2005 was being mapped to all of its RPDOs.

[image]

I used the remapping keyword to apply the remappings for all the tpdos and rpdos. See below for the bus.yml 

```
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 1
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  baudrate: 1000

proxy_device_1:
  node_id: 12
  dcf: "roboteq_motor_controllers_v80.eds"
  driver: "ros2_canopen::ProxyDriver"
  package: "canopen_proxy_driver"
  polling: true
  diagnostics:
    enable: true
```
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 1
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  baudrate: 1000

proxy_device_1:
  node_id: 12
  dcf: "roboteq_motor_controllers_v80.eds"
  driver: "ros2_canopen::ProxyDriver"
  package: "canopen_proxy_driver"
  polling: true
  diagnostics:
    enable: true
    period: 1000 # in milliseconds
  reset_communication: false
  tpdo:
    1:
      enabled: true
      transmission: 0xFF
      cob_id: "auto"
      mapping:
        - { index: 0x2100, sub_index: 1 } # read motor amps
        - { index: 0x2101, sub_index: 1 } # read actual motor command
    2:
      enabled: true
      transmission: 0xFF
      cob_id: "auto"
      mapping:
        - { index: 0x2102, sub_index: 1 } # Read Applied Power Level
        - { index: 0x2103, sub_index: 1 } # read encoder motor speed
    3:
      enabled: true
      transmission: 0xFF
      cob_id: "auto"
      mapping:
        - { index: 0x2104, sub_index: 1 } 
        - { index: 0x2105, sub_index: 1 } 
    4:
      enabled: true
      transmission: 0xFF
      cob_id: "auto"
[... 177 lines cut ...]
index: 8450
subindex: 1
data: 0
---
index: 8451
subindex: 1
data: 0
---
index: 8448
subindex: 1
data: 0
---
index: 8449
subindex: 1
data: 0
---
index: 8450
subindex: 1
data: 0
---
index: 8451
subindex: 1
data: 0
---
```

ipa-cmh, 2023-11-07:
@jclinton830 can you try with a sync? We have not really tested async mode for pdos (we just do not have the use case) usually with drives you want fixed periodic commands and feedback to be able to control them reliably.
Generally if you have async mode enabled for PDOs they are only published on change. That can be pretty misleading when testing. Try to set ``transmission: 0x01`` for each pdo and for master set ``sync_period: 20000`` in bus.yml. All pdos and rpdos should then be send every 20 ms.

Then use ``candump can0`` to monitor the bus.
Can you then send me a the candump  log?

ipa-cmh, 2023-11-09:
@jclinton830 I will see if I can setup a simulation with fake slaves today. With regards to the candump, the startup sequence is most interesting as well as a section during which you sent pdo.

jclinton830, 2023-11-13:
@ipa-cmh any update on this?

jclinton830, 2023-11-22:
@ipa-cmh any chance you can shed some light on what is going on with this setup? 

I still have a few questions:

when I do ros2 topic echo /node_1_controller/rpdo I only ever see one RPDO being reported by the publisher. See below. If this is normal behaviour then my question is what makes the canopen_proxy_controller decide which one of the 4 rpdos to publish?

```
~/gh_ws$ ros2 topic echo /node_1_controller/rpdo 
index: 8451
subindex: 1
data: 0
---
index: 8451
subindex: 1
data: 0
---
index: 8451
subindex: 1
data: 0
---
index: 8451
subindex: 1
data: 0
---
index: 8451
subindex: 1
data: 0
---
```

closed 2024-08-12 by jclinton830