ros-industrial/ros2_canopen #219: ros2_control launch error

alexanderjyuen, 2023-11-03:
At present we are getting the following error when we start our launch file:

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-11-03-00-33-13-233571-polymath-MINIPC-PN50-E1-5019
[INFO] [launch]: Default logging verbosity is set to INFO
<launch.substitutions.path_join_substitution.PathJoinSubstitution object at 0x7f43c1b32e30>
[INFO] [ros2_control_node-1]: process started with pid [5031]
[INFO] [robot_state_publisher-2]: process started with pid [5033]
[INFO] [spawner-3]: process started with pid [5035]
[INFO] [cia402_slave_node-4]: process started with pid [5037]
[INFO] [cia402_slave_node-5]: process started with pid [5039]
[INFO] [spawner-6]: process started with pid [5041]
[INFO] [spawner-7]: process started with pid [5043]
[INFO] [spawner-8]: process started with pid [5045]
[ros2_control_node-1] [INFO] [1698971593.578500513] [resource_manager]: Loading hardware 'canopen_test_system' 
[robot_state_publisher-2] [INFO] [1698971593.584175046] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1698971593.584347007] [robot_state_publisher]: got segment node_1_link
[robot_state_publisher-2] [INFO] [1698971593.584357747] [robot_state_publisher]: got segment node_2_link
[ros2_control_node-1] [INFO] [1698971593.599997164] [resource_manager]: Initialize hardware 'canopen_test_system' 
[ros2_control_node-1] [INFO] [1698971593.602416550] [CanopenSystem]: bus_config: '/colcon_ws/install/hal/share/hal/config/can/bus.yml'
[ros2_control_node-1] [INFO] [1698971593.603843404] [CanopenSystem]: master_config: '/colcon_ws/install/hal/share/hal/config/can/master.dcf'
[ros2_control_node-1] [INFO] [1698971593.605540072] [CanopenSystem]: can_interface_name: 'can0'
[ros2_control_node-1] [INFO] [1698971593.607105314] [CanopenSystem]: master_bin: '""'
[ros2_control_node-1] [INFO] [1698971593.608235974] [resource_manager]: Successful initialization of hardware 'canopen_test_system'
[ros2_control_node-1] [INFO] [1698971593.610293145] [resource_manager]: 'configure' hardware 'canopen_test_system' 
[ros2_control_node-1] [INFO] [1698971593.669473035] [device_container]: Starting Device Container with:
[ros2_control_node-1] [INFO] [1698971593.669641649] [device_container]:      can_interface_name can0
[ros2_control_node-1] [INFO] [1698971593.669658641] [device_container]:      master_config /colcon_ws/install/hal/share/hal/config/can/master.dcf
[ros2_control_node-1] [INFO] [1698971593.669672367] [device_container]:      bus_config /colcon_ws/install/hal/share/hal/config/can/bus.yml
[ros2_control_node-1] [INFO] [1698971593.672908218] [device_container]: Loading Master Configuration.
[ros2_control_node-1] [INFO] [1698971593.674138174] [device_container]: Load Library: /opt/polymathrobotics/canopen_master_driver/lib/libmaster_driver.so
[ros2_control_node-1] [INFO] [1698971593.677489640] [device_container]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[ros2_control_node-1] [INFO] [1698971593.677575290] [device_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[ros2_control_node-1] [INFO] [1698971593.690287261] [master]: NodeCanopenBasicMaster
[ros2_control_node-1] [INFO] [1698971593.690548188] [device_container]: Load master component.
[ros2_control_node-1] [INFO] [1698971593.690794989] [device_container]: Added /master to executor
[ros2_control_node-1] [INFO] [1698971593.717157294] [device_container]: Loading Driver Configuration.
[ros2_control_node-1] [INFO] [1698971593.717342279] [device_container]: Found device left_motor with driver ros2_canopen::Cia402Driver
[ros2_control_node-1] [INFO] [1698971593.717643441] [device_container]: Load Library: /opt/polymathrobotics/canopen_402_driver/lib/libcia402_driver.so
[ros2_control_node-1] [INFO] [1698971593.718779011] [device_container]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::Cia402Driver>
[ros2_control_node-1] [INFO] [1698971593.718849753] [device_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::Cia402Driver>
[ros2_control_node-1] [INFO] [1698971593.733628863] [device_container]: Load driver component.
[ros2_control_node-1] [INFO] [1698971593.734148504] [device_container]: Added /left_motor to executor
[ros2_control_node-1] [ERROR] [1698971593.765973706] [left_motor]: Could not read enable diagnostics from config, setting to false.
[ros2_control_node-1] [INFO] [1698971593.766326916] [left_motor]: scale_pos_to_dev_ 1000.000000
[ros2_control_node-1] scale_pos_from_dev_ 0.001000
[ros2_control_node-1] scale_vel_to_dev_ 67.000000
[ros2_control_node-1] scale_vel_from_dev_ 0.015000
[ros2_control_node-1] 
[ros2_control_node-1] [INFO] [1698971593.769431922] [left_motor]: eds file /colcon_ws/install/hal/share/hal/config/can/roboteq_motor_controllers_v80.eds
[ros2_control_node-1] [INFO] [1698971593.769498717] [left_motor]: bin file /colcon_ws/install/hal/share/hal/config/can/left_motor.bin
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=9
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=a
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=b
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=c
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=d
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=e
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=f
[ros2_control_node-1] Found rpdo mapped object: index=2005 subindex=10
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=1
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=2
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=3
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=4
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=5
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=6
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=7
[ros2_control_node-1] Found tpdo mapped object: index=2106 subindex=8
[ros2_control_node-1] [INFO] [1698971593.788046984] [left_motor]: Driver booted and ready.
[ros2_control_node-1] [INFO] [1698971593.788746319] [left_motor]: Starting with polling mode.
[ros2_control_node-1] [INFO] [1698971593.790816604] [device_container]: Found device right_motor with driver ros2_canopen::Cia402Driver
[... 379 lines cut ...]
    left_motor_controller:
      type: canopen_ros2_controllers/Cia402DeviceController

    right_motor_controller:
      type: canopen_ros2_controllers/Cia402DeviceController

    forward_position_controller:
      type: forward_command_controller/ForwardCommandController

left_motor_controller:
  ros__parameters:
    joint: left_motor

right_motor_controller:
  ros__parameters:
    joint: right_motor

forward_position_controller:
  ros__parameters:
    joints:
      - left_motor
      - right_motor
    interface_name: velocity
```
I am able to run the launch file with a virtual can but it breaks when I try to communicate with a real can interface, anybody have any insight into this?

ipa-cmh, 2023-11-03:
Seems you are starting the fake slaves while on real hardware. That does not work.

alexanderjyuen, 2023-11-03:
> Seems you are starting the fake slaves while on real hardware. That does not work.

Is there a ready made launch file that would work for a real CAN device using 402?

closed 2023-11-03 by alexanderjyuen