ros-industrial/ros2_canopen #223: ci402_setup.launch.py' was not found when running example

SauceQuiPik, 2023-11-06:
Hi,

Having issue to run example. I'm new to all of this (I'm an automation developer)

**Describe the bug**
When i want to run the example :
 ros2 launch canopen_tests ci402_setup.launch.py
I have an error : 
file 'ci402_setup.launch.py' was not found in the share directory of package 'canopen_tests' which is at '/home/canopen_ws/install/canopen_tests/share/canopen_tests'

**To Reproduce**
ROS 2 is running in a container from this image : 
FROM osrf/ros:humble-desktop as base
and i followed installation steps.

**Expected behavior**
I don't know, never was able to run it

**Logs**
file 'ci402_setup.launch.py' was not found in the share directory of package 'canopen_tests' which is at '/home/canopen_ws/install/canopen_tests/share/canopen_tests'

**Setup:**  
- Docker image: FROM osrf/ros:humble-desktop as base
 - Device: Thinkpad P15s
 - OS: ubuntu 20.04 LTS
 - ROS-Distro: Humble
 - Branch/Commit: Humble

**Additional context**
What I have done so far : 
Created this docker image,
Sourced in the ~/.bashrc (inside the container): 
- source /opt/ros/humble/setup.bash
- source /canopen_ws/install/setup.bash
- and of course the bashrc itself

I run the generated image (the container) with this command line : 
canopen_core_1.2 = image name
docker run -it --network="host" canopen_core_1.2 bash
so i'm assuming that the container network is bind the host one
I ran commands to have the vcan interface on the host

I'm sorry if it's a basic issue for beginner, but i did not find a solution with an shamed amount of hours.

thank you in advance

ipa-cmh, 2023-11-06:
@SauceQuiPik can you try cia402_... instead of ci402_setup.launch.py

SauceQuiPik, 2023-11-07:
Hi ipa-cmh, it's working... Thank you very much ! 
In the ros web page of the package, it's written ci402 and not cia402, i just copy-pasted this line...

ipa-cmh, 2023-11-07:
Good catch. Will fix it asap.

closed 2023-11-22 by ipa-cmh