ros-industrial/ros2_canopen #225: Problem with booting the device

C-002, 2023-11-10:
**Describe the bug**
Hello! I'm trying to control a motor with ros2_canopen, following the tutorials ["How to create..."](https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-robot-system.html) and "canopen_tests" packages. However, after launching the launch file, the process stopped at booting the device and it responded "False" while calling the init trigger.

**Logs**

### launch file:

```python
import os
from ament_index_python import get_package_share_directory
from launch import LaunchDescription
import launch 
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    kinco_eds_path = os.path.join(
        get_package_share_directory("kinco_node"),
        "config",
        "kinco_config",
        "kinco.eds"
    )

    ld = LaunchDescription()
    # slave_node_1 = IncludeLaunchDescription(
    #     PythonLaunchDescriptionSource(
    #         [
    #             os.path.join(get_package_share_directory("canopen_fake_slaves"), "launch"),
    #             "/cia402_slave.launch.py",
    #         ]
    #     ),
    #     launch_arguments={
    #         "node_id": "2",
    #         "node_name": "kinco_node_1",
    #         "slave_config": kinco_eds_path,
    #     }.items(),
    # )

    master_bin_path = os.path.join(
        get_package_share_directory("canopen_tests"),
        "config",
        "kinco_config",
        "master.bin",
    )
    if not os.path.exists(master_bin_path):
        master_bin_path = ""

    device_container = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                os.path.join(get_package_share_directory("canopen_core"), "launch"),
                "/canopen.launch.py",
            ]
        ),
        launch_arguments={
            "master_config": os.path.join(
                get_package_share_directory("kinco_node"),
                "config",
                "kinco_config",
                "master.dcf",
            ),
            "master_bin": master_bin_path,
            "bus_config": os.path.join(
                get_package_share_directory("kinco_node"),
                "config",
                "kinco_config",
                "bus.yml",
            ),
            "can_interface_name": "can0",
        }.items(),
[... 225 lines cut ...]
(1699584944.633953) can0 080#
```

### init result:

```shell
user@user-Default-string:~$ ros2 service call /joint_1/init std_srvs/srv/Trigger 
requester: making request: std_srvs.srv.Trigger_Request()

response:
std_srvs.srv.Trigger_Response(success=False, message='')

```

**Setup:**
 - Device: PC
 - OS: Ubuntu 22.04.3 LTS (6.2.0-36-generic)
 - ROS-Distro: Humble
 - Branch/Commit: humble

**Additional context**

The output shows the ros2_canopen process read the device info by 0x1000 and 0x1018 and then it stopped. I'm a fish leg to use CANOpen and it's difficult for me to locate the problem. Any hints or suggestions?

Thanks a lot!

jimowei, 2023-11-13:
_**[device_container_node-1] [WARN] [1699584944.461996910] [joint_1]: Wait for device to boot.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)**_

Do not interrupt with ctrl-c.   when ros2_canopen get  "702#00",the process will continue.

ipa-cmh, 2023-11-14:
@C-002 seems there is no error happening in sdo communication. But it could be that the value of 1018sub2 in eds does not match what is actually on the device.

glaciercoder, 2024-03-26:
Have you solved the problem now? I'm meeting the same problem with you. I'm anxious now. @C-002

ZeroErrControl, 2025-01-25:
You can refer to [this project:https://github.com/ZeroErrControl/eRob_CANopen_Python](https://github.com/ZeroErrControl/eRob_CANopen_Python).