ros-industrial/ros2_canopen #227: Export canopen_fake_slaves include directories

MartinCornelis2, 2023-11-20:
Ubuntu 22.04
ROS Humble

I want to derive my own Mocked devices from either the ros2_canopen::BaseSlave or the ros2_canopen::BasicSlave, however the include folder does not seem to get exported.

Is there a reason why 
```
install(
  DIRECTORY include/
  DESTINATION include
)
ament_export_include_directories(
  include
)
```
is ommitted from the CMakelists.txt in the ROS Humble version (didn't check other versions yet)?

Or am I not supposed to derive my Mock devices from these classes?
With the way things are setup now you also have to override the entire run() of the BaseSlave/BasicSlave, just to feed your own slave class in i.e. in `cia402_slave.hpp`:
`ros2_canopen::CIA402MockSlave slave(timer, chan, slave_config_.c_str(), "", node_id_);` 
or in my case:
`BLinkMockSlave slave(timer, chan, slave_config_.c_str(), "", node_id_);`
I feel like this will lead to a lot of code duplication if you want to create multiple mocked devices.

Please let me know if I can/should open a PR, or if I am not supposed to use canopen_fake_slaves in this way.

ipa-cmh, 2023-11-21:
Honestly, the fake slaves package has not received much love. It would be wonderful to get some help for it.

kurtist123, 2024-03-11:
I implemented this enhancement here: https://github.com/ros-industrial/ros2_canopen/pull/265

I implemented the enhancement on the master branch but I would also like it on the humble branch. What is the process to get this back merged into the humble branch?

closed 2024-04-22 by ipa-vsp