ros-industrial/ros2_canopen #228: Pre-built binaries for Iron/Humble

baptistepetit, 2023-11-21:
Hi, first of all thank you so much for your work!

I just wanted to know if pre-built binaries being deployed for rolling but not humble/iron is a wanted state?
If it is wanted, is the blocker for releasing pre-built binaries for humble and iron to reach the beta release milestone?

ghanim-mukhtar, 2024-04-11:
Hi, also wanted to thank you for your work, 

And had the same question as @baptistepetit : any plans for release an official Debians for humble ?

roncapat, 2024-04-24:
I noticed there is a release in Rosdistro for `humble`, `jazzy` & `rolling`, not for Iron unfortunately...

You can see here: https://github.com/search?q=repo%3Aros%2Frosdistro%20lely_core_libraries&type=code

And here: https://github.com/ros2-gbp/ros2_canopen-release

ghanim-mukhtar, 2024-05-27:
@roncapat sorry for the late reply, yes you are right, I don't know if this has just been released last month or what but it is exactly what I was looking for. 
Thx a lot

closed 2024-05-27 by ipa-vsp