ros-industrial/ros2_canopen #232: cogen_dcf causing build to fail

angstrem98, 2023-11-25:
I tried to implement a simple package, but I'm getting this error during the build:

`sed: can't read /home/dusan/ros2_ws/build/my_canopen_config/config/cia402/preprocessed_bus.yml: No such file or directory
gmake[2]: *** [CMakeFiles/cia402.dir/build.make:74: cia402] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/cia402.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2`

ipa-cmh, 2023-11-25:
@angstrem98 can you provide the complete bus.yml?

angstrem98, 2023-11-25:
```
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 0
  package: "canopen_master_driver"
  driver: "ros2_canopen::MasterDriver"
  baudrate: 500
  sync_period: 10000    # us

defaults:
  dcf: "CDF3000.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  period: 10
  velocity_mode: 1
  revision_number: 0x00010000
  heartbeat_producer: 1000 # Heartbeat every 1000 ms
  scale_pos_to_dev : 1.0
  scale_vel_to_dev : 1.0
  scale_pos_from_dev : 1.0
  scale_vel_from_dev : 1.0
  sdo:
    - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
    - {index: 0x6081, sub_index: 0, value: 420}
    - {index: 0x6083, sub_index: 0, value: 800}
    - {index: 0x6060, sub_index: 0, value: 3} # 3 = profile velocity mode
  tpdo: # TPDO needed statusword, actual velocity, actual position, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6061, sub_index: 0} # mode of operation display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0} # position actual value
        - {index: 0x606c, sub_index: 0} # velocity actual position
  rpdo: # RPDO needed controlword, target position, target velocity, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0} # controlword
      - {index: 0x6060, sub_index: 0} # mode of operation
    2:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x60FF, sub_index: 0} # target velocity

nodes:
  joint1:
    node_id: 1
```
This builds on humble and iron.
Current error when launching:
```
[device_container_node-1] device_container_node: /home/dusan/iron_ws/build/lely_core_libraries/upstream/src/co/dev.c:710: co_dev_read_dcf: Assertion `dev' failed.
```
I'm sure that my physical CAN network is ok, because I was able to read out some data from drive using SDOs through a python library. Note that I'm trying to do 402 velocity control.

angstrem98, 2023-11-25:
Just randomly changed master node id to 5 and there was a lot of can traffic.
Snippet of current errors:
```
[device_container_node-1] [INFO] [1700945133.423022579] [device_container_node]: Load driver component.
[device_container_node-1] [INFO] [1700945133.423135567] [device_container_node]: Added /joint1 to executor
[device_container_node-1] [ERROR] [1700945133.447088988] [joint1]: Could not polling from config, setting to true.
[device_container_node-1] [ERROR] [1700945133.447274112] [joint1]: Could not read enable diagnostics from config, setting to false.
[device_container_node-1] [INFO] [1700945133.447532745] [joint1]: scale_pos_to_dev_ 1.000000
[device_container_node-1] scale_pos_from_dev_ 1.000000
[device_container_node-1] scale_vel_to_dev_ 1.000000
[device_container_node-1] scale_vel_from_dev_ 1.000000
[device_container_node-1] 
[device_container_node-1] [INFO] [1700945133.449494512] [joint1]: eds file /home/dusan/iron_ws/install/keba_canopen_config/share/keba_canopen_config/config/cia402/CDF3000.eds
[device_container_node-1] [INFO] [1700945133.449518593] [joint1]: bin file /home/dusan/iron_ws/install/keba_canopen_config/share/keba_canopen_config/config/cia402/joint1.bin
[device_container_node-1] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-1] Found rpdo mapped object: index=6060 subindex=0
[device_container_node-1] Found rpdo mapped object: index=60ff subindex=0
[device_container_node-1] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6061 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6064 subindex=0
[device_container_node-1] Found tpdo mapped object: index=606c subindex=0
[device_container_node-1] [WARN] [1700945133.486606971] [joint1]: Wait for device to boot.
[device_container_node-1] [ERROR] [1700945133.647628748] [joint1]: Boot Issue: Configuration download failed.
[device_container_node-1] [INFO] [1700945133.647697375] [joint1]: Driver booted and ready.
[device_container_node-1] [INFO] [1700945133.648170464] [joint1]: Starting with polling mode.
[device_container_node-1] [INFO] [1700945134.616780873] [joint1]: Slave 0x1: Switched NMT state to PREOP
```

angstrem98, 2023-11-25:
Changed master node id to 5 randomly and a lot of can traffic appeared.
Snippet of current errors:
```
[device_container_node-1] [INFO] [1700945133.423022579] [device_container_node]: Load driver component.
[device_container_node-1] [INFO] [1700945133.423135567] [device_container_node]: Added /joint1 to executor
[device_container_node-1] [ERROR] [1700945133.447088988] [joint1]: Could not polling from config, setting to true.
[device_container_node-1] [ERROR] [1700945133.447274112] [joint1]: Could not read enable diagnostics from config, setting to false.
[device_container_node-1] [INFO] [1700945133.447532745] [joint1]: scale_pos_to_dev_ 1.000000
[device_container_node-1] scale_pos_from_dev_ 1.000000
[device_container_node-1] scale_vel_to_dev_ 1.000000
[device_container_node-1] scale_vel_from_dev_ 1.000000
[device_container_node-1] 
[device_container_node-1] [INFO] [1700945133.449494512] [joint1]: eds file /home/dusan/ros2_ws/install/my_canopen_config/share/my_canopen_config/config/cia402/CDF3000.eds
[device_container_node-1] [INFO] [1700945133.449518593] [joint1]: bin file /home/dusan/ros2_ws/install/my_canopen_config/share/my_canopen_config/config/cia402/joint1.bin
[device_container_node-1] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-1] Found rpdo mapped object: index=6060 subindex=0
[device_container_node-1] Found rpdo mapped object: index=60ff subindex=0
[device_container_node-1] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6061 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6064 subindex=0
[device_container_node-1] Found tpdo mapped object: index=606c subindex=0
[device_container_node-1] [WARN] [1700945133.486606971] [joint1]: Wait for device to boot.
[device_container_node-1] [ERROR] [1700945133.647628748] [joint1]: Boot Issue: Configuration download failed.
[device_container_node-1] [INFO] [1700945133.647697375] [joint1]: Driver booted and ready.
[device_container_node-1] [INFO] [1700945133.648170464] [joint1]: Starting with polling mode.
[device_container_node-1] [INFO] [1700945134.616780873] [joint1]: Slave 0x1: Switched NMT state to PREOP
```

angstrem98, 2023-11-25:
Changing tpdo and rpdo numbers in bus.yml solved previous errors, rotor shaft position is being read properly, but I cannot give command to the motor.

miksiWan, 2024-02-01:
> Changing tpdo and rpdo numbers in bus.yml solved previous errors, rotor shaft position is being read properly, but I cannot give command to the motor. Pretty much motor driver does not enter operational state, power section is not activated. But both veocity and position are being read by ros.
> 
> Also, calling init service produces:
> 
> ```
> [device_container_node-1] async_sdo_read: id=2 index=0x6502 subindex=0 object does not exist
> [device_container_node-1] [ERROR] [1701003076.702080912] [joint1]: 02:6502:00: Object does not exist in the object dictionary (06020000): Object does not exist in the object dictionary
> ```

I'm having the same problem.
There is nothing for address **0x6502 (Supported Drive Modes)** in the my EDS file.
Is there any progress on this issue?