ros-industrial/ros2_canopen #235: Humble doesn't support create_client with the rclcpp::QoS object

leonce-m, 2023-12-08:
**Describe the bug**
When building canopen_core on a Jetson AGX Orin inside the Isaac ROS docker for ROS2 Humble, I get the following error:
```
--- stderr: canopen_core                               
/workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function ‘void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp::Node]’:
/workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:30:32: error: cannot convert ‘rclcpp::QoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   30 |     init_service_name, rclcpp::QoS(10), client_cbg_);
      |                                ^~~~~~~
      |                                |
      |                                rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:33,
                 from /workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function ‘void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp_lifecycle::LifecycleNode]’:
/workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:74:24: error: cannot convert ‘rclcpp::QoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   74 |     init_service_name, qos, client_cbg_);
      |                        ^~~
      |                        |
      |                        rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:996,
                 from /workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:34,
                 from /workspaces/isaac_ros-dev/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node_impl.hpp:103:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp_lifecycle::LifecycleNode::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  103 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
make[2]: *** [CMakeFiles/node_canopen_driver.dir/build.make:76: CMakeFiles/node_canopen_driver.dir/src/node_interfaces/node_canopen_driver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:143: CMakeFiles/node_canopen_driver.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< canopen_core [7.63s, exited with code 2]
```

**To Reproduce**
1. Clone [isaac_ros_common](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/tree/main) 
2. Follow the steps from nvidia to [install docker
](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/compute/index.html#jetson-platforms) and setup the [dev environment](https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html).
3. Build the default aarch64.ros2_humble docker image using run_dev.sh from [isaac_ros_common](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_common/index.html).
4. Build ros2_canopen inside the docker container.

**Expected behavior**
Build is successful.

**Setup:**
 - Device: Jetson AGX Orin
 - OS: Jetpack 5.1.2-b104
 - ROS-Distro: Humble
 - Branch/Commit: master

**Additional context**
In the latest commit for humble [2c8d2aa](https://github.com/ros2/rclcpp/commit/2c8d2aa4532e5d429b53f21eb911cc090e29057a) a similar issue is mentioned with ComponentManager.

ipa-vsp, 2023-12-09:
Hi @leonce-m,

Currently, master branch only supports rolling and iron. Therefore for humble please use humble branch. 

#216 , #217

leonce-m, 2023-12-09:
Aw shucks! My bad.

closed 2023-12-09 by leonce-m