ros-industrial/ros2_canopen #237: 将功能包git clone之后，直接进行colcon build ，出现了函数报错

woshikami, 2023-12-19:
这是编译报错日志，在/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp这个路径下的代码提示报错error: cannot convert ‘rclcpp::QoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}我该怎么解决这个问题呢？

colcon build
Starting >>> canopen_interfaces
Starting >>> lely_core_libraries
Finished <<< canopen_interfaces [25.0s]                                        
Starting >>> can_control
Finished <<< can_control [2.15s]                                               
Starting >>> canopen_utils
--- stderr: canopen_utils                                                      
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< canopen_utils [1.62s]
[Processing: lely_core_libraries]                                  
--- stderr: lely_core_libraries                                      
正克隆到 'upstream'...
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
HEAD 目前位于 7824cbb2 release v2.3.2
正在检查补丁 configure.ac...
正在检查补丁 python/dcf-tools/Makefile.am...
成功应用补丁 configure.ac。
成功应用补丁 python/dcf-tools/Makefile.am。
configure.ac:17: warning: The macro `AC_PROG_CC_STDC' is obsolete.
configure.ac:17: You should run autoupdate.
./lib/autoconf/c.m4:1666: AC_PROG_CC_STDC is expanded from...
configure.ac:17: the top level
configure.ac:81: warning: $as_echo is obsolete; use AS_ECHO(["message"]) instead
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
m4/ax_pthread.m4:88: AX_PTHREAD is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
configure.ac:81: the top level
configure.ac:434: warning: AC_CHECK_HEADERS($ixxat_header): you should use literals
./lib/autoconf/headers.m4:217: AC_CHECK_HEADERS is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
configure.ac:434: the top level
configure.ac:17: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'
configure.ac:602: installing './tap-driver.sh'
src/can/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
libtool: warning: relinking 'liblely-tap.la'
libtool: warning: relinking 'liblely-util.la'
libtool: warning: relinking 'liblely-can.la'
libtool: warning: relinking 'liblely-co.la'
libtool: warning: relinking 'liblely-io.la'
libtool: warning: relinking 'liblely-ev.la'
libtool: warning: relinking 'liblely-io2.la'
libtool: warning: relinking 'liblely-coapp.la'
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< lely_core_libraries [1min 3s]
Starting >>> canopen_core
Starting >>> canopen_fake_slaves                 
Finished <<< canopen_fake_slaves [29.9s]                                    
--- stderr: canopen_core                                     
/home/shy/can1_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function ‘void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp::Node]’:
/home/shy/can1_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:30:32: error: cannot convert ‘rclcpp::QoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   30 |     init_service_name, rclcpp::QoS(10),client_cbg_);
      |                                ^~~~~~~
      |                                |
      |                                rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/shy/can1_ws/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:33,
                 from /home/shy/can1_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  127 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/shy/can1_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function ‘void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp_lifecycle::LifecycleNode]’:
/home/shy/can1_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:73:32: error: cannot convert ‘rclcpp::QoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   73 |     init_service_name, rclcpp::QoS(10),client_cbg_);
      |                                ^~~~~~~
      |                                |
      |                                rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node.hpp:996,
                 from /home/shy/can1_ws/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:34,
                 from /home/shy/can1_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp_lifecycle/rclcpp_lifecycle/lifecycle_node_impl.hpp:103:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp_lifecycle::LifecycleNode::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’
  103 |   const rmw_qos_profile_t & qos_profile,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/node_canopen_driver.dir/build.make:76：CMakeFiles/node_canopen_driver.dir/src/node_interfaces/node_canopen_driver.cpp.o] 错误 1
gmake[1]: *** [CMakeFiles/Makefile2:211：CMakeFiles/node_canopen_driver.dir/all] 错误 2
gmake[1]: *** 正在等待未完成的任务....
gmake: *** [Makefile:146：all] 错误 2
---
Failed   <<< canopen_core [34.5s, exited with code 2]

Summary: 5 packages finished [1min 38s]
  1 package failed: canopen_core
  3 packages had stderr output: canopen_core canopen_utils lely_core_libraries
  8 packages not processed

peterchenyipu, 2024-03-06:
have you added the -b option to select the correct branch for your ros distro? for example "-b humble" when cloning the source code

G-Apple1, 2024-08-27:
> have you added the -b option to select the correct branch for your ros distro? for example "-b humble" when cloning the source code

Thank you very much for your tips. I have resolved the above issue with the following command:
`git clone https://github.com/ros-industrial/ros2_canopen.git -b humble `