ros-industrial/ros2_canopen #238: When I use PDO mode, an error occurs. How should I configure my own motor eds file

hhz0328, 2023-12-28:
User
ubuntu@iron:~/canopen_demo$ ros2 launch user_pkg pdo_1.launch.py 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-12-28-16-51-29-072150-iron-9859
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/bus.yml
[INFO] [launch.user]: /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/master.dcf
[INFO] [launch.user]: 
[INFO] [launch.user]: can0
[INFO] [cia402_slave_node-1]: process started with pid [9873]
[INFO] [device_container_node-2]: process started with pid [9875]
[device_container_node-2] [INFO] [1703753489.922491824] [device_container_node]: Starting Device Container with:
[device_container_node-2] [INFO] [1703753489.923571502] [device_container_node]: 	 master_config /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/master.dcf
[device_container_node-2] [INFO] [1703753489.923605567] [device_container_node]: 	 bus_config /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/bus.yml
[device_container_node-2] [INFO] [1703753489.923623703] [device_container_node]: 	 can_interface_name can0
[device_container_node-2] [INFO] [1703753489.925036264] [device_container_node]: Loading Master Configuration.
[device_container_node-2] [INFO] [1703753489.926416756] [device_container_node]: Load Library: /home/ubuntu/canopen_demo/install/canopen_master_driver/lib/libmaster_driver.so
[device_container_node-2] [INFO] [1703753489.934396826] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-2] [INFO] [1703753489.934460410] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-2] [INFO] [1703753489.951317208] [master]: NodeCanopenBasicMaster
[device_container_node-2] [INFO] [1703753489.951757326] [device_container_node]: Load master component.
[device_container_node-2] [INFO] [1703753489.951913367] [device_container_node]: Added /master to executor
[device_container_node-2] [INFO] [1703753489.980806848] [device_container_node]: Loading Driver Configuration.
[device_container_node-2] [INFO] [1703753489.980979511] [device_container_node]: Found device cia402_device_1 with driver ros2_canopen::Cia402Driver
[device_container_node-2] [INFO] [1703753489.981578186] [device_container_node]: Load Library: /home/ubuntu/canopen_demo/install/canopen_402_driver/lib/libcia402_driver.so
[device_container_node-2] [INFO] [1703753489.997309011] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::Cia402Driver>
[device_container_node-2] [INFO] [1703753489.997391649] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::Cia402Driver>
[device_container_node-2] [INFO] [1703753490.012270815] [device_container_node]: Load driver component.
[device_container_node-2] [INFO] [1703753490.012533284] [device_container_node]: Added /cia402_device_1 to executor
[device_container_node-2] [ERROR] [1703753490.046114944] [cia402_device_1]: Could not polling from config, setting to true.
[device_container_node-2] [ERROR] [1703753490.046479411] [cia402_device_1]: Could not read enable diagnostics from config, setting to false.
[device_container_node-2] [INFO] [1703753490.046968916] [cia402_device_1]: scale_pos_to_dev_ 1000.000000
[device_container_node-2] scale_pos_from_dev_ 0.001000
[device_container_node-2] scale_vel_to_dev_ 1000.000000
[device_container_node-2] scale_vel_from_dev_ 0.001000
[device_container_node-2] 
[device_container_node-2] [INFO] [1703753490.052038209] [cia402_device_1]: eds file /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/joint_ppm.eds
[device_container_node-2] [INFO] [1703753490.052113234] [cia402_device_1]: bin file /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/cia402_device_1.bin
[device_container_node-2] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-2] Found rpdo mapped object: index=6060 subindex=0
[device_container_node-2] Found rpdo mapped object: index=607a subindex=0
[device_container_node-2] Found rpdo mapped object: index=60ff subindex=0
[device_container_node-2] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-2] Found rpdo mapped object: index=607a subindex=0
[device_container_node-2] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-2] Found rpdo mapped object: index=60ff subindex=0
[device_container_node-2] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-2] Found tpdo mapped object: index=6061 subindex=0
[device_container_node-2] Found tpdo mapped object: index=6064 subindex=0
[device_container_node-2] Found tpdo mapped object: index=606c subindex=0
[device_container_node-2] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-2] Found tpdo mapped object: index=6064 subindex=0
[device_container_node-2] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-2] Found tpdo mapped object: index=606c subindex=0
[device_container_node-2] [WARN] [1703753490.057456316] [cia402_device_1]: Wait for device to boot.
[device_container_node-2] [ERROR] [1703753490.060617811] [cia402_device_1]: Boot Issue: Value of object 1000 from CANopen device is different to value in object 1F84 (Device type).
[device_container_node-2] [INFO] [1703753490.060710255] [cia402_device_1]: Driver booted and ready.
[device_container_node-2] [INFO] [1703753490.061356391] [cia402_device_1]: Starting with polling mode.
[cia402_slave_node-1] [INFO] [1703753490.100649420] [cia402_node_1]: Reaching inactive state.
[INFO] [launch.user]: node 'basic_slave_node' reached the 'inactive' state, 'activating'.
[cia402_slave_node-1] [INFO] [1703753490.106578367] [cia402_node_1]: Reaching active state.
[cia402_slave_node-1] terminate called after throwing an instance of 'std::system_error'
[cia402_slave_node-1]   what():  CanController: No such device
[ERROR] [cia402_slave_node-1]: process has died [pid 9873, exit code -6, cmd '/home/ubuntu/canopen_demo/install/canopen_fake_slaves/lib/canopen_fake_slaves/cia402_slave_node --ros-args -r __node:=cia402_node_1 -r __ns:=/ --params-file /tmp/launch_params_dt9lofcr'].

hhz0328, 2024-01-08:
[cia402_slave_node-1] [INFO] [1704680047.424936841] [cia402_node_1]: Reaching active state.
[cia402_slave_node-1] terminate called after throwing an instance of 'std::system_error'
[cia402_slave_node-1]   what():  CanController: No such device
[device_container_node-2] [ERROR] [1704680047.456682311] [cia402_device_1]: Could not polling from config, setting to true.
[device_container_node-2] [ERROR] [1704680047.457056491] [cia402_device_1]: Could not read enable diagnostics from config, setting to false.
[device_container_node-2] [INFO] [1704680047.457543020] [cia402_device_1]: scale_pos_to_dev_ 1000.000000
[device_container_node-2] scale_pos_from_dev_ 0.001000
[device_container_node-2] scale_vel_to_dev_ 1000.000000
[device_container_node-2] scale_vel_from_dev_ 0.001000
[device_container_node-2] 
[device_container_node-2] [INFO] [1704680047.461573453] [cia402_device_1]: eds file /home/ubuntu/canopen_demo/install/user_pkg/share/user_pkg/config/pdo_1_config/joint_ppm.eds

hhz0328, 2024-01-08:
terminate called after throwing an instance of 'std::system_error'
i don't know what causes it

millejade, 2024-08-02:
Hello!

I'm not sure if this will help but have you initialize your CAN device properly?
```shell
sudo ip link set can0 up type can bitrate 1000000
sudo ip link set can0 txqueuelen 1000
sudo ip link set up can0
```

You should set the `txqueuelen`.

I also encountered error `std::system error` but after properly initializing my CAN device, it now works for.

This might be unrelated but worth it to try.

ZeroErrControl, 2024-11-27:
> User ubuntu@iron:~/canopen_demo$ ros2 launch user_pkg pdo_1.launch.py [INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-12-28-16-51-29-072150-iron-9859 [INFO] [launch]: [...]

You can to follow the prompts to modify the 'master.dcf' file.

closed 2024-12-04 by hhz0328