ros-industrial/ros2_canopen #241: `ConfigurationManager::get_entry()` optional return value is not checked

roncapat, 2024-01-04:
**Describe the bug**
The software can crash in a very uninformative way if ever the `get_entry()` call fails internally.

```
[ros2_control_node-1] [INFO] [1704363943.705185838] [yaml-cpp]: Failed to load entry "driver" for device "" 
[ros2_control_node-1] terminate called after throwing an instance of 'std::bad_optional_access'
[ros2_control_node-1]   what():  bad optional access
```

**Expected behavior**
Return value of `get_entry()` to be cheched for existence before accessing, graceful fail/shutdown with context-aware error message / hint for the user if available to help fix configuration (fo example).

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: rolling
 - Branch/Commit: master

hellantos, 2024-04-18:
Good point, we should fix this.

BatisteAll, 2024-10-22:
Hi, I am getting this same error when launching my real robot and I cannot figure out where is the problem.
I am using the humble branch with ubuntu 22.04.
Here is the exact log after the resource manager of ros2 control has loaded the hardware interface:
`
1729604592.3077662 [ros2_control_node-12] [INFO] [1729604592.307596573] [yaml-cpp]: Failed to load entry "driver" for device "" 
1729604592.3080096 [ros2_control_node-12] terminate called after throwing an instance of 'std::bad_optional_access'
1729604592.3080616 [ros2_control_node-12]   what():  bad optional access
`

Do you have an idea how to fix this?