ros-industrial/ros2_canopen #243: Broken link in documentation

roncapat, 2024-01-04:
**Describe the bug**
At https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-cia301-system.html
the link in the last section is broken (https://github.com/ros-industrial/ros2_canopen/blob/master/canopen_ros2_control/launch/canopen_system.launch.py).

Side note: it would be better to provide also the example XML `<ros2_control>` block to the user.

closed 2024-04-23 by ipa-vsp

roncapat, 2024-04-23:
@ipa-vsp I've seen you closed, later reverted patch. Should this Issue be reopened again?

ipa-vsp, 2024-05-04:
@roncapat The pull request you submitted is for the 'gh-page' branch, which is automatically created by GitHub and will be overwritten during the next build. As a result, I will include it in issue #283.